evcc-io/evcc #20015: Graphical Overview of Solar Production and Consumption

David62215, 2025-03-22:
**Is your feature request related to a problem? Please describe.**  
Currently, there is no clear graphical representation of how much solar power (PV) has been generated and consumed over different time periods. I would like to have an overview of my energy production and consumption, not just "charging sessions" but for my entire household and EV charging. Without this, it is difficult to track daily, weekly, monthly, and yearly energy trends.  

**Describe the solution you'd like**  
I would like to see a graph that displays my solar energy production and consumption throughout the day, week, month, and year. This graph should show how much energy was produced by my PV system and how much was consumed by my house and EV charging. Additionally, I want to compare these values with the previous day, week, etc., to see trends and differences at a glance.  

**Describe alternatives you've considered**  
One alternative is to manually check daily energy reports or logs, but this is not as intuitive or efficient as a visual representation. Another option is using third-party tools, but they often lack integration with my existing system or require additional configuration.  

**Additional context**  
A clear and interactive graph would help users quickly understand their energy usage patterns and optimize their consumption. It would also provide insights into how much of the household energy demand is covered by solar power versus grid consumption.

andig, 2025-03-22:
Doing more charts is in the backlog. If you want to help, how exactly would this look and which information would it provide?

DavidBachDerEchte, 2025-03-22:
I will now think about a design for the graphs. And then upload it here.

andig, 2025-03-23:
Lets reopen with a concrete proposal. And no promisies, especially not short-term!

closed 2025-03-23 by andig

David62215, 2025-03-23:
Reference: #20047 
At the top right, it says "Today." However, when I go back, the date changes to yesterday or tomorrow.

If needed, I can also send you the .psd file.

(These are example data)
[image]

David62215, 2025-03-23:
@naltatis 
Ich glaub ihr könnt auch deutsch, deshalb würde ich jetzt auf Deutsch weitermachen.

Ich habe jetzt nur verstanden aus ihrem "Comment" in #20047, dass ich das Bild hier einfügen soll. Aber ich habe jetzt nicht verstanden was sie meinten mit "Time-based statistic".

naltatis, 2025-03-24:
> Aber ich habe jetzt nicht verstanden was sie meinten mit "Time-based statistic".

@David62215 https://github.com/evcc-io/evcc/issues/13050

stepir, 2025-04-26:
to be honest I'm exactly of the opposite idea - currently I use (in HA) tesla-style visualisation

[image]

and apex chart to have a SE-style graph

Still I wish to implement the simple power graph that EVCC is currently using:

[image]

I find it so immediate to read, simple while 100% representing what I need to know about my system. So if graph/charting will be ever redone, please consider keeping the actual.

No to criticise David's proposal, let me make it clear: it's just is that I think we have already many other systems to represents data over (longer period of) time. Your inverter platform in example? And keeping months, if not years, of data open additional challenges.

In my case I built what SE was already providing in HA because I wanted historical correlations of external data (the consumption of additional loads, the status of each single battery, I have a small external grid-tie PV of another brand) - is this what EVCC is aiming to achieve?

David62215, 2025-04-26:
For me, I just want one app that can display all the information I need or want. I don’t want to use multiple apps or Home Assistant — I prefer having everything in one place to keep it simpler for me. I know you can build custom setups with Home Assistant, but that’s not really my thing. I’m also trying to avoid using the other apps from Sungrow, ABB, or Bluelink. I like it simple: I want to easily see all the data, how each part is performing daily, weekly, monthly, and so on.

stepir, 2025-04-26:
understood and make sense - that's exactly the reason why I use HA, where I've integrated EVCC for handling its specifics advantages. HA integrate with mostly everything, it's basically its scope.