evcc-io/evcc #20032: SMA STP10SE Falsche Werte

Alek19901, 2025-03-23:
### Describe the bug

Hallo zusammen,

beim SMA STP10SE wird leider ein falscher Wert für die PV-Erzeugung hergenommen.

Es wird beim Laden der Batterie die Ladeleistung von der Erzeugung abgezogen. So kommt ein undefinierbarer Wert in der Übersicht zustande.

[image]

[image]

Da noch ein zweiter WR hinterlegt ist, aus welchem auch die Batterie geladen werden kann, kommt dann zustätzlich auch noch der Fehler, dass der Wert der PV-Erzeugung negativ ist.

pv 1 power: -551W is negative - check configuration if sign is correct

### Steps to reproduce

1. SMA STP10SE einbinden
2. Zweiten WR einbinden
3. SMA STP10SE Batterie verbinden
4. Akku laden ;-)

### Configuration details

```yaml
Configuration via UI
```

### Log details

```text
[site ] WARN 2025/03/23 08:32:50 pv 1 power: -1249W is negative - check configuration if sign is correct
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.201.1

michiproep, 2025-03-23:
Ich glaube nicht, dass es an evcc liegt. Ich hab auch den STP10SE und einen zweiten WR. Ich lese die Daten auch im iobroker aus und hab festgestellt, dass das seit dem letzten WR häufig vorkommt, dass die Werte vom Modbus ca. alle 2 bis 4 Minuten einmal komplett "zusammenbrechen".
Ich denke das liegt an SMA

Alek19901, 2025-03-23:
Ich lese via HomeAssistant ebenfalls aus - Da bricht nix zusammen ;-)

Man kann auch permanent nachvollziehen, dass das, was in die Batterie geladen wird vom Erzeugungswert abgezogen wird.
Aufgrund des 2. WR wird dann im Zweifelsfall auch mehr in die Batterie geladen, als der Tripower10 erzeugt. Dann kommt der negativwert zustande.

HomeAssi: (Ost1 + West1 = STP10SE / Ost2 + West2 = STP6)

[image]

EVCC:

[image]

premultiply, 2025-03-23:
Missing config.

premultiply, 2025-03-23:
Ich rate: Falsches Template benutzt.
Mit dem richtigen SMA-Hybrid-Template kann es keine negativen Werte für die PV-Erzeugung geben.

closed 2025-03-23 by premultiply