evcc-io/evcc #20050: Anbindung Wärmepumepensteuerung Dimplex WPM Touch

chackmac, 2025-03-23:
Hallo zusammen,

ich reihe mich hier mal ein würde mir eine Unterstützung der Wärmepumpensteuerung WPM Touch von Dimplex über Modbus TCP wünschen. Wäre toll wenn ihr das Umsetzen könntet.

Funktionen und Details sind hier zu finden: [Modbus TCP Anbindung - Dimplex Wiki](https://dimplex.atlassian.net/wiki/spaces/DW/pages/3303571457/Modbus+TCP+Anbindung)

Danke und Gruß
Martin

andig, 2025-03-23:
Das kannst du selber 😜. Gerne PR.

diddip21, 2025-03-26:
@chackmac du kannst dir z.B. gerne hier abschauen was für eine implementierung benötigt wird.
https://github.com/evcc-io/evcc/blob/master/templates/definition/charger/lambda-zewotherm.yaml
Interessant sind aus meiner Sicht folgende Komponenten:

Generell wie ist die WP angebunden Modbus TCP ?

- setmaxpower ( # PV Überschussleistung)
- power (# aktuelle Aufnahmeleistung [E] der WP)
- energy (# kumulierter Stromverbrauch (Energieaufnahme))
- temp (# IST Temperatur an einer geeigneten Stelle)

Dafür gilt es passende Modbus Register herauszusuchen

chackmac, 2025-03-27:
> https://dimplex.atlassian.net/wiki/spaces/DW/pages/3342204929/Modbus+TCP+-+Eing+nge
> 
> [@chackmac](https://github.com/chackmac) Wie wird die Dimplex gesteuert?

Aktuell nutze ich die Dimplex App über das eingebaute Netzwerkmodul. Angebunden über Modbus TCP.
Aber es sollte doch machbar sein, dass EVCC direkt die aktuelle Leistungsaufnahme der WP abfrägt und ggf. den Boostmodus bzw. eine andere Warmwasser-Solltemperatur aktiviert, oder?
Sorry, ich bin leider nicht so tief drin wie das alles funktioniert und euch allen sehr dankbar für die ganzen Funktionen und Updates. Ehrlichgesagt hab ich selbst alleine keine Ahnung was zu tun ist um das zu implementieren.

Danke und Gruß
Martin

thecem, 2025-03-27:
Es würde schon helfen wenn du mal schaust woher die Leistungswerte kommen, die Steuerung aussieht.
Was kann die Anzeigen welche Werte kann man ändern?
Wie sieht deine App aus? Etc.
Ohne das alles haben wir keine Möglichkeiten.

closed 2025-04-08 by github-actions[bot]