evcc-io/evcc #20064: Doc: PV Vorhersage ist nun eine Liste, in den Beispielen wird aber noch die alte Syntax verwendet

cschlipf, 2025-03-24:
### Describe the bug

Die Einträge unter tariffs/solar sind nun eine Liste: https://docs.evcc.io/docs/tariffs#pv-vorhersage

Bei den Beispielen der Anwender wird aber noch die alte Syntax verwendet, die zur Verwirrung führt: https://docs.evcc.io/docs/tariffs#forecastsolar

### Steps to reproduce

Siehe EVCC Dokumentation: https://docs.evcc.io/docs/tariffs#pv-vorhersage

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-03-24:
Die "alte" Syntax ist ebenso gültig. Das wird intern gehandled.

andig, 2025-03-24:
Gerne konkreter PR für gewünschte Änderung.

cschlipf, 2025-03-24:
> Die "alte" Syntax ist ebenso gültig. Das wird intern gehandled.

Hm... nachdem Import der alten Syntax in die DB lief es nicht mehr bis ich auf die Liste umgewandelt hatte.

andig, 2025-03-24:
Mangels Log/ Fehlermeldung kann man da nur in die Glaskugel schauen.

andig, 2025-03-24:
Ich habs grad probiert. 

```
solar:
  type: fixed
  price: 21
```

landet in den Settings, wird aber nach Neustart nicht angezeigt.

naltatis, 2025-03-24:
Beide Schreibweisen funktionieren. Um Verwirrung vorzubeugen würde find ich aber gut, wie @cschlipf vorgeschlagen hat, immer die Listenschreibweise in der Doku zu verwenden.

Das Problem, dass die Tarif-Box als umkonfiguriert angezeigt wurde hatte zwei Gründe:
- Tarife werden nur gepublisht wenns auch Ladepunkte gibt
- Die Existenz eines Solar Forecasts wurde nicht für die "ist konfiguriert" Logik berücksichtigt.

Beides ist in https://github.com/evcc-io/evcc/pull/20079 gefixt.

closed 2025-03-24 by naltatis