evcc-io/evcc #20065: Nach Umstellung auf DB funktioniert Grünstromindex nicht mehr

cschlipf, 2025-03-24:
### Describe the bug

Als ich noch die evcc.yaml benutzt habe lief die Konfiguration. Aber seitdem evcc migrate bekomme ich den Grünstromindex nicht mehr zum laufen. Fehlermeldungen:

> Request failed with status code 404.
tariff not available: cannot create tariff type 'template': cannot create tariff type 'grünstromindex': unexpected status: 503 (Service Unavailable).
GET http://home:7070/api/tariff/planner

und 

> planner: tariff not available: cannot create tariff type 'template': cannot create tariff type 'grünstromindex': unexpected status: 503 (Service Unavailable)

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
currency: EUR
grid:
    type: fixed
    price: 0.2756
feedin:
    type: fixed
    price: 0.082
co2:
    type: template
    template: grünstromindex
    zip: 71126
solar:
    - type: template
      site: XXXXX-XXXX-XXXX-XXXX
      template: solcast
      token: XXXXXXXXXXXXXXXXXXXX
```

### Log details

```text
[site ] DEBUG 2025/03/24 12:46:46 battery 1 power: -290W
[site ] DEBUG 2025/03/24 12:46:46 pv 2 power: 150W
[site ] DEBUG 2025/03/24 12:46:46 pv power: 670W
[site ] DEBUG 2025/03/24 12:46:47 battery 1 soc: 77%
[site ] DEBUG 2025/03/24 12:46:47 grid powers: [-0 -0 -0]W
[site ] DEBUG 2025/03/24 12:46:47 grid currents: [0.69 1.2 0.72]A
[site ] DEBUG 2025/03/24 12:46:47 battery has priority at soc 77% (< 80%)
[site ] DEBUG 2025/03/24 12:46:47 site power: 100W
[lp-1 ] DEBUG 2025/03/24 12:46:47 charge total import: 7950.484kWh
[lp-1 ] DEBUG 2025/03/24 12:46:47 charger status: B
[lp-1 ] DEBUG 2025/03/24 12:46:47 limitSoc reached: 80.0% > 80%
[site ] DEBUG 2025/03/24 12:46:47 solar forecast: accumulated 73.819kWh, produced 95.983kWh, scale 1.300
[site ] DEBUG 2025/03/24 12:47:01 ----
[site ] WARN 2025/03/24 12:47:01 planner: tariff not available: cannot create tariff type 'template': cannot create tariff type 'grünstromindex': unexpected status: 503 (Service Unavailable)
[lp-1 ] DEBUG 2025/03/24 12:47:01 charge power: 0W
[lp-1 ] DEBUG 2025/03/24 12:47:01 charge currents: [0.006 0.006 0.008]A
[lp-2 ] DEBUG 2025/03/24 12:47:01 charge power: 0W
[lp-2 ] DEBUG 2025/03/24 12:47:01 charge currents: [0.009 0.01 0.01]A
[site ] DEBUG 2025/03/24 12:47:01 grid power: -20W
[site ] DEBUG 2025/03/24 12:47:01 ext 1 power: 0W
[site ] DEBUG 2025/03/24 12:47:01 battery 1 power: 2420W
[site ] DEBUG 2025/03/24 12:47:01 pv 2 power: 150W
[site ] DEBUG 2025/03/24 12:47:01 pv 1 power: 530W
[site ] DEBUG 2025/03/24 12:47:01 pv power: 680W
[site ] DEBUG 2025/03/24 12:47:02 battery 1 soc: 77%
[site ] DEBUG 2025/03/24 12:47:02 grid powers: [-10 -0 -10]W
[site ] DEBUG 2025/03/24 12:47:02 grid currents: [-0.88 2.35 -0.56]A
[site ] DEBUG 2025/03/24 12:47:02 site power: 2500W
[lp-2 ] DEBUG 2025/03/24 12:47:02 charge total import: 2357.643kWh
[lp-2 ] DEBUG 2025/03/24 12:47:02 charger status: B
[lp-2 ] DEBUG 2025/03/24 12:47:02 plan: charge 42m56s between 2025-03-25 07:17:04 +0100 CET until 2025-03-25 08:00:00 +0100 CET (power: 11040W, avg cost: 0.000)
[lp-2 ] DEBUG 2025/03/24 12:47:02 pv charge current: 0A = 0A + -10.9A (2500W @ 1p)
[site ] DEBUG 2025/03/24 12:47:02 solar forecast: accumulated 73.826kWh, produced 95.986kWh, scale 1.300
[site ] DEBUG 2025/03/24 12:47:16 ----
[site ] WARN 2025/03/24 12:47:16 planner: tariff not available: cannot create tariff type 'template': cannot create tariff type 'grünstromindex': unexpected status: 503 (Service Unavailable)
[lp-2 ] DEBUG 2025/03/24 12:47:16 charge power: 0W
[lp-1 ] DEBUG 2025/03/24 12:47:16 charge power: 0W
[lp-1 ] DEBUG 2025/03/24 12:47:16 charge currents: [0.006 0.006 0.008]A
[lp-2 ] DEBUG 2025/03/24 12:47:16 charge currents: [0.009 0.01 0.01]A
[site ] DEBUG 2025/03/24 12:47:16 grid power: 40W
[site ] DEBUG 2025/03/24 12:47:16 ext 1 power: 0W
[site ] DEBUG 2025/03/24 12:47:16 battery 1 power: -320W
[site ] DEBUG 2025/03/24 12:47:16 pv 2 power: 150W
[site ] DEBUG 2025/03/24 12:47:16 pv 1 power: 520W
[site ] DEBUG 2025/03/24 12:47:16 pv power: 670W
[site ] DEBUG 2025/03/24 12:47:17 battery 1 soc: 77%
[site ] DEBUG 2025/03/24 12:47:17 grid powers: [50 -10 -0]W
[site ] DEBUG 2025/03/24 12:47:17 grid currents: [0.76 -1.22 0.74]A
[site ] DEBUG 2025/03/24 12:47:17 battery has priority at soc 77% (< 80%)
[site ] DEBUG 2025/03/24 12:47:17 site power: 140W
[lp-1 ] DEBUG 2025/03/24 12:47:17 charge total import: 7950.484kWh
[lp-1 ] DEBUG 2025/03/24 12:47:17 charger status: B
[lp-1 ] DEBUG 2025/03/24 12:47:17 limitSoc reached: 80.0% > 80%
[site ] DEBUG 2025/03/24 12:47:17 solar forecast: accumulated 73.834kWh, produced 95.989kWh, scale 1.300
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.201.1

andig, 2025-03-24:
Was sagt denn die GSI Doku?

mrvodoo, 2025-03-25:
Ich hatte am Wochenende den selben Fehler allerdings ohne vorher auf DB umgestellt zu haben. Laut Traces und auch per manueller Postman Abfrage der gsi API, gab es wohl ein Problem mit deren Server. Das ist inzwischen behoben. Die o.g. Fehlermeldung ist aber bei mir auch erst nach Neustart von evcc verschwunden. 
Jetzt tut alles wieder auch ohne Token.

closed 2025-03-25 by andig

cschlipf, 2025-03-25:
> Ich hatte am Wochenende den selben Fehler allerdings ohne vorher auf DB umgestellt zu haben. Laut Traces und auch per manueller Postman Abfrage der gsi API, gab es wohl ein Problem mit deren Server. [...]

Ah, das erklärt es natürlich auch. Als ich das Token hinzugefügt habe wurde der Server neu gestartet.