evcc-io/evcc #20070: Polestar no longer able to connect

ottomitschke, 2025-03-24:
### Describe the bug

Hi, 

since today, EVCC Version 0.201.1 (I tried also the nightly) drops the following log error:
`[main ] ERROR 2025/03/24 14:16:15 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'polestar': login failed: missing authorization code`

I have of course validated the login data in the evcc.yaml. 

Not sure if Polestar has again changed something?

Thanks for having a look at this bug!

### Steps to reproduce

1. Start EVCC 
2. Connect Polestar to Wallbox
3. Check the Log

### Configuration details

```yaml
[main  ] INFO 2025/03/24 14:20:32 evcc 0.201.1
[main  ] INFO 2025/03/24 14:20:32 using config file: /etc/evcc.yaml
[db    ] INFO 2025/03/24 14:20:32 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2025/03/24 14:20:32 connecting evcc-804021866 at tcp://172.17.17.10:1883
[mqtt  ] DEBUG 2025/03/24 14:20:32 tcp://172.17.17.10:1883 connected
[main  ] ERROR 2025/03/24 14:20:32 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'polestar': login failed: missing authorization code
[lp-1  ] DEBUG 2025/03/24 14:20:34 charge total import: 19098.432kWh

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 30s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: sonnenbatterie 
  usage: grid  
  host: 172.17.17.51  
  port: 8080  
  name: grid1
- type: template
  template: sonnenbatterie 
  usage: pv  
  host: 172.17.17.51  
  port: 8080  
  name: pv2
- type: template
  template: sonnenbatterie 
  usage: battery  
  host: 172.17.17.51  
  port: 8080  
  name: battery3
  token: *****
  capacity: 10
  chargepower: 4600

chargers:
- type: template
  template: keba-modbus 
  id: 255  
  host: 172.17.17.39  
  port: 502  
  modbus: tcpip  
  welcomecharge: false  
  name: wallbox5

vehicles:
- type: template
  template: polestar 
  title: Polestar 2  
  icon: car  
  user: *****
  password: *****
  vin: *****
  capacity: 78  
  phases: 3  
  cache: 5m  
  mode: pv  
  minCurrent: 6 
  maxCurrent: 16  
  name: ev4

loadpoints:
- title: Wallbox
  charger: wallbox5
  mode: pv

site:
  title: Hubertusweg
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

Version: `0.201.1`
```

### Log details

```text
[main ] ERROR 2025/03/24 14:16:15 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'polestar': login failed: missing authorization code
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.201.1

Lerua123, 2025-03-24:
I have the same problem with Polestar.

loebse, 2025-03-24:
Looks like Polestar changed the way to connect to the API again 😒 

This should fix it: https://github.com/evcc-io/evcc/pull/20081

closed 2025-03-25 by andig