evcc-io/evcc #20073: Show vehicleSoc and vehicleRange in /api/state even when vehice is not connected to loadpoint

etreml, 2025-03-24:
This request may not be of interest to most evcc users, but perhaps a solution can be found.
I use the evcc api to display information on smart devices (e.g. LaMetric Time). However, some of this information (e.g. vehicleSoc and vehicleRange) is not included in the API if the car is not connected to the loadpoint, even though the data is received from the cloud.
It might be possible to attach this data, which comes from the cloud, to the vehicle node.

vedje, 2025-03-24:
Maybe I didn't fully understand your request, but I use the parameter `poll` with `mode` value `allways` for the loadpoint, so the soc is displayed even when the car is not connected. It is then also published via the MQTT API under the topic evcc/loadpoints/1/vehicleSoc.

https://docs.evcc.io/docs/reference/configuration/loadpoints#poll

```
vehicles:
  - name: vehicle1
    title: Car
    type: custom    
    soc:
      source: mqtt
      topic: wican/car/soc

loadpoints:
  - title: Garage
    charger: wallbox1
    vehicle: vehicle1
    mode: pv
    soc:
      poll:
        mode: always
        interval: 5m
```

etreml, 2025-03-25:
Thank you @vedje for your reply. I think this might solve my problem. I have to check the new condiguration when I come home.

etreml, 2025-03-25:
Thanks @vedje, that solved my issue.

closed 2025-03-25 by etreml