evcc-io/evcc #20078: Go-E Controller nicht mehr als Zähler verwendbar

Thorium23, 2025-03-24:
Mit dem Upgrade auf Version 0.200 wurde angeregt, die Geräte über die grafische Oberfläche in der Datenbank anzulegen, statt über die evcc.yaml. Leider ging damit aber die Funktion verloren vom Go-E Controller gezielte Sensorwerte auszulesen. Das Gerät wird leider auch nicht in der Auswahlliste als Zähler für eine Wallbox angeboten.
Das war sehr praktisch, da der Go-E Controller gleich mit sechs Stromzangen geliefert wird und man diese beliebig auf die Verbraucher verteilen konnte. So konnte die Leistung von Wallboxen gemessen werden, die über keinen eigenen Leistungsmesser verfügen.

Über die evcc.yaml konnte man die Werte wie folgt auslesen:
```
name: goEctrl
type: custom
power:
 source: httpv
 uri: http://192.168.xxx.xxx/api/status?filter=ccp
 jq:  .ccp[6]
```

Das Array enthielt alle Werte und über den Index war es dann möglich die Leistung zu erhalten.

Es wäre super, wenn das auch in der neuen Oberfläche wieder möglich ist, den Go-E Controller zur Leistungsmessung zu verwenden.

mfuchs1984, 2025-03-24:
Custom meter sind aktuell nicht über die GUI möglich, aber du kannst sie nach wie vor in der yaml anlegen. Im epic https://github.com/evcc-io/evcc/issues/6029 ist der Punkt noch offen.

andig, 2025-03-24:
> Mit dem Upgrade auf Version 0.200 wurde angeregt, die Geräte über die grafische Oberfläche in der Datenbank anzulegen, statt über die evcc.yaml. 

Das überrascht mich. Gerne ausprobieren, aber Config UI ist immer noch experimentell.

closed 2025-03-24 by andig

Thorium23, 2025-03-25:
Danke für die Info. Dann freue ich mich wenn es so weit ist. Bis dahin teste ich weiter die experimentelle Version.