evcc-io/evcc #20089: Fehler Ladeleistung (fehlerhafter Wert) beim Beladen einer 2ten Batterieeinheit am Goodwe Hybridwechselrichter

VoltMike, 2025-03-25:
### Describe the bug

Guten Morgen evcc - Team,

kurze Übersicht zum WR und Batterie - Setup:

WR1 Hybridwecheselrichter Goodwe GW29.9ET (Anbindung Modbus TCP via Goodwe Dongle V2)
==> Batterie 1 BYD HVS 12,8
==> Batterie 2 BYD HVS 12,8 (**Auffälligkeit**)
WR2 Einspeisewechselrichter Goodwe GW25MT (Anbindung RS485-Modbus TCP - converter)

Uns ist aufgefallen, dass (bei ausreichender PV - Leistung) wenn die Beladung der Batterien startet reproduzierbar die Batterie 2 mit einer fehlerhaften Beladeleistung angezeigt wird. 
Unabhängig von der "realen" Ladeleistung (überprüft mit original Goodwe SolarGo - App) bleibt die Ladeleistung statisch auf (extrem hohe und unrealistischem Wert von) 65,5 kW stehen. Das Ganze ist solange der Fall, bis die Batterie 2 vollständig bis 100 % geladen ist. 
Batterie 1 stellt die Beladeleistung fehlerfrei über den ganzen Ladeverlauf dar. 
Die Auffälligkeit ist nur bei Batterie 2 zu beobachten.

Beim Entladen der beiden Batterien gibt es keine Auffälligkeiten, dies funktioniert problemlos mit durchgängig plausiblen Werten für Batterie 1 als auch für Batterie 2.

Im Anhang ein paar Screenshots mit dem Fehlerbild.

Vorab schon mal besten Dank für den Support.

Beste Grüße
[

[image]
[image]
[image]

### Steps to reproduce

Das Fehlerbild ist in der evcc Übersicht zu erkennen

### Configuration details

```yaml
Die Konfiguration erfolgte per UI, evcc.yaml enthält keine Anweisungen
```

### Log details

```text
[site ] DEBUG 2025/03/25 08:14:02 pv 1 power: 1656W
[site ] DEBUG 2025/03/25 08:14:02 battery 2 soc: 42%
[site ] DEBUG 2025/03/25 08:14:02 battery power: -66222W
[site ] DEBUG 2025/03/25 08:14:02 battery soc: 36%
[site ] DEBUG 2025/03/25 08:14:02 pv power: 2426W
[site ] DEBUG 2025/03/25 08:14:02 site power: -66245W
[lp-1 ] DEBUG 2025/03/25 08:14:02 charge total import: 12324.505kWh
[lp-1 ] DEBUG 2025/03/25 08:14:02 charger status: A
[site ] DEBUG 2025/03/25 08:14:12 ----
[lp-1 ] DEBUG 2025/03/25 08:14:12 charge power: 0W
[lp-1 ] DEBUG 2025/03/25 08:14:12 charge currents: [0 0 0]A
[site ] DEBUG 2025/03/25 08:14:12 grid power: -33W
[site ] DEBUG 2025/03/25 08:14:12 battery 2 power: -65535W
[site ] DEBUG 2025/03/25 08:14:12 battery 1 power: -681W
[site ] DEBUG 2025/03/25 08:14:12 pv 1 power: 1597W
[site ] DEBUG 2025/03/25 08:14:12 pv 2 power: 760W
[site ] DEBUG 2025/03/25 08:14:12 battery 1 soc: 30%
[site ] DEBUG 2025/03/25 08:14:12 battery 2 soc: 42%
[site ] DEBUG 2025/03/25 08:14:12 battery power: -66216W
[site ] DEBUG 2025/03/25 08:14:12 battery soc: 36%
[site ] DEBUG 2025/03/25 08:14:12 pv power: 2357W
[site ] DEBUG 2025/03/25 08:14:12 site power: -66249W
[lp-1 ] DEBUG 2025/03/25 08:14:12 charge total import: 12324.505kWh
[lp-1 ] DEBUG 2025/03/25 08:14:12 charger status: A
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

VoltMike, 2025-03-25:
Nachtrag:

Was noch auffällig ist, dass auch keine Entladeleistung aus Batterie 2 dargestellt wird. (müsste um die 4 kW liegen, referenziert aus Solargo - App).
Auch in der Überischt ist diese fehlende Batterieausspeiseleistung nicht mit eingerechnet und wird als ? angezeigt. 

[image]
[image]

andig, 2025-04-03:
> [site ] DEBUG 2025/03/25 08:14:12 battery 2 power: -65535W

Das ist falsch. Es fehlt die Config:

    evcc config

Was sagt

    evcc meter

VoltMike, 2025-04-04:
[image]

[image]

[image]

Guten Morgen, 

ich habe nochmal aktuelle Screenshots (nahzug zeitgleich erstellt) angehängt.

Wird noch weitere Infromation benötigt um den Fehler einzugrenzen?

Beste Grüße

andig, 2025-04-04:
https://github.com/evcc-io/evcc/issues/20089#issuecomment-2775741079

VoltMike, 2025-04-04:
Hi andig,

wie kann ich dir die gewünschten Daten bereitstellen bzw. auslesen.
Sorry für die Umstände ;)

Beste Grüße

VoltMike, 2025-04-13:
[image]

VoltMike, 2025-04-13:
[image]

andig, 2025-04-13:
Es braucht keine weiteren Screenshots sondern die Ausgabe der beiden Befehle oben.

VoltMike, 2025-04-15:
Wo genau kann ich die Befehle eintragen um die gewünschte Ausgabe zu erzeugen um sie bereitzustellen?

andig, 2025-04-18:
Keine Ahnung. Wenn nicht anders möglich dann einfach ohne HA!

VoltMike, 2025-04-19:
Kann es nicht sein, dass es sich um ein "Umrechnungs-/ Berechnungsthema" für den Leistungswert der BAT2 handelt?

Wegen dem Kommentar oben, könnte eine Direktinstallation von evcc das Problem lösen?

Danke schon mal!

andig, 2025-04-19:
Das kannst du ja leicht ausprobieren?

andig, 2025-04-19:
Mangels Log gewchlossen.

closed 2025-04-19 by andig