evcc-io/evcc #20091: Tatsächlicher Ladestrom pro Fahrzeug (bspw. Zoe)

dennzus, 2025-03-25:
Es wäre super, wenn tatsächliche / reelle Ladeströme einem Fahrzeug individuell angegeben werden könnten. 

Beispiel:
Ich habe bei meinem Setup das Problem, dass die Renault Zoe (phase 2) mit 3P 8A in Wirklichkeit nur 4,1 kW zieht. EVCC erwartet allerdings, dass 5,5 vom Dach kommen. Bei meiner 5,6 kWp Anlage ist das allerdings unrealistisch. Daher wird das 3-phasige Laden bei mir nie aktiv. Manuell ist das allerdings über Stunden kein Problem.

Wenn man EVCC irgendwie beibringen könnte, dass die Zoe bei 3P 8A nur 4,1kW "zieht", wäre das eine große Hilfe.

premultiply, 2025-03-25:
Es wird nur nach der tatsächlichen Wirkleistungsdifferenz am Netzzähler geregelt.

closed 2025-03-25 by premultiply

VolkerK62, 2025-03-25:
Du kannst mit `enable.threshold` eine abweichende Einschaltschwelle festlegen.

dennzus, 2025-03-25:
> Du kannst mit `enable.threshold` eine abweichende Einschaltschwelle festlegen.

Danke dir für den Tipp. Allerdings, wenn ich es richtig verstehe, würde er damit auch 1-phrasig um den entsprechenden Wert früher starten. Das mag eine Zoe leider gar nicht.