evcc-io/evcc #20094: Spotty Tariff Template fails

tpaul1611, 2025-03-25:
### Describe the bug

The tariff somehow fails to run with an 500 error. The Spotty API is available and is reachable on the same device.

### Steps to reproduce

1. configure tariff
2. start evcc or run debug

### Configuration details

```yaml
[main  ] INFO 2025/03/25 09:25:36 evcc 0.201.1
[main  ] INFO 2025/03/25 09:25:36 using config file: /app/evcc.yaml
[db    ] INFO 2025/03/25 09:25:36 using sqlite database: /root/.evcc/evcc.db
[main  ] ERROR 2025/03/25 09:25:37 creating vehicle my_car failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': token expired
[tariff] ERROR 2025/03/25 09:25:37 unexpected status: 500 (Internal Server Error)
[main  ] ERROR 2025/03/25 09:25:37 creating tariff grid failed: cannot create tariff type 'template': cannot create tariff type 'custom': unexpected status: 500 (Internal Server Error)

Konfiguration (/app/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 30s # control cycle interval

site:
  title: Zuhause
  meters:
    grid: my_grid

tariffs:
  grid:
    type: template
    template: spottyenergy
    contractid: *****
    pricetype: CONSUMPTION

vehicles:
  - name: my_car
    type: template
    template: tesla
    title: Model Y
    capacity: 75
    clientId: *****
    accessToken: *****
    refreshToken: *****
    vin: *****

# define your loadpoints according your needs
# see https://docs.evcc.io/en/docs/reference/configuration/loadpoints
loadpoints:
  - title: Garage # display name for UI
    charger: my_charger # charger
    vehicle: my_car # default vehicle

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  # replace with your real grid meter
  - name: my_grid
    type: template
    template: demo-meter
    usage: grid
    power: -1000 # 1 kW feed-in

# replace with your real charger
# see https://docs.evcc.io/docs/devices/chargers
chargers:
  - name: my_charger
    type: template
    template: demo-charger
    status: C # charging
    power: 2000 # 2 kW charging power
    enabled: true # optional

Version: `0.201.1`
```

### Log details

```text
docker run -v $(pwd)/evcc.yaml:/app/evcc.yaml -it evcc/evcc:latest evcc -c evcc.yaml -l debug tariff
[main  ] INFO 2025/03/25 08:56:14 evcc 0.201.1
[main  ] INFO 2025/03/25 08:56:14 using config file: evcc.yaml
[db    ] INFO 2025/03/25 08:56:14 using sqlite database: /root/.evcc/evcc.db
[tariff] ERROR 2025/03/25 08:56:14 unexpected status: 500 (Internal Server Error)
[main  ] ERROR 2025/03/25 08:56:14 creating tariff grid failed: cannot create tariff type 'template': cannot create tariff type 'custom': unexpected status: 500 (Internal Server Error)
grid:
[main  ] FATAL 2025/03/25 08:56:14 tariff not available: cannot create tariff type 'template': cannot create tariff type 'custom': unexpected status: 500 (Internal Server Error)
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

Version: `0.201.1`

andig, 2025-03-25:
Can you share a trace log?

tpaul1611, 2025-03-25:
The Tesla error ist just because I disabled the Tesla API yesterday as seen by "token expired" and should be unrelated.

[main ] INFO 2025/03/25 09:52:01 evcc 0.201.1
[main ] INFO 2025/03/25 09:52:01 using config file: /etc/evcc.yaml
[db ] INFO 2025/03/25 09:52:01 using sqlite database: /root/.evcc/evcc.db
[main ] INFO 2025/03/25 09:52:01 listening at :7070
[main ] ERROR 2025/03/25 09:52:01 creating vehicle model_y failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': token expired
[tariff] ERROR 2025/03/25 09:52:01 unexpected status: 500 (Internal Server Error)
[main ] ERROR 2025/03/25 09:52:01 creating tariff grid failed: cannot create tariff type 'template': cannot create tariff type 'custom': unexpected status: 500 (Internal Server Error)
[site ] INFO 2025/03/25 09:52:01 site config:
[site ] INFO 2025/03/25 09:52:01 meters: grid ✓ pv ✗ battery ✗
[site ] INFO 2025/03/25 09:52:01 grid: power ✓ energy ✗ currents ✗
[site ] INFO 2025/03/25 09:52:01 vehicles:
[site ] INFO 2025/03/25 09:52:01 vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site ] INFO 2025/03/25 09:52:01 tariffs:
[site ] INFO 2025/03/25 09:52:01 grid: TariffType(0)
[site ] INFO 2025/03/25 09:52:01 feed-in: ✗
[site ] INFO 2025/03/25 09:52:01 co2: ✗
[site ] INFO 2025/03/25 09:52:01 solar: ✓
[lp-1 ] INFO 2025/03/25 09:52:01 loadpoint 1:
[lp-1 ] INFO 2025/03/25 09:52:01 mode: off
[lp-1 ] INFO 2025/03/25 09:52:01 charger: power ✓ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1 ] INFO 2025/03/25 09:52:01 meters: charge ✓
[lp-1 ] INFO 2025/03/25 09:52:01 charge: power ✓ energy ✗ currents ✗
[lp-1 ] DEBUG 2025/03/25 09:52:01 phase timer inactive
[lp-1 ] DEBUG 2025/03/25 09:52:01 pv timer inactive
[lp-1 ] INFO 2025/03/25 09:52:01 vehicle updated: unknown -> Model Y
[lp-1 ] DEBUG 2025/03/25 09:52:01 max charge current: 6A
[site ] DEBUG 2025/03/25 09:52:01 ----
[site ] WARN 2025/03/25 09:52:01 planner: tariff not available: cannot create tariff type 'template': cannot create tariff type 'custom': unexpected status: 500 (Internal Server Error)
[lp-1 ] DEBUG 2025/03/25 09:52:01 charge power: 11000W
[site ] DEBUG 2025/03/25 09:52:01 grid power: 0W
[site ] DEBUG 2025/03/25 09:52:01 pv power: 11000W
[site ] DEBUG 2025/03/25 09:52:01 site power: 0W
[lp-1 ] DEBUG 2025/03/25 09:52:01 charger status: C
[lp-1 ] INFO 2025/03/25 09:52:01 car connected
[lp-1 ] DEBUG 2025/03/25 09:52:01 pv timer elapse
[lp-1 ] DEBUG 2025/03/25 09:52:01 pv timer inactive
[lp-1 ] INFO 2025/03/25 09:52:01 start charging ->
[lp-1 ] DEBUG 2025/03/25 09:52:01 wake-up timer: stop
[lp-1 ] DEBUG 2025/03/25 09:52:01 max charge current: 16A
[site ] DEBUG 2025/03/25 09:52:31 ----
[site ] WARN 2025/03/25 09:52:31 planner: tariff not available: cannot create tariff type 'template': cannot create tariff type 'custom': unexpected status: 500 (Internal Server Error)
[lp-1 ] DEBUG 2025/03/25 09:52:31 charge power: 11000W
[site ] DEBUG 2025/03/25 09:52:31 grid power: 0W
[site ] DEBUG 2025/03/25 09:52:31 pv power: 11000W
[site ] DEBUG 2025/03/25 09:52:31 site power: 0W
[lp-1 ] DEBUG 2025/03/25 09:52:31 charger status: C
[site ] DEBUG 2025/03/25 09:53:01 ----
[site ] WARN 2025/03/25 09:53:01 planner: tariff not available: cannot create tariff type 'template': cannot create tariff type 'custom': unexpected status: 500 (Internal Server Error)
[lp-1 ] DEBUG 2025/03/25 09:53:01 charge power: 11000W
[site ] DEBUG 2025/03/25 09:53:01 grid power: 0W
[site ] DEBUG 2025/03/25 09:53:01 pv power: 11000W
[site ] DEBUG 2025/03/25 09:53:01 site power: 0W
[lp-1 ] DEBUG 2025/03/25 09:53:01 charger status: C
[site ] DEBUG 2025/03/25 09:53:31 ----
...

tpaul1611, 2025-03-25:
@andig Could the issue be that the contractid is in quotes? 

as seen in the trace:
`[grid  ] TRACE 2025/03/25 10:10:02 GET https://i.spottyenergie.at/api/prices/CONSUMPTION/'*****'`

The stars are edited by me, but the quotes are there in the original log.

tpaul1611, 2025-03-25:
I have tested and can confirm that the request fails when the contractid is put within single quotes. But how can it be nobody else noticed?

andig, 2025-03-25:
Closed in 16332162b8da9610120292970cbb56441b28930d

closed 2025-03-25 by andig