evcc-io/evcc #20128: Support for Emporia EVSE (charger)

ojiudezue, 2025-03-26:
It's one of the most popular chargers in the US. The only way it does excess solar charging is with an add-on that costs $$

Having EVCC manage it would be a good step for this market

andig, 2025-03-26:
Does it have a usable API?

andig, 2025-03-27:
> Something similar but has some basic GET command to turn on EVSE and Emporia Energy plugs - https://github.com/magico13/PyEmVue/blob/master/api_docs.md

Thanks for sharing that. This does indeed seem to cover EVs.

andig, 2025-03-27:
What we would need now is a full HTTP MITM trace of the interaction between PyEmVue with the EVSE:

- authentication
- get status
- start/stop charging
- change charging current

Based on this we can look at implementation.

magico13, 2025-03-30:
It's pretty much just a single PUT to the `/devices/evcharger` endpoint to control basically everything, PyEmVue implements it [here](https://github.com/magico13/PyEmVue/blob/master/pyemvue/pyemvue.py#L241). Authentication is through AWS cognito, the client id and pool are detailed [here](https://github.com/magico13/PyEmVue/blob/master/pyemvue/auth.py#L10). Anyone with one of these EVSEs can just log into the web portal and save all the traffic. 

I had done some [documentation of their API](https://github.com/magico13/PyEmVue/blob/master/api_docs.md) but this is a bit outdated, and they're moving to a new V2 that I haven't built support for.

andig, 2025-04-10:
Unfortunately we didn't get enough details to find an approach. Closing.

closed 2025-04-10 by andig