evcc-io/evcc #20134: evcc 0.202.0 not starting

polojoe, 2025-03-26:
### Describe the bug

evcc 0.202 not starting any more, 0.201.1 has no issues.

[image]

### Steps to reproduce

1. pull evcc 0.202.2
2. start container

### Configuration details

```yaml
no evcc.yaml anymore
```

### Log details

```text
see debug log uploadet
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.202.0

andig, 2025-03-26:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2025-03-26 by andig

andre68723, 2025-03-26:
@polojoe same for me, are you using home assistant by any chance?

polojoe, 2025-03-26:
@andig https://paste.libreelec.tv/singular-buck.log
@andre68723 I'm using docker on libreelec

jove01, 2025-03-26:
same in homeassistant addon