evcc-io/evcc #20149: circuit power: read failed: EOF  2. Versuch vermutlichen Bug zu melden

nuiler, 2025-03-27:
### Describe the bug

Seit dem Release und Installation von 0.202.0 erscheint die Meldung circuit power: read failed: EOF

In der Version davor ist die Meldung nicht erschienen.

### Steps to reproduce

1. Installation von 0.202.0 
2.
3.
...

### Configuration details

```yaml
/app # evcc dump --cfg                                                           
[main  ] INFO 2025/03/27 06:29:21 evcc 0.202.0                                   
[main  ] INFO 2025/03/27 06:29:21 using config file: /etc/evcc.yaml              
[db    ] INFO 2025/03/27 06:29:21 using sqlite database: /root/.evcc/evcc.db     
                                                                                 
Fehlermeldung:                                                                   
                                                                                 
    database database is locked (5) (SQLITE_BUSY)                                
                                                                                 
Konfiguration (/etc/evcc.yaml):                                                  
                                                                                 
network:                                                                         
  # schema is the HTTP schema                                                    
  # setting to `https` does not enable https, it only changes the way URLs are ge
nerated                                                                          
  schema: http                                                                   
  # host is the hostname or IP address                                           
  # if the host name contains a `.local` suffix, the name will be announced on MD
NS                                                                               
  # docker: MDNS announcements don't work. host must be set to the docker host's 
name.                                                                            
  host: evcc.local                                                               
  # port is the listening port for UI and api                                    
  # evcc will listen on all available interfaces                                 
  port: 7070                                                                     
                                                                                 
interval: 15s # control cycle interval                                           
                                                                                 
log: debug                                                                       
                                                                                 
telemetry: true                                                                  
                                                                                 
meters:                                                                          
- type: template                                                                 
  template: varta-energiespeicher                                                
  usage: grid                                                                    
  host: 192.168.178.11  # IP-Adresse oder Hostname                               
  port: 502  # Port                                                              
  name: grid1                                                                    
- type: template                                                                 
  template: solarlog                                                             
  usage: pv                                                                      
  host: 192.168.178.10  # IP-Adresse oder Hostname                               
  port: 502  # Port                                                              
  name: pv2                                                                      
- type: template                                                                 
  template: varta-energiespeicher                                                
  usage: battery                                                                 
  host: 192.168.178.11  # IP-Adresse oder Hostname                               
  port: 502  # Port                                                              
  name: battery3                                                                 
  capacity: 11.7 # in kWh                                                        
- type: template                                                                 
  template: siemens-pac2200                                                      
[... 486 lines cut ...]
[heidel] TRACE 2025/03/27 06:53:24 modbus: recv 01 03 02 00 00 b8 44
[circuit-Wallbox] TRACE 2025/03/27 06:53:24 validate current: 0.0574A + (0A -> 0A) <= 16A ok
[circuit-Wallbox] TRACE 2025/03/27 06:53:24 validate power: 11.134W + (0W -> 0W) <= 11000W ok
[influx] TRACE 2025/03/27 06:53:24 write greenShareHome=map[value:0.35645933014354064] (map[])
[site ] DEBUG 2025/03/27 06:53:24 solar forecast: accumulated 0.391kWh, produced 0.028kWh, scale 0.072
[influx] TRACE 2025/03/27 06:53:24 write greenShareLoadpoints=map[value:0] (map[])
[influx] TRACE 2025/03/27 06:53:24 write tariffPriceHome=map[value:0.22111076555023929] (map[])
[influx] TRACE 2025/03/27 06:53:24 write tariffCo2Home=map[value:168.60765550239236] (map[])
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.202.0

VolkerK62, 2025-03-27:
copy/paste Fehler oder sind `maxCurrent` und `maxPower` tatsächlich leer?
```                                                                       
- name: main                                                                     
  title: Main Circuit # fuer das UI                                              
  maxCurrent:                                                                    
  maxPower:                                                                      
  meter: grid1                                                                   
  parent:    
```

nuiler, 2025-03-27:
circuits:
- name: main
  title: Main Circuit # fuer das UI
  maxCurrent: 
  maxPower: 
  meter: grid1
  parent: 
- name: Wallbox
  title: Wallbox # fuer das UI
  maxCurrent: 16
  maxPower: 11000
  meter: Wallboxen
  parent: main 
   
maxCurrent und maxPower sind wirklich leer

VolkerK62, 2025-03-27:
setz da mal (plausible) Werte ein. Ich vermute, dann ist der Fehler weg.

nuiler, 2025-03-27:
Ich habe folgene Meldung

[main ] FATAL 2025/03/27 08:48:44 circuit cannot create circuit 'main': meter does not support phase currents

nachdem ich die Config wie folgt angepasst habe

circuits:
- name: main
  title: Main Circuit # fuer das UI
  maxCurrent: 50 # 50A (optional) 
  maxPower: 30000 # 30kW (optional) 
  meter: grid1
  parent: 
- name: Wallbox
  title: Wallbox # fuer das UI
  maxCurrent: 16
  maxPower: 11000
  meter: Wallboxen
  parent: main

andig, 2025-03-27:
In https://github.com/evcc-io/evcc/pull/20093 we've changed the handling of meter failures for modbus by moving the retry from library into evcc core. When this was done, we've missed adding the required retry in the circuit logic.

closed 2025-03-27 by andig

nuiler, 2025-03-27:
Vielen Dank für den schnellen Bug-Fix