evcc-io/evcc #20160: evcc 0.202.1 doesn’t start in Homeassistant

gimmel, 2025-03-27:
### Describe the bug

Yesterday I got an evcc update message in Homeassistant for version 0.202.1. After updating, evcc didn’t start anymore. I jumped back to 0.201.1, which worked again. Tried a second time today, but 0.202.1 won’t start at all in Homeassistant.

### Steps to reproduce

1. Update evcc in Homeassistant to 0.202.1

### Configuration details

```yaml
# Es ist 1095 Tage gültig und kann jederzeit neu erzeugt werden.
sponsortoken: ..

# unique installation id
plant: 

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

interval: 30s # control cycle interval

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

modbusproxy:
  - port: 5021
    uri: 192.168.172.168:502
  #    # rtu: true
  #    # readonly: true # use `deny` to raise modbus errors

meters:
  - name: jrk39_grid
    type: template
    template: huawei-dongle-powersensor
    usage: grid

    modbus: tcpip
    id: 1
    host: 192.168.172.168
    port: 502
    timeout: 15s

  - name: jrk39_pv
    type: template
    template: huawei-dongle-powersensor
    usage: pv

    modbus: tcpip
    id: 1
    host: 192.168.172.168
    port: 502
    timeout: 15s

  - name: jrk39_battery
    type: template
    template: huawei-dongle-powersensor
    usage: battery

    modbus: tcpip
    id: 1
    host: 192.168.172.168
    port: 502
    timeout: 15s

chargers:
  - name: walle
    type: template
    template: zaptec
    user: 
    password: 

vehicles:
  - name: elmo
    title: Tesla Model Y
    type: custom
    icon: car
    capacity: 60
    soc:
      source: http
      uri: http://homeassistant.local:8123/api/states/sensor.elmo_batteriestand_last
      method: GET
      jq: .state
      headers:
        - content-type: application/json
        - Authorization: Bearer ..-
    range:
      source: http
      uri: http://homeassistant.local:8123/api/states/sensor.elmo_batteriereichweite_last
      method: GET
      jq: .state
      headers:
        - content-type: application/json
        - Authorization: Bearer ..-
    odometer:
      source: http
      uri: http://homeassistant.local:8123/api/states/sensor.elmo_kilometerzahler_last
      method: GET
      jq: .state
      headers:
        - content-type: application/json
        - Authorization: Bearer ..-
    status:
      source: http
      uri: http://homeassistant.local:8123/api/states/sensor.tesla_status_for_evcc
      method: GET
      jq: .state
      headers:
        - content-type: application/json
        - Authorization: Bearer ..-
    
  - name: zoe
    title: Renault Zoe
    type: template
    template: renault
    icon: car
    vin: 
    capacity: 50
    user: 
    password: 
    mode: pv # enable PV-charging when vehicle is identified
    mincurrent: 8
    maxcurrent: 32

# site describes the EVU connection, PV and home battery
site:
  title: JRK39
  meters:
    grid: jrk39_grid
    pv:
    - jrk39_pv
    battery:
    - jrk39_battery
  residualPower: 100 # additional household usage margin

loadpoints:
  - title: Wallbox # display name for UI
    charger: walle # Wallbox Referenz
    mode: pv # charge mode (off, now, minpv, pv)
    #resetOnDisconnect: true # set defaults when vehicle disconnects
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: -2000 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 5m # threshold must be exceeded for this long
      threshold: 2000 # maximum import power (W)
    guardDuration: 5m # switch charger contactor not more often than this (default 5m)

tariffs:
  grid:
    type: tibber
    token: "" # access token
    homeid: ""

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # EUR/kWh

  co2:
    # co2 tariff provides co2 intensity forecast and is for co2-optimized target charging if no variable grid tariff is specified
    type: grünstromindex # GrünStromIndex (Germany only)
    zip: 

  solar:
    type: template
    template: solcast
    site: 
    token:

# mqtt message broker
mqtt:
  broker: 192.168.172.20:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  user: 
  password: 

# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
    soc: # vehicle soc update event
      title: Soc updated
      msg: Battery charged to ${vehicleSoc:%.0f}%
    guest: # vehicle could not be identified
      title: Unknown vehicle
      msg: Unknown vehicle, guest connected?
  services:
```

### Log details

```text
There’s no log, the add-on doesn’t start at all.
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.202.1

mfuchs1984, 2025-03-27:
Usually, even when the addon does not successfully start, there are logs. Go to configuration, add-ons, evcc, log. Another place to look is configuration, system, logs.

andig, 2025-03-27:
/cc @thecem I wouldn't even know where to look?

gimmel, 2025-03-27:
@mfuchs1984: You are totally right. I forgot that adding have their own logs in Homeassistant. I added the log above. It’s a „panic: runtime error: invalid memory address or nil pointer dereference“

What’s really weird: now, after the fifth try, it suddenly works.

mfuchs1984, 2025-03-27:
Was the car connected during all 5 attempts?

andig, 2025-03-27:
Closed in ee157ee4473de8b7ca244cb71dfabe84bd4730a0

closed 2025-03-27 by andig