evcc-io/evcc #20171: Merkwürdiges Verhalten bei Forecast

userwithoutpassword, 2025-03-27:
### Describe the bug

[image]

Mir ist mit der Version 202.0 erstmal ein merkwürdiges Verhalten aufgefallen.
Der Screenshot zeigt den Forecast. etwa alle paar Sekunden wird die Kurve etwas höher bzw kleiner obwohl "Adjust" ausgeschaltet ist. Die angezeigten Werte bleiben identisch aber die Grafik änder sich immer leicht.
Mit Version 202.1 ist der Forecast komplett verschwunden sodass ich wieder zur 202.0 zurück bin.

### Steps to reproduce

n/a

### Configuration details

```yaml
n/a
```

### Log details

```text
n/a
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.202.0

userwithoutpassword, 2025-03-27:
Ich nutze solcast

marcelGoerentz, 2025-04-04:
Mir ist aufgefallen, dass nach der Zeitumstellung die Vorhersage nicht mehr mit der tatsächlichen Leistung übereinstimmt, die zu erwartenden Erträge starten vor Sonnenaufgang?

[image]
Sonnenaufgang ist aber erst ab ca. 7:00 morgen!

naltatis, 2025-04-04:
@marcelGoerentz das klingt nach einem Problem mit der Integration deiner genutzten Vorhersagequelle. Mach dazu gerne einen spezifischen Issue mit den Rohdaten: `http://evcc.local:7070/api/state?jq=.forecast` und den Daten, die du bei der Quelle angezeigt bekommst auf.

marcelGoerentz, 2025-04-04:
> [@marcelGoerentz](https://github.com/marcelGoerentz) das klingt nach einem Problem mit der Integration deiner genutzten Vorhersagequelle. Mach dazu gerne einen spezifischen Issue mit den Rohdaten: [...]

#20371 Erledigt

userwithoutpassword, 2025-04-04:
> [@marcelGoerentz](https://github.com/marcelGoerentz) das klingt nach einem Problem mit der Integration deiner genutzten Vorhersagequelle. Mach dazu gerne einen spezifischen Issue mit den Rohdaten: [...]

Habt ihr eigentlich beim letzten Update irgendetwas geändert beim Forecast weil Solcast funktioniert nicht mehr. Wird nicht mehr angezeigt mit der neuesten Version

closed 2025-04-16 by github-actions[bot]