evcc-io/evcc #20180: Tesla vehicle --start fails

tpaul1611, 2025-03-27:
### Describe the bug

Running the cli command "vehicle" with flag "--start" fails if no commandProxy and proxyToken are configured. If they are missing, the fleet API credentials should be used instead.

### Steps to reproduce

1. Configure a Tesla vehicle 
2. Run evcc vehicle <name> --start

### Configuration details

```yaml
vehicles:
  - name: tesla
    type: template
    template: tesla
    title: Tesla
    capacity: 75
    clientId: <redacted>
    accessToken: <redacted>    
    refreshToken: <redacted>
```

### Log details

```text
[main  ] INFO 2025/03/27 16:56:52 evcc 0.202.0
[main  ] INFO 2025/03/27 16:56:52 using config file: evcc.yaml
[db    ] INFO 2025/03/27 16:56:52 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2025/03/27 16:56:52 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/03/27 16:56:52 CREATE TABLE `settings` (`key` text,`value` text,PRIMARY KEY (`key`)) 0 <nil>
[db    ] TRACE 2025/03/27 16:56:52 SELECT * FROM `settings` 0 <nil>
[db    ] TRACE 2025/03/27 16:56:52 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2025/03/27 16:56:53 CREATE TABLE `configs` (`id` integer PRIMARY KEY AUTOINCREMENT,`class` integer,`type` text,`title` text,`icon` text,`product` text,`value` text) 0 <nil>
[db    ] TRACE 2025/03/27 16:56:53 SELECT * FROM `configs` WHERE `configs`.`class` = 3 0 <nil>
[tesla ] TRACE 2025/03/27 16:56:53 POST https://auth.tesla.com/oauth2/v3/token
[tesla ] TRACE 2025/03/27 16:56:53 client_id=***&grant_type=refresh_token&refresh_token=***
--
{"access_token":"redacted","expires_in":28800,"token_type":"Bearer"}
[tesla ] TRACE 2025/03/27 16:56:53 GET https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/users/region
[tesla ] TRACE 2025/03/27 16:56:53 {"response":{"region":"eu","fleet_api_base_url":"https://fleet-api.prd.eu.vn.cloud.tesla.com"}}
[tesla ] TRACE 2025/03/27 16:56:53 GET https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles
[tesla ] TRACE 2025/03/27 16:56:54 {"response":[{"id":redacted,"vehicle_id":redacted,"vin":"redacted","color":null,"access_type":"OWNER","display_name":"","option_codes":null,"granular_access":{"hide_private":false},"tokens":null,"state":"offline","in_service":false,"id_s":"redacted","calendar_enabled":true,"api_version":84,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false}],"pagination":{"previous":null,"next":null,"current":1,"per_page":1,"count":1,"pages":1},"count":1}
[tesla ] TRACE 2025/03/27 16:56:54 POST https://api.myteslamate.com/api/1/vehicles/redacted/command/charge_start
[tesla ] TRACE 2025/03/27 16:56:55 {"error":"Invalid API key: go to app.myteslamate.com\/fleet to check it"}
[main  ] ERROR 2025/03/27 16:56:55 start charge: 401 Unauthorized: Invalid API key: go to app.myteslamate.com/fleet to check it
[db    ] TRACE 2025/03/27 16:56:55 INSERT INTO `settings` (`key`,`value`) VALUES ("redacted") ON CONFLICT (`key`) DO UPDATE SET `value`=`excluded`.`value` 1 <nil>
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.202.0

andig, 2025-03-27:
Tbo, I don't recall how this used to work. But sending commands imho requires virtual key being installed on the vehicle that _that_ requires a command proxy? Or how are you ensuring that the vehicle key is present?

tpaul1611, 2025-03-27:
Yes I can ensure the presence of the key.

But I just checked and the command still would need to be signed, which is handled by the proxies. But if the private_key were a parameter of the vehicle, evcc could sign the commands and make it work without a proxy.

andig, 2025-03-27:
Happy to take a PR. Given this addresses a small tech-savvy aufience, adding command signing is nothing the core team is planning to work on.

tpaul1611, 2025-03-27:
Hmm I will think about it, but now that I have discovered it, its probably easier to just host my own proxy :)

closed 2025-03-27 by tpaul1611