evcc-io/evcc #20184: Seit Update 0.202.0 modbus Fehler

ganzton, 2025-03-27:
### Describe the bug

Hallo,

seit 0.202.0 gibt es unten stehende modbus Fehlermeldung. Nach downgrade auf 201.1 ist es wieder stabil.

### Steps to reproduce

1. Update auf von 0.201.1 auf 0.202.0 oder 0.202.1
2.
3.
...

### Configuration details

```yaml
leider weiß ich nicht, wie ich die Konfiguration aus dem neuen UI bekomme.
```

### Log details

```text
[site ] DEBUG 2025/03/27 17:59:03 pv 1 power: 1552W
[site ] DEBUG 2025/03/27 17:59:03 battery 1 power: 0W
[site ] DEBUG 2025/03/27 17:59:03 grid currents: [4.15 -1.5 -1.8]A
[site ] DEBUG 2025/03/27 17:59:03 battery 1 soc: 100%
[site ] DEBUG 2025/03/27 17:59:03 site power: 277W
[lp-1 ] DEBUG 2025/03/27 17:59:03 detected active phases: 1p
[lp-1 ] ERROR 2025/03/27 17:59:04 charge total import: modbus: exception '4' (server device failure), function '4'
[lp-1 ] ERROR 2025/03/27 17:59:04 charger status: modbus: exception '4' (server device failure), function '4'
[site ] DEBUG 2025/03/27 17:59:04 solar forecast: accumulated 0.133kWh, produced 0.300kWh, scale 2.250
[site ] DEBUG 2025/03/27 17:59:13 ----
[lp-1 ] DEBUG 2025/03/27 17:59:13 charge power: 1391W
[lp-1 ] DEBUG 2025/03/27 17:59:13 charge currents: [6 0 0]A
[site ] DEBUG 2025/03/27 17:59:13 grid power: 251W
[site ] DEBUG 2025/03/27 17:59:13 pv 1 power: 1483W
[site ] DEBUG 2025/03/27 17:59:13 battery 1 power: 0W
[site ] DEBUG 2025/03/27 17:59:13 grid currents: [4.24 -1.4 -1.69]A
[site ] DEBUG 2025/03/27 17:59:13 battery 1 soc: 100%
[site ] DEBUG 2025/03/27 17:59:13 site power: 351W
[lp-1 ] DEBUG 2025/03/27 17:59:13 detected active phases: 1p
[lp-1 ] ERROR 2025/03/27 17:59:13 charge total import: modbus: exception '4' (server device failure), function '4'
[lp-1 ] DEBUG 2025/03/27 17:59:13 charger status: C
[lp-1 ] DEBUG 2025/03/27 17:59:13 soc estimated: 70.01% (vehicle: 70.00%)
[lp-1 ] DEBUG 2025/03/27 17:59:13 vehicle soc: 70%
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0202.1

andig, 2025-03-27:
Bitte Remotezugang zum Charger an info@evcc.io

ganzton, 2025-03-27:
Oh, damit bin ich leicht überfordert ...

gibt es eine Anleitung dazu?

Danke für die schnelle Hilfe!

Robert

ganzton, 2025-03-27:
Es ist ein Sungrow Charger.

andig, 2025-03-27:
Leider nein. Wireguard, FritzVPN, SSH etc wären möglichkeiten. Ich mach zu bis Config/ Zugang vorliegen. Ohne gehts leider nicht.

ganzton, 2025-03-27:
Alles klar.

ganzton, 2025-03-27:
Noch eine Frage:
Ist mit Remote-Zugang die Modbus über TCP Verbindung gemeint? Etwas anderes wüsste ich nicht.

ganzton, 2025-03-27:
Und noch die Frage:
Wie komme ich an die Config? Seit dem neuen UI ist meine yaml sehr leer.

andig, 2025-03-28:
evcc config

> Ist mit Remote-Zugang die Modbus über TCP Verbindung gemeint? Etwas anderes wüsste ich nicht.

Genau. Am allereinfachsten ist Portweiterleitung, aber dann hängt Dein Gerät offen im Internet...