evcc-io/evcc #20185: Power of Battery 2 wrong on Goodwe Hybrid Inverters

lutzvahl, 2025-03-27:
### Describe the bug

This is related to https://github.com/evcc-io/evcc/issues/15085

IMO the modbus register for battery 2 power is wrong.

### Steps to reproduce

1. Try to setup a battery - validate the modbus connection. 
2. Wrong Power values are shown
3.
...

### Configuration details

```yaml
![Image](https://github.com/user-attachments/assets/d91ad998-cde9-40f6-b02b-11617697844c)
```

### Log details

```text
[site ] INFO 2025/03/27 18:16:31 pv 1: power ✓ energy ✗ currents ✗
[site ] INFO 2025/03/27 18:16:31 battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✓
[site ] INFO 2025/03/27 18:16:31 battery 2: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓

[site ] DEBUG 2025/03/27 18:16:31 battery 1 power: 557W
[site ] DEBUG 2025/03/27 18:16:31 battery 1 soc: 51%
[site ] DEBUG 2025/03/27 18:16:31 battery 2 soc: 93%
[site ] DEBUG 2025/03/27 18:16:31 battery power: 36569646W
[site ] DEBUG 2025/03/27 18:16:31 battery soc: 68%
[site ] DEBUG 2025/03/27 18:16:31 site power: 36569749W <----
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.202.1 and evcc 0.202.1 (6f5c0b64) Nightly

andig, 2025-03-27:
Missing config

lutzvahl, 2025-03-27:
> Missing config

Configured via the UI, added a yaml now too

lutzvahl, 2025-03-27:
Looking at other [HA GoodWe Modbus integrations](https://github.com/marcelblijleven/goodwe/blob/58c7f8e4b0b95ce6c3ff2fbd61b4c32a6c3665c2/goodwe/et.py#L156C32-L156C37) the correct register seems to be **35264** and not 35265

andig, 2025-03-27:
@lutzvahl gerne PR.

/cc @UserChristian ich wundere mich, dass das in 1 Jahr niemand bemerkt hat?

lutzvahl, 2025-03-27:
> [@lutzvahl](https://github.com/lutzvahl) gerne PR.
> 
> /cc [@UserChristian](https://github.com/UserChristian) ich wundere mich, dass das in 1 Jahr niemand bemerkt hat?

Das ist eine gute Frage!

Habe es eben validiert.
[image]
HA Modbus config
[image]

PR kommt sofort

lutzvahl, 2025-03-27:
PR https://github.com/evcc-io/evcc/pull/20193
@andig PTAL

closed 2025-03-27 by andig

lutzvahl, 2025-03-31:
Confirming this is working now with nightly.20250331-51234aa