evcc-io/evcc #20198: Dynamic battery or vehicle optimization

m4ttb1ss, 2025-03-28:
**Is your feature request related to a problem? Please describe.**
Depending on the days when I or my wife need to drive to work earlier or later, I would like to be able to dynamically change the settings for Home battery or vehicle charging prioritization.
By default, I've set the settings to "Prioritize home battery until it reaches 90%", then the last 10% battery supported charging. That means in the morning, before we drive to work, the battery is charged first, but I would like the car using the first surplus before we start driving to work. On other days, for example on weekends, I'd like the battery to be charged first and then the car can charge the rest of the day.
This feature from my POV is especially helpful in the summer months, where there is production already in the morning from 6am on.

**Describe the solution you'd like**
I see two possibilities:

1. A schedule for home battery settings similar to the charging schedule
2. A one-time setting that changes back to default once the car is unplugged, similar to the charging mode, which is for instance by default on "PV only", can be set to "Fast" manually, and changes back to "PV only" once the car is unplugged.

**Describe alternatives you've considered**
Alternatives are currently to set manually Min+PV in the morning after waking up, or to change battery settings in the morning manually and then back to "default" manually later.

**Additional context**
None

andig, 2025-03-28:
> That means in the morning, before we drive to work, the battery is charged first, but I would like the car using the first surplus before we start driving to work. On other days, for example on weekends, I'd like the battery to be charged first and then the car can charge the rest of the day.

What you're really saying is that the requirements depend. For this purpose. evcc has the APIs.

> A schedule for home battery settings similar to the charging schedule

That's something we're considering, see https://github.com/evcc-io/evcc/issues/17424

> A one-time setting that changes back to default once the car is unplugged, similar to the charging mode, which is for instance by default on "PV only", can be set to "Fast" manually, and changes back to "PV only" once the car is unplugged.

There's already the experimental battery boost.

m4ttb1ss, 2025-03-28:
> There's already the experimental battery boost.

Heißt, wenn ich den Battery Boost einschalte und die Batterie ist leer, dann lädt evcc mit dem Überschuss, auch wenn die Battery Settings auf Battery first sind? Das würde schon mal helfen

andig, 2025-03-28:
Das sollte so sein.

m4ttb1ss, 2025-03-28:
Ok. Teste ich morgen mal. Danke soweit, ich geb hier Rückmeldung.

Chris45671, 2025-03-28:
Wenn Du zusätzlich zu EVCC auch Home Assistent einsetzt, könntest du folgendes probieren. 
#20105 

Über diese Automatisierung wird das laden der Hausbatterie in der früh auf 0% gesetzt und über den Tag nach oben geschraubt damit zum Abend die Batterie wieder voll ist, aber in der früh der Strom ins Auto fließen kann.

m4ttb1ss, 2025-03-28:
> Wenn Du zusätzlich zu EVCC auch Home Assistent einsetzt, könntest du folgendes probieren. [#20105](https://github.com/evcc-io/evcc/discussions/20105)
>
> Über diese Automatisierung wird das laden der [...]

Ich nutze Smarthomeng, eventuell bau ich mir da mal ein Python-Script zusammen. Danke für den Denkanstoß

closed 2025-03-28 by andig