evcc-io/evcc #20203: Ladeplan für "diesen"/ einen Tag pausieren

alpenseppl, 2025-03-28:
Hallo,
ich hatte einen Tag in der Woche frei und habe deswegen den Ladeplan ausgeschaltet, damit das Fz tagsüber über solar laden kann. Natürlich habe ich am Folgetag wieder vergessen den Plan einzuschalten, was zu einem zu niedrigen Ladestand am darauffolgenden Tag (3ter Tag) führte.

Deswegen die Idee, bei der Ladeplanung einen Schalter mit der Funktion einzurichten, der die Ladeplanung für einen Tag aussetzt und sich dann wieder selbst "abhakt" und der Ladeplan automatisch wieder aufgenommen wird.

Ggf finden das noch bei anderen Anklang.

detlefheese, 2025-03-28:
Wirklich nette Idee.

mfuchs1984, 2025-03-28:
Wäre auch für mich ein hilfreiches Feature.

naltatis, 2025-03-28:
Ich sehe den Nutzen. Der Aufwand/Komplexität für eine saubere und nachvollziehbare Umsetzung (Restart, Kommunikation in der UI, Fehlerpotential, ...) würde ich relativ hoch einschätzen. Daher würd ich das jetzt hier erst einmal schließen. Wir haben einige andere Themen (saubere Integration von Wärmeerzeuger, Fertigstellung Config UI, Auswertungen) die höhere Priorität haben. Wir können das aber zu einem späteren Zeitpunkt gerne noch mal wieder aufmachen.

closed 2025-03-28 by naltatis

docolli, 2025-09-30:
Dieses Feature würde ich auch begrüßen! Am besten noch mit REST API, damit man den "Skip" auch von extern auslösen kann.

pjjjv, 2026-08-19:
New PR: https://github.com/evcc-io/evcc/pull/32972/