evcc-io/evcc #20220: Application is crashing when starting the charge

narcismatei, 2025-03-28:
### Describe the bug

After the latest update evcc fails to start with the following stacktrace
Mar 28 11:45:37 raspberrypi evcc[110759]: panic: runtime error: invalid memory address or nil pointer dereference
Mar 28 11:45:37 raspberrypi evcc[110759]: [signal SIGSEGV: segmentation violation code=0x1 addr=0x28 pc=0xf34094]
Mar 28 11:45:37 raspberrypi evcc[110759]: goroutine 247 [running]:
Mar 28 11:45:37 raspberrypi evcc[110759]: github.com/evcc-io/evcc/charger.(*Zaptec).Identify(0x57d52b0?)
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/evcc-io/evcc/charger/zaptec.go:312 +0xf4
Mar 28 11:45:37 raspberrypi evcc[110759]: github.com/evcc-io/evcc/core.(*Loadpoint).createSession(0x4000cfc908)
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/evcc-io/evcc/core/loadpoint_session.go:43 +0x1dc
Mar 28 11:45:37 raspberrypi evcc[110759]: github.com/evcc-io/evcc/core.(*Loadpoint).evVehicleConnectHandler(0x4000cfc908)
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/evcc-io/evcc/core/loadpoint.go:493 +0x204
Mar 28 11:45:37 raspberrypi evcc[110759]: reflect.Value.call({0x217f300?, 0x400122c4c0?, 0x7ffefd458a78?}, {0x285572e, 0x4}, {0x585d1c0, 0x0, 0x6b738?})
Mar 28 11:45:37 raspberrypi evcc[110759]: 	reflect/value.go:581 +0x978
Mar 28 11:45:37 raspberrypi evcc[110759]: reflect.Value.Call({0x217f300?, 0x400122c4c0?, 0x0?}, {0x585d1c0?, 0x10?, 0x40008bcd70?})
Mar 28 11:45:37 raspberrypi evcc[110759]: 	reflect/value.go:365 +0x94
Mar 28 11:45:37 raspberrypi evcc[110759]: github.com/asaskevich/EventBus.(*EventBus).doPublish(0x2351400?, 0x400124d320, {0x2861a6e?, 0x400122c4d0?}, {0x0?, 0x1db9ff8?, 0x40012cd888?})
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/asaskevich/EventBus@v0.0.0-20200907212545-49d423059eef/event_bus.go:160 +0x60
Mar 28 11:45:37 raspberrypi evcc[110759]: github.com/asaskevich/EventBus.(*EventBus).Publish(0x4000c529e0, {0x2861a6e, 0x7}, {0x0, 0x0, 0x0})
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/asaskevich/EventBus@v0.0.0-20200907212545-49d423059eef/event_bus.go:144 +0x588
Mar 28 11:45:37 raspberrypi evcc[110759]: github.com/evcc-io/evcc/core.(*Loadpoint).updateChargerStatus(0x4000cfc908)
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/evcc-io/evcc/core/loadpoint.go:1065 +0x484
Mar 28 11:45:37 raspberrypi evcc[110759]: github.com/evcc-io/evcc/core.(*Loadpoint).Update(0x4000cfc908, 0xc0ad2c5e353f7cee, 0x0, {0x4000f56008, 0xac, 0x124}, 0x0, 0x0, 0x3ff0000000000000, 0x4000ad0f50, ...)
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/evcc-io/evcc/core/loadpoint.go:1775 +0x2c8
Mar 28 11:45:37 raspberrypi evcc[110759]: github.com/evcc-io/evcc/core.(*Site).update(0x40000e7208, {0x3230748, 0x4000cfc908})
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/evcc-io/evcc/core/site.go:924 +0x7a4
Mar 28 11:45:37 raspberrypi evcc[110759]: github.com/evcc-io/evcc/core.(*Site).Run(0x40000e7208, 0x4000fcf490, 0x6fc23ac00)
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/evcc-io/evcc/core/site.go:1043 +0x1a4
Mar 28 11:45:37 raspberrypi evcc[110759]: github.com/evcc-io/evcc/cmd.runRoot.func4()
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/evcc-io/evcc/cmd/root.go:325 +0x2c
Mar 28 11:45:37 raspberrypi evcc[110759]: created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/evcc-io/evcc/cmd/root.go:324 +0x1bf4
Mar 28 11:45:37 raspberrypi systemd[1]: evcc.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Mar 28 11:45:37 raspberrypi systemd[1]: evcc.service: Failed with result 'exit-code'.
^C

### Steps to reproduce

Upgrade to 0.202.2 version and now the application fails to start

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: raspberrypi.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 8c89f4f77bf6f67324c99a7363856f2e0e3b5a0482f8672f0538ac7b1cb999d0

sponsortoken: 

interval: 30s # control cycle interval

meters:
- type: template
  template: solaredge-inverter
  id: 1
  host: 192.168.0.11
  port: 1502
  usage: pv
  modbus: tcpip
  timeout: 30s
  name: pv2

site:
  title: Narcis House
  residualPower: 1500
  meters:
    pv:
    - pv2

tariffs:
  currency: CHF
  grid:
    type: fixed
    price: 0.29 # EUR/kWh (default)
    zones:
      - days: Mo-So
        hours: 7-21
        price: 0.2917
      - days: Mo,So
        hours: 21-0
        price: 0.1881
      - days: Mo,So
        hours: 0-7
        price: 0.1881
      - days: Mo,So
        hours: 7-21
        price: 0.29
  feedin:
    type: fixed
    price: 0.14

chargers:
  - name: ZAP339850
    type: template
    template: zaptec
    id: 51267d05-8b23-4bf7-84e9-e1dd0258cc9e
    user: 
    password: 

vehicles:
  - name: kia_ev6
    type: template
    template: kia
    title: Kia EV6
    capacity: 77 # kWh
    user: 
    password: 
    vin: 
~
```

### Log details

```text
Mar 28 11:45:37 raspberrypi evcc[110759]: panic: runtime error: invalid memory address or nil pointer dereference
Mar 28 11:45:37 raspberrypi evcc[110759]: [signal SIGSEGV: segmentation violation code=0x1 addr=0x28 pc=0xf34094]
Mar 28 11:45:37 raspberrypi evcc[110759]: goroutine 247 [running]:
Mar 28 11:45:37 raspberrypi evcc[110759]: github.com/evcc-io/evcc/charger.(*Zaptec).Identify(0x57d52b0?)
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/evcc-io/evcc/charger/zaptec.go:312 +0xf4
Mar 28 11:45:37 raspberrypi evcc[110759]: github.com/evcc-io/evcc/core.(*Loadpoint).createSession(0x4000cfc908)
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/evcc-io/evcc/core/loadpoint_session.go:43 +0x1dc
Mar 28 11:45:37 raspberrypi evcc[110759]: github.com/evcc-io/evcc/core.(*Loadpoint).evVehicleConnectHandler(0x4000cfc908)
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/evcc-io/evcc/core/loadpoint.go:493 +0x204
Mar 28 11:45:37 raspberrypi evcc[110759]: reflect.Value.call({0x217f300?, 0x400122c4c0?, 0x7ffefd458a78?}, {0x285572e, 0x4}, {0x585d1c0, 0x0, 0x6b738?})
Mar 28 11:45:37 raspberrypi evcc[110759]: 	reflect/value.go:581 +0x978
Mar 28 11:45:37 raspberrypi evcc[110759]: reflect.Value.Call({0x217f300?, 0x400122c4c0?, 0x0?}, {0x585d1c0?, 0x10?, 0x40008bcd70?})
Mar 28 11:45:37 raspberrypi evcc[110759]: 	reflect/value.go:365 +0x94
Mar 28 11:45:37 raspberrypi evcc[110759]: github.com/asaskevich/EventBus.(*EventBus).doPublish(0x2351400?, 0x400124d320, {0x2861a6e?, 0x400122c4d0?}, {0x0?, 0x1db9ff8?, 0x40012cd888?})
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/asaskevich/EventBus@v0.0.0-20200907212545-49d423059eef/event_bus.go:160 +0x60
Mar 28 11:45:37 raspberrypi evcc[110759]: github.com/asaskevich/EventBus.(*EventBus).Publish(0x4000c529e0, {0x2861a6e, 0x7}, {0x0, 0x0, 0x0})
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/asaskevich/EventBus@v0.0.0-20200907212545-49d423059eef/event_bus.go:144 +0x588
Mar 28 11:45:37 raspberrypi evcc[110759]: github.com/evcc-io/evcc/core.(*Loadpoint).updateChargerStatus(0x4000cfc908)
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/evcc-io/evcc/core/loadpoint.go:1065 +0x484
Mar 28 11:45:37 raspberrypi evcc[110759]: github.com/evcc-io/evcc/core.(*Loadpoint).Update(0x4000cfc908, 0xc0ad2c5e353f7cee, 0x0, {0x4000f56008, 0xac, 0x124}, 0x0, 0x0, 0x3ff0000000000000, 0x4000ad0f50, ...)
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/evcc-io/evcc/core/loadpoint.go:1775 +0x2c8
Mar 28 11:45:37 raspberrypi evcc[110759]: github.com/evcc-io/evcc/core.(*Site).update(0x40000e7208, {0x3230748, 0x4000cfc908})
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/evcc-io/evcc/core/site.go:924 +0x7a4
Mar 28 11:45:37 raspberrypi evcc[110759]: github.com/evcc-io/evcc/core.(*Site).Run(0x40000e7208, 0x4000fcf490, 0x6fc23ac00)
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/evcc-io/evcc/core/site.go:1043 +0x1a4
Mar 28 11:45:37 raspberrypi evcc[110759]: github.com/evcc-io/evcc/cmd.runRoot.func4()
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/evcc-io/evcc/cmd/root.go:325 +0x2c
Mar 28 11:45:37 raspberrypi evcc[110759]: created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
Mar 28 11:45:37 raspberrypi evcc[110759]: 	github.com/evcc-io/evcc/cmd/root.go:324 +0x1bf4
Mar 28 11:45:37 raspberrypi systemd[1]: evcc.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Mar 28 11:45:37 raspberrypi systemd[1]: evcc.service: Failed with result 'exit-code'.
^C
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-03-28:
> I have verified that the issue is reproducible with the latest nightly build

Really?

closed 2025-03-28 by andig

narcismatei, 2025-03-28:
> > I have verified that the issue is reproducible with the latest nightly build
> 
> Really?

Sorry about this, I assumed that since the last version was released yesterday there will be no fix so quickly.
Anyway, thanks for the fix.