evcc-io/evcc #20225: Vehicle pre heating

enboer, 2025-03-28:
**Is your feature request related to a problem? Please describe.**
Currently I cannot enable the car heater from the EVCC app/webpage. That would be nice. Another option would be if I enable pre heating in the (Renault app) it is detected by EVCC. If it does, enable it to charge. This helps not to take power from the main car battery.

**Describe the solution you'd like**
Make it possible from EVCC to start heating the car, and charging then as well.

**Describe alternatives you've considered**
If heating is set from the car app, EVCC should start to charge.

**Additional context**
Add any other context or screenshots about the feature request here.

StefanSchoof, 2025-03-28:
Similar issue https://github.com/evcc-io/evcc/issues/18622

andig, 2025-03-28:
> If heating is set from the car app, EVCC should start to charge.

We already do this where possible. Note https://github.com/evcc-io/evcc/issues/19690 for another approach.

closed 2025-03-28 by andig