evcc-io/evcc #20233: Sungrow fehler bei Batterie Entsperren

comessoon, 2025-03-29:
### Describe the bug

Hallo seit dem Update von Version 201.1 auf Version 202.1 kommt es zum unten genauen Fehler wenn EVCC meinen Sungrow Batterie zu entsperren. Das Sperren glápt noch Fehler frei so wie es soll. Wenn aber die Batterie entsperrt werden soll kommt dieser Fehler und die Batterie bleibt gesperrt

### Steps to reproduce

1. update von 201.1 auf 2002.1 
2. 

### Configuration details

```yaml
Konfiguration wurde in GUI Durchgeführt
```

### Log details

```text
site ] ERROR 2025/03/29 09:10:32 battery mode: modbus: exception '4' (server device failure), function '6'
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

202.2

andig, 2025-03-29:
Passiert das auch mit

    evcc meter

?

comessoon, 2025-03-29:
Das ist die einzige Fehlermeldung dich im Log hatte und auch sonst ist mir kein Fehler aufgefallen. Bin jetzt erstmal zurück zur 201.1 kann aber jeder Zeit zum testen zurück zur aktuellen Version.

closed 2025-03-29 by andig

comessoon, 2025-03-29:
Hallo @andig warum ist der Fall jetzt abgeschlossen?

andig, 2025-03-29:
Mangels Antwort

comessoon, 2025-03-29:
Hallo @andig dann hab ich einfach deine Frage nicht richtig verstanden was meinst du denn. Mit: passiert das auch mit evcc Meter?

comessoon, 2025-03-29:
/app # evcc meter
[main  ] INFO 2025/03/29 12:53:52 evcc 0.202.1
[main  ] INFO 2025/03/29 12:53:52 using config file: /etc/evcc.yaml
[db    ] INFO 2025/03/29 12:53:52 using sqlite database: /root/.evcc/evcc.db
db:8
----
Power:        4244W
Energy:       24433.0kWh
Max AC power: 10000W

db:9
----
Power:          -5W
Energy:         18383.7kWh
Current L1..L3: -5.32A 10.6A -5.26A

db:10
-----
Power:        -142W
Energy:       4798.1kWh
Soc:          43%
Capacity:     9.6kWh
Controllable: true

andig, 2025-03-29:
> Wenn aber die Batterie entsperrt werden soll kommt dieser Fehler und die Batterie bleibt gesperrt

Du musst bitte genau das ausprobieren, nicht irgendwas ;) Siehe Optionen.

comessoon, 2025-03-29:
@andig ich muss ja zugeben das ich die Parameter ja alle kenne, kannst du nächste mal vielleicht kurz einen Tipp mit geben wo ich die finden und wie sparen uns beide etwas schreib Arbeit und währen jetzt schon weiter.

/app # evcc meter --battery-mode hold
[main  ] INFO 2025/03/29 14:32:42 evcc 0.202.1
[main  ] INFO 2025/03/29 14:32:42 using config file: /etc/evcc.yaml
[db    ] INFO 2025/03/29 14:32:42 using sqlite database: /root/.evcc/evcc.db
/app # evcc meter --battery-mode normal
[main  ] INFO 2025/03/29 14:32:46 evcc 0.202.1
[main  ] INFO 2025/03/29 14:32:46 using config file: /etc/evcc.yaml
[db    ] INFO 2025/03/29 14:32:46 using sqlite database: /root/.evcc/evcc.db
[main  ] FATAL 2025/03/29 14:32:46 set battery mode: modbus: exception '4' (server device failure), function '6'
/app #

andig, 2025-03-29:
> ich muss ja zugeben das ich die Parameter ja alle kenne

Wie bei jeder anderen Anwendung auch findest Du die mit `--help`. Es hat ja geklappt :) Mach mal bitte

    evcc meter --battery-mode normal --log trace,db:error

mit aktueller und letzter funktionierender Version. Mir ist noch nicht richtig klar was sich da geändert haben sollte.

comessoon, 2025-03-29:
Hallo @andig hier die beider Ergebnisse:

Version 201.1 Fehler frei:

evcc meter --battery-mode normal --log trace,db:error
[main  ] INFO 2025/03/29 18:06:56 evcc 0.201.1
[main  ] INFO 2025/03/29 18:06:56 using config file: /etc/evcc.yaml
[db:10 ] TRACE 2025/03/29 18:06:57 modbus: send 00 01 00 00 00 06 01 06 32 f9 00 00
[db:10 ] TRACE 2025/03/29 18:06:57 modbus: recv 00 01 00 00 00 06 01 06 32 f9 00 00
[db:10 ] TRACE 2025/03/29 18:06:57 modbus: send 00 02 00 00 00 06 01 06 32 fa 00 cc
[db:10 ] TRACE 2025/03/29 18:06:57 modbus: recv 00 02 00 00 00 06 01 06 32 fa 00 cc
[db:10 ] TRACE 2025/03/29 18:06:57 modbus: send 00 03 00 00 00 06 01 06 81 17 04 24
[db:10 ] TRACE 2025/03/29 18:06:57 modbus: recv 00 03 00 00 00 06 01 06 81 17 04 24

Version 202.1 Fehlerhaft:

/app # evcc meter --battery-mode normal --log trace,db:error
[main  ] INFO 2025/03/29 18:10:13 evcc 0.202.1
[main  ] INFO 2025/03/29 18:10:13 using config file: /etc/evcc.yaml
[db:10 ] TRACE 2025/03/29 18:10:14 modbus: send 00 01 00 00 00 06 01 06 32 f9 00 00
[db:10 ] TRACE 2025/03/29 18:10:14 modbus: recv 00 01 00 00 00 06 01 06 32 f9 00 00
[db:10 ] TRACE 2025/03/29 18:10:14 modbus: send 00 02 00 00 00 06 01 06 32 fa 00 cc
[db:10 ] TRACE 2025/03/29 18:10:14 modbus: recv 00 02 00 00 00 06 01 06 32 fa 00 cc
[db:10 ] TRACE 2025/03/29 18:10:14 modbus: send 00 03 00 00 00 06 01 06 81 17 29 68
[db:10 ] TRACE 2025/03/29 18:10:15 modbus: recv 00 03 00 00 00 03 01 86 04
[main  ] FATAL 2025/03/29 18:10:15 set battery mode: modbus: exception '4' (server device failure), function '6'

mfuchs1984, 2025-03-30:
Vorher: maxchargepower 1060
`[db:10 ] TRACE 2025/03/29 18:06:57 modbus: send 00 03 00 00 00 06 01 06 81 17 04 24`

Nachher: maxchargepower 10600
`[db:10 ] TRACE 2025/03/29 18:10:14 modbus: send 00 03 00 00 00 06 01 06 81 17 29 68`

Wie hast du maxchargepower konfiguriert?

mfuchs1984, 2025-03-30:
1060 entspricht wohl 10,6 kW bzw. 10600 W, nicht gerade intuitiv ;)
https://github.com/evcc-io/evcc/pull/19882/files

comessoon, 2025-03-30:
Hallo @mfuchs1984 

Einmal ein Screenshot aus den Einstellungen  denke aber, wenn ich darüber nach  denke das der Wert zu hoch für den Akku ist 

[image]

mfuchs1984, 2025-03-30:
Wenn deine Batterie 10600 kann, trage dort mal 1060 ein. Wenn sie weniger kann, entsprechen die Watt / 10.

comessoon, 2025-03-30:
@mfuchs1984 ja bin da gerade schon  am testen mit 1060 läuft es auf jeden Fall schon mal wieder.

mfuchs1984, 2025-03-30:
@andig @naltatis hier muss irgendwo zwischen Benutzeingabe und Modbus Nachricht noch ein scale Faktor von 0.1 rein.

andig, 2025-03-30:
Closed in f713679eebd8234e13e4f0e6ed67580b5826ea29. Unfortunately that change was never tested :(

mfuchs1984, 2025-03-30:
@comessoon mit dem nächsten Release musst du dann wieder 10600 eintragen oder den Wert einfach entfernen.

comessoon, 2025-03-30:
Hi @mfuchs1984 , danke für den Hinweis, damit hatte ich auch schon gerechnet