evcc-io/evcc #20238: Bitte Lambda Support um "Negative Energieüberschussmessung" erweitern

anbie, 2025-03-29:
Liebes evcc-Team,

erstmal ein großes Lob für eure tolle Arbeit an evcc! Die Integration von Wärmepumpen, besonders die Unterstützung für Lambda Wärmepumpen, ist echt klasse.

Ich habe meine Wärmepumpe auf “Negativen Energieüberschuss” einstellen lassen. Allerdings kann ich das nicht selber umstellen und muss für die Einstellung Lambda oder meinen Heizungsbauer kontaktieren. Es wäre super, wenn evcc auch direkt "Negative Energieüberschussmessung" unterstützen würde.
Vielen Dank schon mal im Voraus und macht weiter so!

Aus der Lambda Modbus-Spec:
"Fall 2: Negative Energieüberschussmessung: Der Energiezähler misst die Energiemenge in der Netzzuleitung das Gebäudes.  Die Wärmepumpe versucht die Überschüssige Energie (PV Ertrag abzüglich aller eletrischen Verbraucher im Gebäude), welche ins Netz eingespeißt werden muss, zu verbrauchen.  Negativ bedeuted hier, dass bei Messung eines Überschusses (Strom wird ins Netz eingespeißt) ein negativer Wert angezeigt wird. "

[image]

andig, 2025-03-29:
Das braucht einen Parameter im Template. Gerne PR.

closed 2025-04-10 by github-actions[bot]

anbie, 2025-06-20:
Hi zusammen,

ich würde das Issue gerne wieder öffnen – habe endlich etwas Zeit gefunden und will mich an einer Implementierung und PR machen.

Ich bin neu in der EVCC-Entwicklung. Habe aber schon lauffähigen Bau- und Testsetup aufgesetzt mit meiner Lambda und angefangen, einen neuen Parameter im Template umzusetzen.

Hier mein bisheriger Ansatz für lambda-zewotherm.yaml mittels scale:
```
params:
  - name: host
  - name: tempsource
    type: choice
    choice: ["warmwater_top", "warmwater_bottom", "buffer_top", "buffer_bottom"]
  - name: excess
    type: choice
    choice: ["positive", "negative"]
    default: "positive"
    description:
      de: E-Überschuss("positive" oder "negative")
      en: Excess("positive" or "negative")  
  - name: phases
    deprecated: true
  - name: watchdog
    type: duration
    default: 60s
    advanced: true 
render: |
  type: heatpump
  setmaxpower:
    source: watchdog
    timeout: {{ .watchdog }} # re-write at timeout/2
    initial: 0
    set:
      source: modbus
      uri: {{ .host }}:9001
      id: 1
      register:
        address: 102 # PV Überschussleistung
        type: writemultiple # λ erwartet single value als FC16
        decode: int16
      scale: {{ if eq .excess "positive" }} 1 {{ else }} -1 {{ end }}
```

Das Problem: Ich bekomme es nicht hin, der Wärmepumpe einen negativen Wert zu übermitteln. Wenn ich testweise "scale: 2" (anstatt "-1") setze, sehe ich in der Lambda, wie sich der Wert verdoppelt – das klappt. Bei "scale: -1" wird im Energie-Management-Display der Lambda aber immer nur 0.00 angezeigt.

Hat jemand eine Idee, woran das liegen könnte? Bin für jeden Hinweis oder Vorschlag dankbar!

Viele Grüße
Andreas

andig, 2025-06-20:
Logfile statt raten wäre ein guter Ansatz.

anbie, 2025-06-20:
Mein Ansatz für die Erweiterung:

```
params:
  - name: host
  - name: tempsource
    type: choice
    choice: ["warmwater_top", "warmwater_bottom", "buffer_top", "buffer_bottom"]
  - name: excess
    type: choice
    choice: ["positive", "negative"]
    default: "positive"
    description:
      de: E-Überschuss("positive" oder "negative")
      en: Excess("positive" or "negative")  
  - name: phases
    deprecated: true
  - name: watchdog
    type: duration
    default: 60s
    advanced: true 
render: |
  type: heatpump
  setmaxpower:
    source: watchdog
    timeout: {{ .watchdog }}   # re-write at timeout/2
    initial: 0
    set:
      source: modbus
      uri: {{ .host }}:9001
      id: 1
      register:
        address: 102            # PV Überschussleistung
        type: writemultiple     # λ erwartet single value als FC16
        decode: uint16 
      scale: {{ if eq .excess "positive" }}1{{ else }}-1{{ end }}
```

Damit sehe ich im log-file bei einem Lambda Charging Point mit  "positive" angelegt diese modbus requests für Register 102 (laut der Lambda-Spec soll in Reg 102 (0x66) ein  INT16 mit dem "actual excess power" in Watt stehen). In diesem Fall ist dann Data 0x2b20 = 11040 W. Das entspricht meiner Erwartung.
```
[db:1  ] TRACE 2025/06/20 23:04:44 modbus: send 00 18 00 00 00 09 01 10 00 66 00 01 02 2b 20
[db:1  ] TRACE 2025/06/20 23:04:44 modbus: recv 00 18 00 00 00 06 01 10 00 66 00 01 
```

Wenn ich nun die Lambda mit "negative" anlege, dann sehe ich folgende Requests im Logfile:
```
[db:1 ] TRACE 2025/06/20 23:57:49 modbus: send 01 3e 00 00 00 09 01 10 00 66 00 01 02 00 00
[db:1 ] TRACE 2025/06/20 23:57:49 modbus: recv 01 3e 00 00 00 06 01 10 00 66 00 01
```
Gleicher Request, nur Data ist 0x0000 = 0W. Erwartet hätte ich in etwa -11040W.

Es scheint, als würde irgendwo im Kontext des WriteMultipleRegister-Call im modbus-client die Daten auf 0x0000 gesetzt zu werden. 
Dazu 2 Fragen:
1. Gibt es andere Ansätze die ich versuchen kann?
2. Kennt sich jemand in der modbus-client Ecke aus, der helfen könnte dem Problem auf die Schliche zu kommen?

(sagt Bescheid, falls ich weitere oder andere Debug-Daten erzeugen soll).

andig, 2025-06-21:
Gehts denn mit writesingle?

anbie, 2025-06-21:
Wenn ich writemultiple mit writesingle ersetze, dann "failed" das validate schon beim Anlegen der Lambda:
```
cannot create charger type 'template': cannot create charger type 'heatpump': modbus: exception '1' (illegal function), function '6'
```

Aus dem Log:
```
[httpd ] TRACE 2025/06/21 09:36:48 POST /api/config/test/charger
[modbus] TRACE 2025/06/21 09:36:48 modbus: send 00 01 00 00 00 06 01 06 00 66 00 00
[modbus] TRACE 2025/06/21 09:36:48 modbus: recv 00 01 00 00 00 03 01 86 01
```

Scheint so, als wäre writesingle in diesem Kontext nicht erlaubt. Bin aber noch nicht tief genug vorgedrungen um zu sagen, ob das eine Limitation des clients oder der Lambda ist.

andig, 2025-06-21:
Das kann nicht sein. Beim anlegen wird noch kein Strom gesetzt. An der falschen Stelle geändert?

anbie, 2025-06-21:
🤷🏻‍♂️ - zur Sicherheit habe ich nochmal gecheckt. Die Änderungen ist nur in setmaxpower. 
Den ersten Modbus-Request, den ich sehe beim Anlegen einer Lambda ist ein Schreib-Versuch ins Register 102 (0x66) beim Validate:

Hier der Log:
```
[httpd ] TRACE 2025/06/21 10:10:41 GET /api/config/products/charger
[httpd ] TRACE 2025/06/21 10:10:48 GET /api/config/templates/charger
[httpd ] TRACE 2025/06/21 10:10:59 POST /api/config/test/charger
[modbus] TRACE 2025/06/21 10:10:59 modbus: send 00 01 00 00 00 06 01 06 00 66 00 00
[modbus] TRACE 2025/06/21 10:10:59 modbus: recv 00 01 00 00 00 03 01 86 01
```
[evcc-20250621-101147-trace.log](https://github.com/user-attachments/files/20845735/evcc-20250621-101147-trace.log)

Hier wieder mit writemultiple (sonst keine weiteren Änderungen in lambda-zewotherm.yaml)
```
[httpd` ] TRACE 2025/06/21 10:18:11 GET /api/config/products/charger
[httpd ] TRACE 2025/06/21 10:18:16 GET /api/config/templates/charger
[httpd ] TRACE 2025/06/21 10:18:23 POST /api/config/test/charger
[modbus] TRACE 2025/06/21 10:18:23 modbus: send 00 01 00 00 00 09 01 10 00 66 00 01 02 00 00
[modbus] TRACE 2025/06/21 10:18:23 modbus: recv 00 01 00 00 00 06 01 10 00 66 00 01
[modbus] TRACE 2025/06/21 10:18:23 modbus: send 00 02 00 00 00 06 01 03 00 67 00 01
[modbus] TRACE 2025/06/21 10:18:24 modbus: recv 00 02 00 00 00 05 01 03 02 00 00
[modbus] TRACE 2025/06/21 10:18:24 modbus: send 00 03 00 00 00 06 01 03 03 fc 00 02
[modbus] TRACE 2025/06/21 10:18:24 modbus: recv 00 03 00 00 00 07 01 03 04 00 05 80 30
[modbus] TRACE 2025/06/21 10:18:24 modbus: send 00 04 00 00 00 06 01 03 07 d2 00 01
[modbus] TRACE 2025/06/21 10:18:24 modbus: recv 00 04 00 00 00 05 01 03 02 01 fb
[modbus] TRACE 2025/06/21 10:18:24 modbus: send 00 05 00 00 00 06 01 03 08 02 00 01
[modbus] TRACE 2025/06/21 10:18:24 modbus: recv 00 05 00 00 00 05 01 03 02 02 17
[httpd ] TRACE 2025/06/21 10:18:28 POST /api/config/devices/charger
[modbus] TRACE 2025/06/21 10:18:28 modbus: send 00 06 00 00 00 09 01 10 00 66 00 01 02 00 00
[modbus] TRACE 2025/06/21 10:18:28 modbus: recv 00 06 00 00 00 06 01 10 00 66 00 01
[db    ] TRACE 2025/06/21 10:18:28 INSERT INTO `configs` (`class`,`type`,`title`,`icon`,`product`,`value`) VALUES (1,"template","","","Lambda EU-L Series","{""excess"":""positive"",""host"":""homebridge"",""template"":""lambda-zewotherm"",""tempsource"":""warmwater_top"",""watchdog"":""60s""}") RETURNING `id` 1 <nil>
```
[evcc-20250621-101938-trace.log](https://github.com/user-attachments/files/20845736/evcc-20250621-101938-trace.log)
Auch hier sehe ich als ersten Modbus-Request ein schrieben auf Register 102 (mit Daten 0x0000).

andig, 2025-06-21:
> Hier der Log:

Ich sehe keinen Fehler.

andig, 2025-06-21:
> Ich auch nicht. Aber im UI sehe ich das:

Mach bitte 

    evcc charger -i 6 --log trace

Es hilft gar nichts, immer in den ganzen Wald zu schauen wenn man nur einen Pilz sucht ;)

anbie, 2025-06-21:
Gute Idee:

```
❯ ./evcc charger -i 6 --log trace
[main  ] INFO 2025/06/21 12:16:41 evcc empty (c5596faa)
[main  ] INFO 2025/06/21 12:16:41 using config file: /Users/anbie/evcc/evcc/evcc.yaml
[db    ] INFO 2025/06/21 12:16:41 using sqlite database: /Users/anbie/.evcc/evcc.db
[db    ] TRACE 2025/06/21 12:16:41 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/06/21 12:16:41 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/06/21 12:16:41 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/06/21 12:16:41 SELECT * FROM `settings` 0 <nil>
[db    ] TRACE 2025/06/21 12:16:41 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2025/06/21 12:16:41 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/06/21 12:16:41 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/06/21 12:16:41 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
```

andig, 2025-06-21:
Ich sehe da nix. V.a. kein Modbus.

anbie, 2025-06-21:
sorry 🙈 - vermutlich zu heiß heute🙃

Hier nochmal. Charger ist angelegt mit writemultiple und negative

```
❯ ./evcc charger -i 6 --log trace
[main  ] INFO 2025/06/21 13:58:33 evcc empty (c5596faa)
[main  ] INFO 2025/06/21 13:58:33 using config file: /Users/anbie/evcc/evcc/evcc.yaml
[db    ] INFO 2025/06/21 13:58:33 using sqlite database: /Users/anbie/.evcc/evcc.db
[db    ] TRACE 2025/06/21 13:58:33 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/06/21 13:58:33 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/06/21 13:58:33 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/06/21 13:58:33 SELECT * FROM `settings` 2 <nil>
[db    ] TRACE 2025/06/21 13:58:33 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2025/06/21 13:58:33 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/06/21 13:58:33 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/06/21 13:58:33 SELECT * FROM `configs` WHERE `configs`.`class` = 1 1 <nil>
[db:1  ] TRACE 2025/06/21 13:58:33 modbus: send 00 01 00 00 00 09 01 10 00 66 00 01 02 00 00
[db:1  ] TRACE 2025/06/21 13:58:33 modbus: recv 00 01 00 00 00 06 01 10 00 66 00 01
[db:1  ] TRACE 2025/06/21 13:58:33 modbus: send 00 02 00 00 00 09 01 10 00 66 00 01 02 00 00
[db:1  ] TRACE 2025/06/21 13:58:33 modbus: recv 00 02 00 00 00 06 01 10 00 66 00 01
```