evcc-io/evcc #20241: Huawei EMMA - Support External Meter for cases in which not all loads are connected to EMMA

mrandt, 2025-03-29:
**Is your feature request related to a problem? Please describe.**

For energy measuring point _meters: usage: grid_ I'd like to be able to configure the Huawei EMMA meter integration / template so it uses measurements from an external meter connected to EMMA via Modbus RTU rather than the built-in energy flow sensors.

Huawei EMMA includes energy flow sensors and in many cases will be the only meter required in a Huawei PV setup.

But EMMA also supports connecting an external meter (like a DTSU666 or YDS60-80) via Modbus RTU on the secondary RS485 COMM port.

An external meter might be necessary, if not all loads are connected behind EMMA - either because the loads require too much current in total or the metering concept requires an energy flow measurement at a different location.

I have recently installed the brand new Huawei SmartGuard backup power solution, which always includes EMMA.

In preparation for a cascaded metering solution ("_Messkonzept mit Kasakadenmessung zur separaten Messung der Energie für SteuVE in Deutschland_") not all loads could be connected behind EMMA but the EV charger and at some point in the future a heatpump are connected in front of the SmartGuard and thus EMMA.

Maybe a picture explains it better than many words:

[image]

I hope it is obvious from the picture, that EMMA's internal energy flow sensors (pink) will only capture energy flow from and to household + PV. For surplus charging, this measurement is totally off. It will actually assume the energy goint to the wallbox was exported to grid...

The correct measurement needed is the data from the YDS60-80 (blue) which measures the overall energy flow between household and public grid.

Within EMMA, I could configure the external meter to be used for battery charge / discharge control and statistics in Fusion portal and app.

**Describe the solution you'd like**

I'd like an optional configuration parameter for meters with template huawei-emma (https://docs.evcc.io/en/docs/devices/meters#emma)  which switches to use the external meter rather than the internal one.

I do assume that EMMA provides those values via some Modbus register, even though I don't know which one.

However, if you let me know I can best help analyze, develop and test this, I am absoultely willing to help.

Thank you in advance for considering this!

**Describe alternatives you've considered**

In theory, we could add a third-party meter (unrelated to Huawei) next to the external meter (directly behind the blue one in my drawing) and use that device's data. But it seems a bit cumbersome to add yet another meter if - in theory - all the data should already be available.

**Additional context**

Please find a description of how and when to use an external meter in the Huawei Smart Guard manual here:
https://support.huawei.com/enterprise/de/doc/EDOC1100366781/26745df?idPath=258788303|258788489|258789981|21102413|255740899

mrandt, 2025-03-29:
@Mungg1818 I saw that you have contributed the huawei-emma template. Could you please have a look at this issue? Thanks!

mrandt, 2025-03-29:
More digging; seems that template huawei-emma reads register `31679` via Modbus for grid energy:

https://github.com/evcc-io/evcc/blob/e93e42003745743af9ce58accd9de539609bba90/templates/definition/meter/huawei-emma.yaml#L30

If I understand correctly, in my scenario the register `31935` would be needed instead - see screenshot from Huawei docs:

[image]

So basically an option which switches between 316979 (internal energy meter) and 31935 (external energy meter) would to the trick?!

premultiply, 2025-03-30:
> I hope it is obvious from the picture, that EMMA's internal energy flow sensors (pink) will only capture energy flow from and to household + PV. For surplus charging, this measurement is totally off. It will actually assume the energy goint to the wallbox was exported to grid...

> The correct measurement needed is the data from the YDS60-80 (blue) which measures the overall energy flow between household and public grid.

The position of "EMMA" and "EnFuRi" needs to be swapped.
Somebody misunderstood the concept.
Z1 is the grid meter. Z2 is **only** for billing reasons to divide between consumption of the "SteuVE" and all other loads.