evcc-io/evcc #20243: Netwerkverbindung

hrampf, 2025-03-29:
### Describe the bug

Es tritt ein Lesefehler grid auf

### Steps to reproduce

Fehler tritt immer auf

### Configuration details

```yaml
network:
  port: 7070

log: debug

interval: 30s

meters:

  - name: sofar-battery-1
    type: template
    template: sofarsolar-g3
    usage: battery
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.87 # Hostname
    port: 8899 # Port  
    delay: '0s'
    capacity: 12
    minsoc: 10
    maxsoc: 100

  - name: sofar-grid
    type: template
    template: sofarsolar-g3
    usage: grid
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.87 # Hostname
    port: 8899 # Port  
    delay: '0s'

  - name: sofar-pv
    type: template
    template: sofarsolar-g3
    usage: pv
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.87 # Hostname
    port: 8899 # Port  
    delay: '0s'

  - name: shelly
    type: template
    template: shelly-3em # Shelly 3EM als Grid-Meter
    usage: grid
    host: 192.168.178.89

chargers:
  type: template
  template: elvi
  stationid: EVB-P22217851
  connector: 1
  connecttimeout: 5m
  timeout: 2m
  meter: false
  meterinterval: 4s
  name: wallbox1

vehicles:
  type: template
  template: tesla
  title: Tesla
  icon: car
  accessToken: 
  refreshToken: 
  capacity: 70
  phases: 3
  mode: minpv
  minCurrent: 5
  maxCurrent: 16
  priority: 1
  name: ev1
  #onIdentify:
  #  mode: pv
  #  minSoC: 1
  #  targetSoC: 90
  #cloud: false  

site:
  title: Zuhause
  meters:
    grid: sofar-grid
    pv: sofar-pv
    battery: sofar-battery-1
  #prioritySoC: 90 # give home battery priority up to this soc (0 to disable)
  #bufferSoC: 80
  residualPower: 100

loadpoints:
  - title: Carport
    charger: wallbox1
    mode: pv
    phases: 3
    #meter: sofar-grid
    meter: shelly
    vehicle: ev1

tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    type: fixed
    price: 0.30 # EUR/kWh
  feedin:
    type: fixed
    price: 0.082 # EUR/kWh
  co2:
    type: grünstromindex
    zip: 83661

influx:
  #url: http://localhost:8086
  url: http://influxdb:8086
  database: evccdb
  user: 
  password:
```

### Log details

```text
grid power: read failed: read tcp 172.21.0.3:60524->192.168.178.87:8899: i/o timeout
vor 3 Minuten

grid power: read failed: read tcp 172.21.0.3:58400->192.168.178.87:8899: i/o timeout
vor 3 Minuten

battery 1 power: read failed: read tcp 172.21.0.3:41510->192.168.178.87:8899: i/o timeout
vor 3 Minuten

pv 1 power: add[0]: read failed: read tcp 172.21.0.3:36550->192.168.178.87:8899: i/o timeout
vor 4 Minuten

grid power: read failed: read tcp 172.21.0.3:37024->192.168.178.87:8899: i/o timeout
vor 6 Minuten

grid power: read failed: read tcp 172.21.0.3:42172->192.168.178.87:8899: i/o timeout
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

Installierte Version: 0.201.1

Chris45671, 2025-03-29:
Es handelt sich dabei ja um ganz andere IP Adressen in einem anderen Bereich? 
Läuft das in einem Docker?
Können die IP Adressen von dem Rechner angepingt werden? 
Sieht nach einem reinen Netzwerkproblem aus.

andig, 2025-03-30:
Ist der Fehler neu oder ging das noch nie mit einer alten Version?