evcc-io/evcc #20244: Neues Auto und der Clou Login funktioniert bei BMW nicht mehr

AniGerm, 2025-03-30:
### Describe the bug

Hallo,
ich habe aus ehemals 1x BMW und 1x Audi jetzt 2x BMW gemacht. Vorher ging alles einwandfrei, ich hatte die hcaptcha eingerichtet und alles war gut. Nun habe ich den Audi aus der yaml gelöscht und einen zweiten BMW hinzugefügt und nun kann ich machen was ich will, ich bekomme über die .yaml keine Verbindung mit der Cloud. Wenn ich das Fahrzeug über UI hinzufügen will, sehe ich bei „Prüfen“, dass alles funktioniert, ich kann es jedoch nicht speichern, da ich nicht in die Database speichern könne 😞
Am liebste wäre es mir wie vorher: 2 Fahrzeuge in der yaml und gut ist es. Vielleicht hat noch jemand eine Idee! Vielen Dank 👍🏼

### Steps to reproduce

1. sobald das Fahrzeug in der .yaml steht
2. auch verschiede, nur eins oder zwei verschiedene neue hcaptchas machen keinen Unterschied. 

### Configuration details

```yaml
open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 30s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: huawei-sun2000 
  id: 1  
  host: 192.168.1.139  
  port: 502  
  usage: grid  
  modbus: tcpip  
  timeout: 15s  
  name: grid1
- type: template
  template: huawei-sun2000 
  id: 1  
  host: 192.168.1.139  
  port: 502  
  usage: pv  
  modbus: tcpip  
  timeout: 15s  
  name: pv2
- type: template
  template: huawei-sun2000 
  id: 1  
  host: 192.168.1.139  
  port: 502  
  usage: battery  
  modbus: tcpip  
  timeout: 15s  
  name: battery3

chargers:
- type: template
  template: abl 
  id: 1  
  host: 192.168.1.194  
  port: 502  
  modbus: rs485tcpip  
  timeout: 10s  
  name: wallbox6

vehicles:
- type: template
  template: bmw 
  user: *****
  password: *****
  vin: *****
  title: i5  
  capacity: 81.2  
  hcaptcha: P1_eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJwYXNza2V5IjoiRE1rRjh0ekVHU2ZHOWh0Y3VOa1RSNkFPN2ttc3lGNzhGUVpDMzBVZm9USG9ha1pHUytSNTYvWWJyNUFJWWdYYmZRM3l4MWhzaStJZVVmN0VsTnhQNVlSVzhXWmcxeDdTV3NIdmV5M3A0V2h6WjJwOGcrZEllWTVKckNod2YrUndkUTNUcEo1bkVFdVR4Z01vd0YvVG9DZitwN3d2dW9HZVg1OE5FN01vanIwUis5NnpHNzVtOE5OUzFtdlZreGpaUTVZSThraHZKSE9zWWkxVTRHcDMweTlmcEtJR2lHVmJZdk40SFI5U2dYY0M1VTgwVXl5aUhYU2JwUkwzUmJDdzh6cHZjRVoraHAxR05TYWRNV0ZhNjFhMzJ0R2w2R1k0VUJNOTBPc3pUZ09CSVFKa0Zwa2llZ3RUMCtqR2Z1NUFjQ3BHcDlJY0pPVXBxYW9hcEJiMlloL2s3UTNDbnZYRE85UUpScThuL3ZpRDJyT21GYW1aN2dKUHVLTkVCZzhRbUZDNktsRVNBMytOenBWbEVNVlhHYjlhZHZZVWRHeFpMM0pTL2gvazBnUXN2WGxzNkRpb3grT1VrRlRRWlg1dy96aS9Hd1ZUcTFqYnEyR0gvRCtSazZNelhzQm1QY2NycjMyamx4V1dxUjJLaGJtczBJK1VHWldPd2Zhelc1Y1BzT1hjNnlVMVJWRW5pNi9nVnN2MFR1dGFQZWZ5RnhpZm9Mb0pPYVNRenRyckZuVEtaOXdGcTB3Qys4Z3VlUnhSdThKdzMycEFrQUlIdWhoM1U3YUJ0aVZzVmN4UFBMZ2d2N0creW9MdnZSdEgxSVJKU0RIRDN1bXFTdHBSL0d0NThzejJxMnBVaWNjVVdxYnV5ZGdIYUZONWZYVXlyR2swdWkwM0pTQjQvUFRQREJyejdqbDVJSmtvVkU3VW1Qa2RXS1pxdGFTb01NbUErM3psWlUzK1JFQ295d3g0WFY3OWM4OGYzTEFGdmRsVEp1YmhzSDlWOVYxcjFQa0xMWElaNDBCRzl6Z2xJUXVxRFdYM1A5VTVnTjlCZ1BiNXFtVWlNRjVVd0RQZldPWDVNaHk3dGUzRGtsRnozbDZlQWhuc2hnT3FkWTVGbnhJV1o3cUc1aTU2a2pPeHdvS25ud3dRK3Q3aFFSemRERmRYMWRETWhDS0FtbWUrVGNnWDR0YnhpaGpsWmZGRkZPWXY0bEp6dUt6Rjk5TVN3K0gwclk1VTBwQlZOd1hIeTJ4eklneUtkVldTbU9VQzNnZ3VmWitwS1A5SWNuNU1HWjF3UHM4ZUxKMVVSRjNlS1NTbWhhZWhkWVlsaVUxRkViaVZsSEJoVTY0N1ZydFd1bVhXNUNQVTNudWZSakZ0M25XUTE3T3BrQWcwTTRRRnZFUlFVMzdLb0dtVUkrc3NEdlMrU3BvbG04Qng3T3JxSEdaUGhxQjV2NURDQS9VUFgxcmhwQ01xNGlNaWdYcVE3Rmdha2owQjE1bXZaVzViUDZXNXdraGdsWm1SRHhCN0R2czhXRWxXT1VDNW1MeWsvRnY4NDFOQTNpemYxMEVHSkRscjAxUm1UL2czRnQvL2orOWpWYTFZSXNjUFU4MXlnc01EZEMvb3ZWdEIySEdCdEJ6Z3BDWnpYYlY1dXpyeSttaVhhTEJCUmRxS1oySVRWSzVqTXhYOU9pQ1V1Mi9WbWx1eXBWanN3cHc1YzhWUGp5eUlydjRKMlloUDErZnNuVkVvMC9VbEp2UVE0N0lqZnRxQm9ZbG1ENnh0cG5ieWwyVFJ3Nm1LMyt0OW9RPT0iLCJleHAiOjE3NDMyOTM0ODAsInNoYXJkX2lkIjozMzk1MTAzMDMsImtyIjoiNjNmN2ZhZCIsInBkIjowfQ.wELKCOw50WZbXRkdPQcoNL2rWgGd_KfEA3mXpT6nLhA  
  name: ev4
- type: template
  template: bmw 
  user: *****
  password: *****
  vin: *****
  title: iX1  
  capacity: 64.8  
  hcaptcha: P1_eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJwYXNza2V5IjoiRE1rRjh0ekVHU2ZHOWh0Y3VOa1RSNkFPN2ttc3lGNzhGUVpDMzBVZm9USG9ha1pHUytSNTYvWWJyNUFJWWdYYmZRM3l4MWhzaStJZVVmN0VsTnhQNVlSVzhXWmcxeDdTV3NIdmV5M3A0V2h6WjJwOGcrZEllWTVKckNod2YrUndkUTNUcEo1bkVFdVR4Z01vd0YvVG9DZitwN3d2dW9HZVg1OE5FN01vanIwUis5NnpHNzVtOE5OUzFtdlZreGpaUTVZSThraHZKSE9zWWkxVTRHcDMweTlmcEtJR2lHVmJZdk40SFI5U2dYY0M1VTgwVXl5aUhYU2JwUkwzUmJDdzh6cHZjRVoraHAxR05TYWRNV0ZhNjFhMzJ0R2w2R1k0VUJNOTBPc3pUZ09CSVFKa0Zwa2llZ3RUMCtqR2Z1NUFjQ3BHcDlJY0pPVXBxYW9hcEJiMlloL2s3UTNDbnZYRE85UUpScThuL3ZpRDJyT21GYW1aN2dKUHVLTkVCZzhRbUZDNktsRVNBMytOenBWbEVNVlhHYjlhZHZZVWRHeFpMM0pTL2gvazBnUXN2WGxzNkRpb3grT1VrRlRRWlg1dy96aS9Hd1ZUcTFqYnEyR0gvRCtSazZNelhzQm1QY2NycjMyamx4V1dxUjJLaGJtczBJK1VHWldPd2Zhelc1Y1BzT1hjNnlVMVJWRW5pNi9nVnN2MFR1dGFQZWZ5RnhpZm9Mb0pPYVNRenRyckZuVEtaOXdGcTB3Qys4Z3VlUnhSdThKdzMycEFrQUlIdWhoM1U3YUJ0aVZzVmN4UFBMZ2d2N0creW9MdnZSdEgxSVJKU0RIRDN1bXFTdHBSL0d0NThzejJxMnBVaWNjVVdxYnV5ZGdIYUZONWZYVXlyR2swdWkwM0pTQjQvUFRQREJyejdqbDVJSmtvVkU3VW1Qa2RXS1pxdGFTb01NbUErM3psWlUzK1JFQ295d3g0WFY3OWM4OGYzTEFGdmRsVEp1YmhzSDlWOVYxcjFQa0xMWElaNDBCRzl6Z2xJUXVxRFdYM1A5VTVnTjlCZ1BiNXFtVWlNRjVVd0RQZldPWDVNaHk3dGUzRGtsRnozbDZlQWhuc2hnT3FkWTVGbnhJV1o3cUc1aTU2a2pPeHdvS25ud3dRK3Q3aFFSemRERmRYMWRETWhDS0FtbWUrVGNnWDR0YnhpaGpsWmZGRkZPWXY0bEp6dUt6Rjk5TVN3K0gwclk1VTBwQlZOd1hIeTJ4eklneUtkVldTbU9VQzNnZ3VmWitwS1A5SWNuNU1HWjF3UHM4ZUxKMVVSRjNlS1NTbWhhZWhkWVlsaVUxRkViaVZsSEJoVTY0N1ZydFd1bVhXNUNQVTNudWZSakZ0M25XUTE3T3BrQWcwTTRRRnZFUlFVMzdLb0dtVUkrc3NEdlMrU3BvbG04Qng3T3JxSEdaUGhxQjV2NURDQS9VUFgxcmhwQ01xNGlNaWdYcVE3Rmdha2owQjE1bXZaVzViUDZXNXdraGdsWm1SRHhCN0R2czhXRWxXT1VDNW1MeWsvRnY4NDFOQTNpemYxMEVHSkRscjAxUm1UL2czRnQvL2orOWpWYTFZSXNjUFU4MXlnc01EZEMvb3ZWdEIySEdCdEJ6Z3BDWnpYYlY1dXpyeSttaVhhTEJCUmRxS1oySVRWSzVqTXhYOU9pQ1V1Mi9WbWx1eXBWanN3cHc1YzhWUGp5eUlydjRKMlloUDErZnNuVkVvMC9VbEp2UVE0N0lqZnRxQm9ZbG1ENnh0cG5ieWwyVFJ3Nm1LMyt0OW9RPT0iLCJleHAiOjE3NDMyOTM0ODAsInNoYXJkX2lkIjozMzk1MTAzMDMsImtyIjoiNjNmN2ZhZCIsInBkIjowfQ.wELKCOw50WZbXRkdPQcoNL2rWgGd_KfEA3mXpT6nLhA  
  name: ev5

loadpoints:
- title: Garten
  charger: wallbox6
  mode: pv

site:
  title: Xxx
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

Version: `0.202.1`
```

### Log details

```text
[main ] INFO 2025/03/30 01:20:31 evcc 0.202.1
[main ] INFO 2025/03/30 01:20:31 using config file: /etc/evcc.yaml
[main ] INFO 2025/03/30 01:20:32 listening at :7070
[bmw ] DEBUG 2025/03/30 01:20:32 identity.Login - database token found
[bmw ] DEBUG 2025/03/30 01:20:32 identity.Login - database token found
[bmw ] DEBUG 2025/03/30 01:20:32 identity.Login - database token invalid. Proceeding to login via user, password and captcha.
[bmw ] DEBUG 2025/03/30 01:20:32 identity.Login - database token invalid. Proceeding to login via user, password and captcha.
[main ] ERROR 2025/03/30 01:20:32 creating vehicle ev5 failed: cannot create vehicle type 'template': cannot create vehicle type 'bmw': unexpected status: 429 (Too Many Requests)
[main ] ERROR 2025/03/30 01:20:32 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'bmw': unexpected status: 429 (Too Many Requests)
[main ] ERROR 2025/03/30 01:21:31 cannot save settings: attempt to write a readonly database (8)
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.202.1

StefanSchoof, 2025-03-30:
`429 (Too Many Requests)`

Wahrscheinlich hast du zu oft probiert und jetzt ist das erstmal für eine gewisse Zeit beim Hersteller gesperrt. Nach einiger Wartezeit noch einmal probieren.

mfuchs1984, 2025-03-30:
Laut https://www.home-assistant.io/integrations/bmw_connected_drive/ 24h warten ohne die API zu benutzen.

AniGerm, 2025-03-30:
Hallo! 
Und brauche ich jetzt für jedes Fahrzeug ein eigenes hcaptcha (wenn ja, das selbe oder verschiedene)?
Komischer Weise werden mir in der UI alle Daten korrekt angezeigt, dort kann ich es aber nicht speichern. Würde mich sehr freuen wenn es die Zeit behebt 😅

AniGerm, 2025-03-30:
Bzw.  wie verhindere ich in evcc das „Polling“? Ich habe keinen HA laufen. Danke!

mfuchs1984, 2025-03-30:
Nicht laden, evcc abschalten oder die Fahrzeuge temporär aus der config nehmen denke ich.

AniGerm, 2025-03-30:
Ok, werde ich versuchen! 
Ich kann mich jetzt auch nicht mehr über die bmw App einloggen, scheinbar ist auch mein Account gesperrt- immerhin geht der von meiner Frau noch;)
Aber über die Art und Anzahl habe ich leider nirgends eine Valide Aussage gefunden.

mfuchs1984, 2025-03-30:
Spannend, eventuell ergibt sich daraus ein Lösungsweg für mehrere BMW: pro BMW in evcc / Home Assistant einen eigenen Account verwenden.

mfuchs1984, 2025-03-30:
Es geht wohl beides, gleiches und unterschiedliches hcaptcha. Mit zwei accounts wäre dann aber wohl ein eigenes jeweils notwendig.

andig, 2025-03-30:
Alternative: ioBroker Integration nutzen.