evcc-io/evcc #20249: heating vs cooling

michiproep, 2025-03-30:
Hallo zusammen,
jetzt da sich die Sonne wieder zeigt und wir langsam wieder Richtung Sommer kommen, fallen mir natürlich wieder neue Features ein.
Ihr habt ja in letzter Zeit viel in das Thema "heating" investiert. Wäre es da nicht ein Leichtes auch "cooling" anzubieten, was im Prinzip ja das Selbe wäre, nur dass der TargetSoc natürlich in die andere Richtung gehen würde.

Ein etwas komplizierterer Fall wäre vielleicht wenn ein Gerät sowohl heizen als auch kühlen kann. Dann bräuchte ein charger/integratedDevice vielleicht noch sowas wie einen "mode", den man zwischen Heizen und Kühlen umstellen kann (auch via Api oder mqtt).

Was haltet ihr davon?

andig, 2025-03-30:
Spricht nichts dagegen. Statt einer globalen Diskussion wäre das aber eine Frage des konkreten Gerätes. Da Klimaanlagen normalerweise keinen Kältespeicher haben, fallen mir wenig sinnvolle Anwendungen ein. Ein konkretes Gerät zeigt vllt. eine.

/cc @premultiply

closed 2025-03-30 by andig

michiproep, 2025-03-30:
Naja, ich hab zwei Geräte die das können. 
Die Pool-Wärmepumpe (via iobroker->tuya) kann zwischen heizen, kühlen und automatik umgestellt werden.
Hier stelle ich manuell eigentlich nur manuell um, weil das an max 7 Tage im Jahr passiert.

Die normale Haus-Wärmepumpe (via Luxtronik2) hat eine Heizgrenze (bei mir Durchschnitt 16°/24h) und kann auch Kühlen (Kühlgrenze 22°/24h).
Das hier entscheidet die WP auf Basis der Grenzen ob sie im Heiz- oder Kühlmodus arbeitet

andig, 2025-03-30:
> Die Pool-Wärmepumpe (via iobroker->tuya) kann zwischen heizen, kühlen und automatik umgestellt werden.

Das passt doch perfekt. Wenn evcc die auf "boost" stellen kann kann sie selbst entscheiden was zu tun ist.

michiproep, 2025-03-30:
@andig: ja, aber zu dem Zeitpunkt is targetSoc > soc  (soll 24°, ist 26°) => evcc startet die WP nicht

michiproep, 2025-03-30:
Technisch möglich, aber nicht chiq. Dann müsste man einen Dummy-Wert einstellen. Soll = 50°
Wir stellen halt gerne auf unserem Evcc-Wand-Display die gewünschte Soll-Temperatur ein.

Lass uns da noch ein wenig Brainstormen. Wie gesagt: Mir fallen halt immer wieder lustige Features ein...

michiproep, 2025-03-30:
Und es gibt noch ein Problem bei deinem Vorschlag: Ich hab ja mehrere Loadpoints. Wenn man einen DummyWert einstellt, wird der ja nie erreicht. D.h. je nach Priorisierung der Loadpoints versucht evcc dann immer Leistung zur WP zu schieben obwohl die eigentlich fertig ist.
Ich denke, das cooling Feature wäre zum Start (anstatt zwischen modi umzuschalten) echt cool. Dann könnte man zumindest ein zweites Gerät anlegen und via Api das eine oder andere auf "aus" schalten.

andig, 2025-03-30:
> D.h. je nach Priorisierung der Loadpoints versucht evcc dann immer Leistung zur WP zu schieben obwohl die eigentlich fertig ist.

…und? Wenn die ihr Limit erreicht hat ist das irrelevant.

michiproep, 2025-03-30:
technisch ja, aber in der UI und in Statistiken unschön.
Und bei vielen Loadpoints plus zugehörigen Threadholds warten man so eh schon ewig bis der nächste Loadpoint mal an der Reihe ist.
Und man kann halt seine Wunschtemperatur dann nicht via evcc setzen.
Ok, war nur ein Vorschlag. Sollen wir's verwerfen?

andig, 2025-03-30:
Da workaround vorhanden: ja.