evcc-io/evcc #20250: Erweiterung von Hardybarth-salia um die Funktionen Identify und Authorize

LembkeM, 2025-03-30:
**Is your feature request related to a problem? Please describe.**
Derzeit implementiert der Hardybarth-salia Charger nicht die Schnittstellen für die Identifikation und Autorisierung. Dies führt dazu, dass wichtige Daten aus den Salia-Rohdaten – insbesondere das strukturierte Feld authorization_request – nicht ausgelesen oder für eine Autorisierung genutzt werden können, was in Integrationsszenarien zu Problemen führt.

**Describe the solution you'd like**
Ich schlage vor, die Hardybarth-salia Implementierung um zwei Funktionen zu erweitern:

Eine Identify()-Methode, die das Feld authorization_request ausliest und mittels eines strukturierten Typs (z. B. AuthorizationRequest mit den Feldern Protocol und Key) eine formatierte Identifikation zurückgibt.
Eine Authorize(key string)-Methode, die einen POST-Request an einen definierten Endpunkt sendet, um die Autorisierung mit dem übergebenen Schlüssel zu veranlassen.

**Describe alternatives you've considered**
Eine Alternative wäre, das Feld authorization_request weiterhin als []string zu verarbeiten. Dies bietet jedoch weniger Typsicherheit und führt zu unklaren Semantiken im Code. Ebenso könnte man den Autorisierungsprozess komplett extern implementieren, was jedoch die Integration und Wiederverwendbarkeit der Funktionalitäten im Charger einschränkt.

**Additional context**
Die Erweiterung um Identify und Authorize verbessert die Integration des Chargers in Systeme, die auf eindeutige Fahrzeugidentifikation und Autorisierung angewiesen sind. Eine strukturierte Implementierung ermöglicht zudem eine bessere Validierung der Daten und erleichtert zukünftige Erweiterungen und Fehlerdiagnosen. Screenshots der relevanten JSON-Daten und Beispielcode (siehe interne Diskussion) können bei Bedarf bereitgestellt werden.

andig, 2025-03-30:
> Eine Identify()-Methode,

Gerne PR.

> Eine Authorize(key string)-Methode, die einen POST-Request an einen definierten Endpunkt sendet, um die Autorisierung mit dem übergebenen Schlüssel zu veranlassen.

Das machen wir nirgendwo sonst in evcc. Wozu wäre das notwendig?

LembkeM, 2025-03-30:
Hi,
also ich habe ein generic vehicle.
Das würde ich gerne über die RFID Karte am Ladepunkt identifizieren.

Ich habe die Lösung auch schon umgesetzt und würde gerne den Branch feature/hardybarth-salia pushen.
Dann auch gerne einen Pull Request starten.

Bei mir lokal läuft das alles schon im Test.

andig, 2025-03-30:
> Das würde ich gerne über die RFID Karte am Ladepunkt identifizieren.

Das ist Identifier. Wie gesagt- gerne PR. Der zweite Teil ist unklar.

LembkeM, 2025-03-30:
Aktuell konnte ich den Branch nicht pushen.

Oder wie arbeitet Ihr?

andig, 2025-03-30:
Du machst einen PR. Siehe Github Doku.

closed 2025-03-31 by andig

LembkeM, 2025-04-01:
@andig ich habe gerade die nightly mit meiner evcc.db gestartet.
Haben wir da etwas übersehen?
In der DB ist die Wallbox (alter Stand) angelegt. Wenn evcc startet bekomme ich folgende Fehlermeldung.
Muss da noch ne Migration gemacht werden?

[main  ] FATAL 2025/04/01 18:07:15 charger [db:5] cannot create charger 'db:5': cannot create charger type 'template': cannot create charger type 'hardybarth-salia': json: cannot unmarshal string into Go struct field .Secc.Port0.rfid.authorization_request of type []string

andig, 2025-04-01:
F***. Du sagtest doch es läuft- wieso jetzt nicht mehr?

andig, 2025-04-01:
Welches Json schickt die Box da?

LembkeM, 2025-04-01:
Ich glaube ich habe nicht das korrekte nightly.

nightly.20250401-9000244

Sollte ich die Änderungen nun im Master sehen?

LembkeM, 2025-04-01:
Also wenn ich in den Master schaue, dann ist dort 

	RFID             struct {
		Type                 string   `json:"type"`
		Available            string   `json:"available"`
		Authorizereq         string   `json:"authorizereq"`
		AuthorizationRequest []string `json:"authorization_request,string"`
	} `json:"rfid"`

Und mein Code sieht aber so aus.

	RFID struct {
		Type                 string               `json:"type"`
		Available            string               `json:"available"`
		Authorizereq         string               `json:"authorizereq"`
		AuthorizationRequest AuthorizationRequest `json:"authorization_request"`
	} `json:"rfid"`

type AuthorizationRequest struct {
	Protocol string
	Key      string
}

func (a *AuthorizationRequest) UnmarshalJSON(data []byte) error {
	// Versuche, die äußeren Anführungszeichen direkt zu entfernen.
	s, err := strconv.Unquote(string(data))
	if err != nil {
		// Falls das Unquoting fehlschlägt, behandeln wir den Input als ungültig.
		*a = AuthorizationRequest{}
		return nil
	}

	// Jetzt enthält s beispielsweise: ["ISO14443","9af18400"]
	var arr []string
	if err := json.Unmarshal([]byte(s), &arr); err != nil {
		*a = AuthorizationRequest{}
		return nil
	}
	if len(arr) == 2 {
		*a = AuthorizationRequest{
			Protocol: arr[0],
			Key:      arr[1],
		}
		return nil
	}
	*a = AuthorizationRequest{}
	return nil
}

andig, 2025-04-01:
Ja, wir sprachen drüber:

> Ich habs mal vereinfacht- könntest Du bitte prüfen ob das immer noch funktioniert? Dann kanns rein :)
Ich werde es heute Abend noch mal prüfen und rückmelden.
Ich habe es gerade überprüft und bei mir funktioniert es.

Aber das hilft jetzt auch nicht ;) Deshalb:

> Welches Json schickt die Box da?

LembkeM, 2025-04-01:
Ich habe es gerade nicht im Zugriff da ich unterwegs bin.

Wenn Du Deine letzten zwei Commita rückgängig machst, funktioniert alles.

LembkeM, 2025-04-02:
Guten Morgen andig,

hier ist der Request.

"rfid":{"type":"stronglink-modbus","available":"1","authorizereq":"","authorization_request":"[\"ISO14443\",\"9cx18400\"]"}

Mfg.
Markus

LembkeM, 2025-04-02:
Dann ist hier wohl das Missgeschick entstanden.

Ich hatte nicht gesehen, dass Du meinen Code gelöscht hattest und ich pullen muss.

Ich hatte meinen Code getestet und der läuft einwandfrei.

> Ja, wir sprachen drüber:
>
> Ich werde es heute Abend noch mal prüfen und rückmelden.
> Ich habe es gerade überprüft und bei mir funktioniert es.
>
> Aber das hilft jetzt auch nicht ;) Deshalb:

andig, 2025-04-02:
> hier ist der Request.

Mit Verlaub: aber das ist schwer zu glauben. Denn das ist kein gültiges JSON, egal ob oder wie wir es dekodieren. Das kannst  Du unter https://jsonlint.com selbst prüfen.

LembkeM, 2025-04-02:
Guten Abend Andig,

es ist nicht mühsam. An sich war das meine Einstigsfrage... wie arbeitet Ihr.

Ich werde am Sonntag wieder Zeit haben, und dann werde ich den Change erneut machen.
Ich werde es mit der Minimallösung machen und dann kannst Du noch mal dürber schauen.

Bis dahin wünsche ich Dir eine gute und erfolgreiche Zeit.

andig, 2025-04-02:
Schau mal ins nightly- ist in deiner Version schon wieder drin 🤗

LembkeM, 2025-04-02:
Mega,

ist drinnen und funzt.

[image]

Gerade gestartet und gleich erkannt.

Wenn Du einfach Tickets hast, kannste mir gerne das nächste zuweisen.
Dann schaue ich was ich machen kann.

Mfg.
Markus

PR #20255 (2025-03-31): hardybarth-salia: add rfid identifier
closes: #20250