evcc-io/evcc #20259: Preislimit für günstiges Netzladen zeitlich begrenzen

aengelhorn, 2025-03-30:
Als Nutzer der Funktion "günstiges Netzladen" würde ich mir ein Häkchen neben der Auswahl der Preisgrenze wünschen. Wenn ich das setze, gilt die ausgewählte Preisgrenze nur für die nächsten 24 Stunden. Damit entfällt das manuelle Rücksetzen des Preislimits.

endurance1968, 2025-03-30:
Warum nicht 48h oder eine Woche oder Monat oder Winter / Sommer? Sprich warum ausgerechnet 24h? 
Macht die Preisgrenze bei Dir nur kurzfristig sinn und wenn ja warum?

andig, 2025-03-30:
Dafür gibts die Apis. Weitere Optionen möchten wir an der Stelle nicht ergänzen.

/cc @naltatis

closed 2025-03-30 by andig

naltatis, 2025-03-31:
@aengelhorn hier gibts einen Sammelthread zum Thema "Ausbaumöglichkeiten der Netzladen Funktion":
https://github.com/evcc-io/evcc/discussions/16364