evcc-io/evcc #20267: show Loadpoint prio

diddip21, 2025-03-30:
Entwurf um Priorität am Ladepunkt anzuzeigen.
[image]
mehrere Loadpoints
[image]

weiterhin könnte dann noch eine Konfigmöglichkeit geschaffen werden:
[image]

**Warum erachte ich das als sinnvoll ?**
Bei dem Einzug der Wärmepumpen oder Heizstäben, möchte ich ggf. die Priorität eher auf die WP legen statt sonst auf die Wallbox.

closed 2025-03-31 by andig

diddip21, 2025-03-31:
Schade... Einfach ohne Kommentar zu gemacht?
Wars so eine besch... Idee ??

andig, 2025-03-31:
Mach dir bitte die Mühe genau hin zu schauen.  Dann wirst Du sehen dass es das Thema bereits gibt. Danke.

diddip21, 2025-03-31:
Ich entschuldige mich, habe deinen Verweis nicht gesehen :(