evcc-io/evcc #20272: Show SoC of all cars in database

eweri, 2025-03-31:
Hello!

I have one wallbox but two EVs - I decide which one to load after sun-forecast, SoC of both cars and expected and the next ride . For this I need the sun-forecast and the SoC of both cars so I have to use two apps to get the needed information. 

I think it would be a great idea, if I could see the SoC of all cars that are configured in database. I think an update every hour would be enough. And this information must be shown at the bottom of evcc.

I open evcc and have sun-forecast, SoC of all of my cars in a single view. This will help to load more solar power  into the right car. 

Thanks, for building evcc.

Bye,
eweri

An-Wi, 2025-03-31:
What about the view below the configuration?

[image]

Is it sufficient for your purposes?

naltatis, 2025-03-31:
Hi @eweri we had a lengthy discussion a while back to this exact topic. There the idea was to introduce a list of all vehicles to the main ui. We decided against this. However, as @An-Wi already said, you can now get this overview in the config view.

Vehicle info update interval is also configurable (via loadpoint).

closed 2025-03-31 by naltatis

eweri, 2025-03-31:
Ah - okay - Thanks - not very convortable but okay.