evcc-io/evcc #20275: Forbidden status Renault

enboer, 2025-03-31:
### Describe the bug

It seems that if there was a charging session ended, EVCC is unable to start/resume a next session. I see forbidden errors in the trace log. If I click in the Renault App to start charging it will eventually start. Attached a trace log of the sections where is went wrong. I remove userid,api keys and tokens. So if you see xxxxx that is because I changed them in the log.

### Steps to reproduce

Let EVCC start a charging session. 
EVVC stops, eg because not enough sun, to much site power, charging settings (<50%)
eg change charging setting to <80%
Evvc tries to charge again, but that fails

In the time between stop and start, car is not removed from the charger.

### Configuration details

```yaml
not applicable. It can communicate correct.
```

### Log details

```text
[site  ] DEBUG 2025/03/30 09:46:49 ----
[lp-1  ] DEBUG 2025/03/30 09:46:49 charge power: 10408W
[lp-1  ] DEBUG 2025/03/30 09:46:49 charge currents: [15.3 15.4 15.2]A
[circuit-carport] DEBUG 2025/03/30 09:46:49 power: 10408W
[circuit-carport] DEBUG 2025/03/30 09:46:49 current: 15.4A
[site  ] DEBUG 2025/03/30 09:46:49 grid power: 9967W
[site  ] DEBUG 2025/03/30 09:46:49 grid currents: [14 14 14]A
[site  ] DEBUG 2025/03/30 09:46:49 pv 1 power: 1108W
[site  ] DEBUG 2025/03/30 09:46:49 site power: 9967W
[lp-1  ] DEBUG 2025/03/30 09:46:49 detected active phases: 3p
[lp-1  ] DEBUG 2025/03/30 09:46:49 charge total import: 171.929kWh
[lp-1  ] DEBUG 2025/03/30 09:46:49 charger status: C
[lp-1  ] DEBUG 2025/03/30 09:46:49 vehicle soc: 51%
[lp-1  ] DEBUG 2025/03/30 09:46:49 vehicle range: 162km
[lp-1  ] DEBUG 2025/03/30 09:46:49 available power 441W < 4140W min 3p threshold
[lp-1  ] DEBUG 2025/03/30 09:46:49 start phase scale1p timer
[lp-1  ] DEBUG 2025/03/30 09:46:49 phase scale1p in 3m0s
[lp-1  ] DEBUG 2025/03/30 09:46:49 pv charge current: 1.56A = 16A + -14.4A (9967W @ 3p)
[lp-1  ] DEBUG 2025/03/30 09:46:49 projected site power 7207W >= 0W disable threshold
[lp-1  ] DEBUG 2025/03/30 09:46:49 pv disable in 0s
[lp-1  ] DEBUG 2025/03/30 09:46:49 pv disable timer elapsed
[renault] TRACE 2025/03/30 09:46:49 GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/xxxxxxxxxxxxxxxxxx/kamereon/kca/car-adapter/v1/cars/xxxxxxxxxxxxxxx/hvac-status?country=DE
[renault] TRACE 2025/03/30 09:46:50 {"data":{"id":"xxxxxxxxxxxxx","attributes":{"internalTemperature":0.0,"hvacStatus":"off","socThreshold":15.0,"lastUpdateTime":"2025-03-29T18:40:51Z"}}}
[circuit-carport] TRACE 2025/03/30 09:46:50 validate current: 15.4A + (16A -> 0A) <= 25A ok
[lp-1  ] DEBUG 2025/03/30 09:46:50 charger disable
[lp-1  ] DEBUG 2025/03/30 09:46:50 wake-up timer: stop
[site  ] DEBUG 2025/03/30 09:46:50 solar forecast: accumulated 31.379kWh, produced 43.440kWh, scale 1.384
[site  ] DEBUG 2025/03/30 09:47:19 ----
[lp-1  ] DEBUG 2025/03/30 09:47:19 charge power: 0W
[lp-1  ] DEBUG 2025/03/30 09:47:19 charge currents: [0 0 0]A
[circuit-carport] DEBUG 2025/03/30 09:47:19 power: 0W
[circuit-carport] DEBUG 2025/03/30 09:47:19 current: 0A
[site  ] DEBUG 2025/03/30 09:47:19 grid power: -657W
[site  ] DEBUG 2025/03/30 09:47:19 grid currents: [1 1 0]A
[site  ] DEBUG 2025/03/30 09:47:19 pv 1 power: 1089W
[site  ] DEBUG 2025/03/30 09:47:19 site power: -657W
[lp-1  ] DEBUG 2025/03/30 09:47:19 charge total import: 171.935kWh
[lp-1  ] DEBUG 2025/03/30 09:47:19 charger status: B
[lp-1  ] INFO 2025/03/30 09:47:19 stop charging <-
[lp-1  ] DEBUG 2025/03/30 09:47:19 charge total import: 171.935kWh
[renault] TRACE 2025/03/30 09:47:19 GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/xxxxxxxxxxxx/kamereon/kca/car-adapter/v2/cars/xxxxxxxxxxxxxx/battery-status?country=DE
[renault] TRACE 2025/03/30 09:47:19 {"data":{"id":"xxxxxxxxxxxxxxxxx","attributes":{"timestamp":"2025-03-30T07:47:13Z","batteryLevel":52,"batteryAutonomy":167,"plugStatus":1,"chargingStatus":0.3,"chargingRemainingTime":111,"V2L_SystemStatusDisplay":0}}}
[lp-1  ] DEBUG 2025/03/30 09:47:19 vehicle soc: 52%
[lp-1  ] DEBUG 2025/03/30 09:47:19 vehicle range: 167km
[lp-1  ] DEBUG 2025/03/30 09:47:19 available power 657W < 4140W min 3p threshold
[lp-1  ] DEBUG 2025/03/30 09:47:19 phase scale1p in 0s
[lp-1  ] DEBUG 2025/03/30 09:47:20 switched phases: 1p
[lp-1  ] DEBUG 2025/03/30 09:47:20 phase timer inactive
[... 439 lines cut ...]
[site  ] DEBUG 2025/03/30 10:08:19 grid power: -2785W
[site  ] DEBUG 2025/03/30 10:08:19 grid currents: [3 3 4]A
[site  ] DEBUG 2025/03/30 10:08:19 pv 1 power: 2703W
[site  ] DEBUG 2025/03/30 10:08:19 site power: -2785W
[lp-1  ] DEBUG 2025/03/30 10:08:19 charge total import: 171.935kWh
[lp-1  ] DEBUG 2025/03/30 10:08:19 charger status: B
[lp-1  ] DEBUG 2025/03/30 10:08:19 pv charge current: 12.1A = 0A + 12.1A (-2785W @ 1p)
[circuit-carport] TRACE 2025/03/30 10:08:19 validate current: 0A + (6.17A -> 12.1A) <= 25A ok
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.202.1

mfuchs1984, 2025-03-31:
Aready tried `alternativewakeup: true`? See https://docs.evcc.io/docs/devices/vehicles#renault "show advanced options".

enboer, 2025-04-01:
I still have this issue. So first charge is ok, but a planned charge during the night fails. alternativewakeup=true does not help.

What I see in the trace is this POST call: https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/xxxxx/kamereon/kca/car-adapter/v1/cars/xxxxx/actions/charging-start?country=DE

If I try this, this call gives an error: Missing Authorization. 

I do not know why this POST is here, since when the successfull charge is started I do not see this POST call.
If I use the api calls after this post they run ok.
Could it be that because the POST call fails evcc stops the start fith the 403 forbidden error.

enboer, 2025-04-01:
But both give the same error:
   error "missing_authorization"
   error_description "Missing API Key header"

mfuchs1984, 2025-04-01:
Maybe Renault changed something?

enboer, 2025-04-01:
I see on the Renault API github people mentioning similar issues....
But as a another solution, it is an option that EVCC does not send start/stop charging to the car? My car is set to charge always or I can set a certain schedule. EVCC just controls the charger, and the car will follow as he gets no power if EVCC decides to stop charging.

mfuchs1984, 2025-04-01:
The wakeup is used only when the car refuses to charge.

enboer, 2025-04-03:
I noticed the following ( I do not think the cause of this forbidden error). The car is plugged in, EVCC tries to get the battery status that fails. But then it logs into the renault account, and does another battery status (orange) , that succeeds. See picture.

[image]

In the picture you see at 18:33 another fail (a few successfulls before that). I am wondering if the time between the calls is on the edge, that Renault requires a smaller time between login and getting the status (currently ~15m30s).

Later in the night (EVCC planned charge), it fails initially again. But after logging in it succeeds

enboer, 2025-04-08:
the endpoint EVCC is using for Renault are not correct for R5 Etech and Megane Etech. (https://github.com/hacf-fr/renault-api/issues/1348)
Planner settings: /commerce/v1/accounts/<account>/kamereon/kcm/v1/vehicles/<vin>/ev/settings

Although it is not clear how to trigger the car  to start charging if it is in sleep mode. See this https://gist.github.com/remo-tech/e9ffa42ddb19b94b21687ffd766dd910

And as in the previous comment, EVCC get the battery status, but without logging in. That is failing as well. Endpoint is ok, but it first needs to go through the authorisation process.

closed 2025-04-20 by github-actions[bot]

javydekoning, 2025-12-11:
> This new endpoint is implemented already, use this in your car settings of EVCC (advanced settings):
> <img alt="Image" width="519" height="358" src="https://private-user-images.githubusercontent.com/47273942/525340710-3643fd73-eb19-4d28-861e-8256040274d3.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjU0NTI0NDIsIm5iZiI6MTc2NTQ1MjE0MiwicGF0aCI6Ii80NzI3Mzk0Mi81MjUzNDA3MTAtMzY0M2ZkNzMtZWIxOS00ZDI4LTg2MWUtODI1NjA0MDI3NGQzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTEyMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMjExVDExMjIyMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdiMjRjMzVlNTNjNDM0MWEwNDRiYWVhZWJhNmY4ZTA5NjdlMmY1MjFhYTM4YmI5ZWMxNjNiZDViNzJiNzU2MzImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.qr1Z8qLtQX6gLjFD5f4B-ZlKx9FxdhreNQGgLVFaeBM">
> 
> I do not have any issues anymore charging my R5

Thank you, I'll try that!