evcc-io/evcc #20279: Anbindung an Teslalogger schlägt fehl (neue config ui issue?)

JanFellner, 2025-03-31:
### Describe the bug

Vorab: Ich nutze die experimentelle Config UI. Möglicherweise ist das Problem auch hierin zu finden.

Ich habe heute evcc from scratch neu aufgesetzt.
Teslalogger läuft schon seit ~2 Jahren ist aber auch hier die aktuellste Version (1.63.4.0)

Versuche ich den Teslalogger als Fahrzeug im evcc hinzu zu fügen scheitere ich nach Eingabe aller Daten beim überprüfen.

🔍 Im Browser log sehe ich - EVCC macht eine HTTP-POST Anfrage an:
https://evcc.meinzuhause.local/api/config/test/vehicle
`
{
	"template": "teslalogger",
	"type": "template",
	"icon": "car",
	"id": "1",
	"port": 5010,
	"title": "Tesla",
	"url": "http://192.168.111.99",
	"capacity": 75
}

Hier als CURL aufruf aus dem Browser geklaut:
`
curl "https://evcc.evcc.meinzuhause.local/api/config/test/vehicle" -X POST -H "User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:137.0) Gecko/20100101 Firefox/137.0" -H "Accept: application/json" -H "Accept-Language: de,en-US;q=0.7,en;q=0.3" -H "Accept-Encoding: gzip, deflate, br, zstd" -H "Content-Type: application/json" -H "Origin: https://evcc.evcc.meinzuhause.local" -H "DNT: 1" -H "Connection: keep-alive" -H "Referer: https://evcc.evcc.meinzuhause.local/" -H "Cookie: auth=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJzdWIiOiJhZG1pbiIsImV4cCI6MTc1MTE4MzExNX0.fXHKzHeuwjIi3Bd9nlLERkfpl4cmICL1NfjUeo1D4Z4" -H "Sec-Fetch-Dest: empty" -H "Sec-Fetch-Mode: cors" -H "Sec-Fetch-Site: same-origin" -H "Priority: u=0" --data-raw "{""template"":""teslalogger"",""type"":""template"",""icon"":""car"",""id"":""1"",""port"":5010,""title"":""Tesla"",""url"":""http://192.168.111.99"",""capacity"":75}"
`

🔥 Und erhalte als Antwort:
`
<html>
<head><title>502 Bad Gateway</title></head>
<body>
<center><h1>502 Bad Gateway</h1></center>
<hr><center>openresty</center>
</body>
</html>
`

🕵️‍♂️ Rufe ich selbst:
http://192.168.111.99:5010/currentjson/1

✅ Bekomme ich valide Werte:
[image]

🤔 Was mir hierbei noch auffällt ich aber nicht bewerten kann:
In der Yaml zum Teslalogger gibt es drei URLs für wake_up, charge_start_stop und set_charging_amps
templates\definition\vehicle\teslalogger.yaml
welche ich nicht aufrufen darf.

[image]

Das Console Log vom TeslaLogger ist hierzu leider auch nicht wirklich aufschlussreich...
`
31.03.2025 12:30:42 : #1[Thread Pool Worker:21]: PostCommand: wake_up - 
31.03.2025 12:30:42 : #1[Thread Pool Worker:21]: Use ProxyServer
31.03.2025 12:30:42 : #1[Thread Pool Worker:62]: Response: <!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>403 Forbidden</title>
</head><body>
<h1>Forbidden</h1>
<p>You don't have permission to access this resource.</p>
<hr>
<address>Apache/2.4.38 (Debian) Server at teslalogger.de Port 4444</address>
</body></html>
`

### Steps to reproduce

- Configured EVCC using the wizard on Windows
- Setup EVCC from scratch as docker image on a synology nas
- Switched to experimental ui
- Add a TeslaLogger

### Configuration details

```yaml
!!! fetched from inside the docker container (no clue how to get the mapped evcc.yaml synced once i switch to the config ui...) !!!

[main  ] INFO 2025/03/31 12:35:37 evcc 0.202.1
[main  ] INFO 2025/03/31 12:35:37 using config file: /etc/evcc.yaml
[db    ] INFO 2025/03/31 12:35:37 using sqlite database: /root/.evcc/evcc.db
[lp-1  ] DEBUG 2025/03/31 12:35:38 charge total import: 812.910kWh

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:80
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 80

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 30s # control cycle interval

meters:
- type: template
  template: sma-hybrid
  id: 3
  host: 192.168.111.140
  port: 502
  usage: pv
  modbus: tcpip
  maxacpower: 10000
  name: pv1
- type: template
  template: sma-hybrid
  id: 3
  host: 192.168.111.140
  port: 502
  usage: battery
  modbus: tcpip
  maxacpower: 10000
  name: battery
- type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.111.226
  name: grid
- type: template
  template: sma-inverter-modbus
  id: 3
  host: 192.168.111.141
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: sma-inverter-modbus
  id: 3
  host: 192.168.111.142
  port: 502
  usage: pv
  modbus: tcpip
  name: pv3

chargers:
- type: template
  template: evsewifi
  host: 192.168.111.246
  name: wallbox6

loadpoints:
- title: Garage
  charger: wallbox6
  mode: pv

site:
  title: Mein Zuhause
  meters:
    grid: grid
    pv:
    - pv1
    - pv2
    - pv3
    battery:
    - battery

hems:
  type: sma
  AllowControl: false

tariffs:
  grid:
    type: template
    template: tibber
    token: *****

Version: `0.202.1`
```

### Log details

```text
I see nothing in the logs when using the config ui
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.202.1

JanFellner, 2025-03-31:
🧨💣 Möchte in dem Zuge auch hinzufügen das der TeslaLogger den Aufruf von set_charging_amps mittlereweile blockiert. So ganz schlau werde ich nicht aber es gibt hierzu hinweise und angepasste Doku von letzter Woche...

https://github.com/bassmaster187/TeslaLogger/issues/1499
https://github.com/bassmaster187/TeslaLogger/commit/e62a7cd52c2e11d3361662d24cc7031c8c1331d2

andig, 2025-03-31:
> I see nothing in the logs when using the config ui

This needs a trace log for tesla logger to understand what exactly fails.

Adminius, 2025-03-31:
> der TeslaLogger den Aufruf von set_charging_amps mittlereweile blockiert

Das ist richtig. Jeder Aufruf von den "commands" kostet jetzt richtig Geld. Es gab "Spezialisten" die den selben Stromwert alle 10 Sekunden immer wieder geschickt haben. Das hat immense Kosten verursacht. Aber auch wenn man nur alle paar Minuten sendet, kostet es trotzdem relativ viel... Hier empfehle ich auf BLE umzusteigen: https://github.com/wimaha/TeslaBleHttpProxy funktioniert perfekt mit EVCC (tesla-ble template).

Wake und charge_start_stop hingegen funktionieren noch, wenn man entsprechende Berechtigungen (Virtual Key) in Tesla hinterlegt hat.
Ich bin mir nicht sicher ob ich set_charging_amps aus dem EVCC Template wieder ausbauen soll oder nicht, denn es gibt Leute, wie mich, die Telemetry Server auf eigene Kosten betreiben, da kann ich selbst für die Commands zahlen und da ist nichts blockiert.
Was ich nicht probiert habe ist TeslaLogger Template für HTTPProxy für die Commands (es gibt ja Option für external Proxy mit BLE).
Da wiederum bräuchte man TeslaLogger nicht mehr, weil alle Daten über BLE kommen.