evcc-io/evcc #20296: Plan: Ladestromanpassung bringt nach Plan-Deaktivierung normales Ladeverhalten durcheinander

An-Wi, 2025-04-01:
### Describe the bug

Nach Deaktivierung eines aktiv laufenden Plans (mit reduziertem Ladestrom) und anschließender Ladeströmerhöhung wird im Min+PV Modus mit maximalem Ladestrom geladen.

Neben Plan 1 habe ich zwei weitere wiederholende Pläne eingerichtet, die aber zeitlich weit genug in der Zukunft liegen.

UI: Zeigt nach Deaktiverung von Plan 1 die Zielzeit von Plan 2 in grüner Schrift an (also "laufende" und Zielzeit wird erreicht)

Erst nach Deaktivieren **aller** Pläne wird die Leistung entsprechend der Vorgaben von Min+PV reduziert.
Danach lassen sich auch die beiden wiederholenden Pläne wieder aktivieren ohne dass diese nun das "normale" Verhalten beeinflussen.

### Steps to reproduce

1. Modus "PV"
2. Ladestrom Max auf 6A begrenzen
3. Plan 1 aktivieren
4. [Ladeplan startet nach Erreichen der geplanten Startzeit]
5. Änderung Modus PV+Min
6. Deaktivierung Plan 1
7. Ladestrom Max auf 16A
8. =>[Lädt mit 10.574W]

### Configuration details

```yaml
# open evcc at http://evcc.local:7070

network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: fupc00041.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 20s   #25s #30s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: ey***

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info   #debug #trace
levels:
  site: debug
  lp-1: debug #trace
  lp-2: debug
  cache: error
  db: error #trace #error
[... 490 lines cut ...]
[lp-1  ] DEBUG 2025/03/30 08:23:08 charger status: C
[lp-1  ] DEBUG 2025/03/30 08:23:08 soc estimated: 57.46% (vehicle: 57.00%)
[lp-1  ] DEBUG 2025/03/30 08:23:08 vehicle soc: 57%
[lp-1  ] DEBUG 2025/03/30 08:23:08 vehicle range: 295km
[lp-1  ] DEBUG 2025/03/30 08:23:08 plan: charge 1h32m8s between 2025-03-31 05:27:52 +0200 CEST until 2025-03-31 07:00:00 +0200 CEST (power: 11040W, avg cost: 0.280)
[lp-1  ] DEBUG 2025/03/30 08:23:08 pv charge current: min 6A > 1.44A (3144W @ 3p, battery: false)
[site  ] DEBUG 2025/03/30 08:23:08 solar forecast: accumulated 62.759kWh, produced 55.037kWh, scale 0.877
[site  ] DEBUG 2025/03/30 08:23:28 ----
```

### What type of operating system or environment does evcc run on?

Windows

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.202.1

VolkerK62, 2025-04-01:
Ein Plan "löschen" heißt wohl nicht, dass die Plan-Ladung sofort aufhört
```
[site  ] DEBUG 2025/03/30 08:20:10 set ev1 plan soc: 0 @ 0001-01-01 01:00:00 +0100 CET
[lp-1  ] DEBUG 2025/03/30 08:20:17 set max current: 16
[site  ] DEBUG 2025/03/30 08:20:28 ----
[lp-1  ] DEBUG 2025/03/30 08:20:28 plan: continuing until end of slot
[lp-1  ] DEBUG 2025/03/30 08:20:28 max charge current: 16A
```

naltatis, 2025-04-01:
\\cc @Maschga

Maschga, 2025-04-01:
Kann ich mir gerne nach #20291 anschauen.

naltatis, 2025-04-02:
> Kann ich mir gerne nach https://github.com/evcc-io/evcc/pull/20291 anschauen.

Super, wenn damit nicht direkt zusammenhängt mach gerne nen separaten PR. Single-Issue PRs sind immer einfacher.

andig, 2025-04-12:
Gibts hier noch was zu tun? Sonst würde ich schließen.

Maschga, 2025-04-12:
Ich hatte noch keine Zeit mir das Issue anzusehen.
Ist wahrscheinlich ein Bug.

Maschga, 2025-04-20:
Es lässt sich reproduzieren und ist ein Bug.

Ich verstehe nicht, warum es zu diesem Verhalten kommt. Soetwas zu debuggen fällt mir nicht leicht, weshalb ich gerne das Issue abgeben würde.

Meine Überlegung:
Bei 6A ist der Plan aktiv und wes wird mit dem Maximum geladen, um das Planziel zu erreichen.
Bei der Umstellung auf 16A wird einfach weiterhin mit dem Maximum geladen, unbeirrt, dass sich der maximale Ladestrom geändert hat.

andig, 2025-04-20:
> Es lässt sich reproduzieren und ist ein Bug.

Mir fehlt hier jegliche nachvollziehbare Fehlerbeschreibung. Soweit ich sehe ändert sich bei 16A der Plan weil weniger Slots gebraucht werden. Passt.

Falls es eine nachvollziehbare Fehlerbeschreibung gibt können wir gerne wieder auf machen.

closed 2025-04-20 by andig

An-Wi, 2025-04-21:
> > Es lässt sich reproduzieren und ist ein Bug.
> 
> Mir fehlt hier jegliche nachvollziehbare Fehlerbeschreibung. Soweit ich sehe ändert sich bei 16A der Plan weil weniger Slots gebraucht werden. Passt.
> 
> Falls es eine nachvollziehbare Fehlerbeschreibung gibt können wir gerne wieder auf machen.

@andig Mh, ich weiß nicht, wie nachvollziehbar die Beschreibung noch sein muss.
Ich habe dieses Problem bei mir mehrfach reproduziert bevor ich ein Issue hier aufgemacht habe und dabei genau mein Vorgehen inklusive der geforderten Logs sowie der YAML-Konfiguration hinterlegt.
Selbst Maschga konnte es reproduzieren und sieht es als Bug - also warum für dich ist es jetzt plötzlich nicht mehr nachvollziehbar?
Ich habe hier sogar bewusst auf Screenshots verzichtet, da ich weiß, dass dir Logs viel wichtiger sind. 

Das Verhalten von evcc ist in oben beschriebenen Fall einfach nicht nachvollziehbar und unlogisch.
Aber sag einfach ganz offen, wenn ich auf weitere Issues dieser Art in Zukunft verzichten soll! Gefühlt werden manche Issuues lieber als "completed" geschlossen, als sie zu beheben.
Bitte nicht falsch verstehen, ich finde dieses Projekt unheimlich genial in seiner Funktion, Design und Bedienbarkeit. Und es ist absolute spitze, wie es sich permanent weiterentwickelt - daher auch mein innerlicher Wunsch, an Verbesserungen teilzuhaben. Allerdings frage ich mich manchmal, warum ich Stunden in die Analyse verschwende, um eine möglichst detailierte Fehlerbeschreibung zu verfassen, damit dann am Ende das Ganze als "nicht nachvollziehbar" geschlossen wird. Bitte nicht persönlich nehmen und nur als konstruktive Kritik sehen - Danke.

andig, 2025-04-21:
> Ich habe dieses Problem bei mir mehrfach reproduziert

Es ist ganz einfach: bisher hat sich niemand dieses Issues angenommen und ich verstehe leider nicht, was "das Problem" ist. Also machen wir entweder zu weil unlösbar oder wir müssen es schaffen, dass ich das Problem verstehe und vielleicht auch lösen kann.

> Bitte nicht persönlich nehmen und nur als konstruktive Kritik sehen - Danke.

Ich nehme es nicht persönlich, aber es hilft halt auch nix.

> Selbst Maschga konnte es reproduzieren und sieht es als Bug - also warum für dich ist es jetzt plötzlich nicht mehr nachvollziehbar?
> Das Verhalten von evcc ist in oben beschriebenen Fall einfach nicht nachvollziehbar und unlogisch.

Lass es uns abkürzen. Was ist- in einfachen Worten- "das Problem"? Eine gute Beschreibung vergleicht erwartetes mit beobachtetem Verhalten. Aus `=>[Lädt mit 10.574W]` kann zumindest ich das nicht interpretieren.

VolkerK62, 2025-04-21:
> Ein Plan "löschen" heißt wohl nicht, dass die Plan-Ladung sofort aufhört
>
> ```
> [site  ] DEBUG 2025/03/30 08:20:10 set ev1 plan soc: 0 @ 0001-01-01 01:00:00 +0100 CET
> [lp-1  ] DEBUG 2025/03/30 [...]

@An-Wi wenn du das reproduzieren kannst, dann sollte sich das gewünschte Verhalten zur nächsten vollen Stunde einstellen. In obigem Fall, sollte also ab 9:00 Uhr in MinPV mit 6A geladen werden.

Das ist mMn. das Zusammentreffen von "Slot nicht unterbrechen" i.V.m. zukünfigen Plänen.
Erst wenn es keine zukünfigen Pläne mehr gibt, dann endet die Plan-Ladung sofort.

andig, 2025-04-21:
Ist denn das- Beenden des Slots- der wahr genommene Fehler?

VolkerK62, 2025-04-21:
in dem Fall ist das NICHT-Beenden des Slots der Fehler, weil der Plan gelöscht wurde.

Hier passt die gewollte Nicht-Unterbrechnung eines Plans, nicht mit evtl. zukünfigen Plänen zusammen.

An-Wi, 2025-04-21:
> Lass es uns abkürzen. Was ist- in einfachen Worten- "das Problem"? Eine gute Beschreibung vergleicht erwartetes mit beobachtetem Verhalten. Aus `=>[Lädt mit 10.574W]` kann zumindest ich das nicht interpretieren.

@andig ok, ich dachte das ging eindeutig aus Step 5 "Änderung Modus PV+Min" hervor, dass nicht mit maximalen 11kW geladen werden sollte...

> in dem Fall ist das NICHT-Beenden des Slots der Fehler, weil der Plan gelöscht wurde.
> 
> Hier passt die gewollte Nicht-Unterbrechnung eines Plans, nicht mit evtl. zukünfigen Plänen zusammen.

Danke @VolkerK62 
Genau das scheint hier das Problem zu sein. Der zukünftige Plan liegt so weit in der Zukunft (Ziel-Ende fast 23h später vom Ausgangszeitpunkt), dass eine "Nicht-Unterbrechung" hier nicht logisch nachvollziehbar ist.

Maschga, 2025-04-21:
Ich habe gerade eben versucht, das Problem erneut zu reproduzieren und dabei ein Logfile und ein Video aufzunehmen, um die Änderungen im UI und das Problem nachvollziehbarer zu machen.
Das ist mir aber nicht gelungen.

Meinen Fehler glaube ich gefunden zu haben: Ich habe gestern - wie in der Beschreibung beschrieben - noch einen wiederholenden Plan eingestellt. Dieser lag aber nicht weit genug in der Zukunft, weshalb evcc dann nach der Deaktivierung des statischen Plans auf den wiederholenden Plan übergesprungen ist und deshalb mit 11 kW geladen hat.

Entschuldigung dafür.

andig, 2025-04-21:
Leider ist das doch nicht so einfach: `SetPlanSoc` ist eine Methode des Fahrzeugs und hat überhaupt keine Verbindung zum Ladepunkt. Entweder müsste man die jetzt krude rausfinden oder es bräuchte eine zusätzliche Bedingung die den Abbruch des Ladeplanes vorziehen könnte in

```go
case lp.clock.Now().Before(lp.planSlotEnd) && !lp.planSlotEnd.IsZero():
	// don't stop an already running slot if goal was not met
	lp.log.DEBUG.Printf("plan: continuing until end of slot at %s", lp.planSlotEnd.Round(time.Second).Local())
	return true
```

An-Wi, 2025-04-22:
Mal ins Unreine gesprochen...
Steht es nicht eigentlich zwischen diese beiden Zeilen:
<pre>
[lp-1  ] DEBUG 2025/03/30 08:20:28 plan: charge 1h33m41s between 2025-03-31 05:26:19 +0200 CEST until 2025-03-31 07:00:00 +0200 CEST (power: 11040W, avg cost: 0.280)
[lp-1  ] DEBUG 2025/03/30 08:20:28 plan: continuing until end of slot
</pre>

- Zeit der Prüfung war: 2025/03/30 08:20:28
- Nächster Plan startet: 2025-03-31 05:26:19
- Zeitliche Differenz = 21h 5min 51s
- Wenn zeitliche Differenz größer als 1-2h (?), dann nicht direkt fortsetzen

Denke ich zu einfach?

andig, 2025-04-22:
Ich vermute wir wollten vermeiden, dass Schwankungen im Estimator dazu führen könnten, dass ein Slot rein und raus fällt. In Anbetracht der Tatsache, dass die Ursache im Logfile klar ist _und_ das Szenario nur dann auftritt wenn ein aktiver Plan gelöscht schlage ich vor in diesem Falle nichts zu tun. Bitte nicht vergessen: es handelt sich um die Situation wo ein Plan abgelaufen ist und das Ladeziel nicht erfüllt! Die saubere Alternative wäre wohl eher, den Ladepunkt für das Fahrzeug zu finden und dort den Plan zu deaktivieren, an den Bedingungen aber nichts zu ändern.

An-Wi, 2025-04-22:
Nochmal kurz zum Verständnis. Was ist für dich der Unterschied zwischen einem aktiven Plan "löschen" und "deaktivieren"?
Und wieso war der Plan "abgelaufen und Ladeziel nicht erfüllt"?
Das Szenario war, dass ich Plan 1 (also den normale, nicht wiederholenden Plan ganz oben) "Aktiv=false" im UI gesetzt hatte.
Plan 2 (wiederholend) war aktiv und dessen Aktiv-Status wurde nicht verändert.

andig, 2025-04-22:
> Nochmal kurz zum Verständnis. Was ist für dich der Unterschied zwischen einem aktiven Plan "löschen" und "deaktivieren"?

In dem Fall gibts keinen da wir uns nicht den aktiven Plan merken sondern einfach ob ein (irgendein) Planslot aktiv ist. Wir wissen dann nicht mehr, zu welchem Plan der gehört.

> Und wieso war der Plan "abgelaufen und Ladeziel nicht erfüllt"?

Weil es einen noch aktiven Planslot gibt, aber keinen Plan mehr zu dem der passen würde (da inaktiv oder gelöscht).

Solange wir diese Verbindung- aktiver Slot zu zugehörigem Plan- nicht haben wird das immer eine Heuristik bleiben. Deshalb würde ich Stand jetzt nix dran ändern.

An-Wi, 2025-04-22:
Ok, das verstehe ich.

In meinem Fall würde meistens eine Heuristik wie "aktiver Plan könnte erst in >10h starten, daher stoppen" ausreichen, aber auch das wäre nur die halbe Lösung für das generelle Problem und könnte blöde Seiteneffekte hervorrufen.
Von daher kann ich mit diesem Hintergrundwissen gut damit leben - durch Deaktivieren der wiederholenden Pläne lässt sich das Verhalten umgehen bzw. beenden, wenn man das Deaktivieren zuvor vergessen hat.
"Closed until revision" 😉

Danke für die vertiefenden Einblicke und deine kostbare Zeit.

andig, 2025-04-22:
Danke :)