evcc-io/evcc #20298: octopus-api apiKey not accepted

a-teece, 2025-04-01:
### Describe the bug

I am attempting to use an API Key with the Octopus Energy template for grid, but it is failing the validation.

My key starts with sk_live_jgg.... and the total length is 32 characters (including the sk_live).

Clearly the configuration is picking up the Api key because the code branch requires ApiKey is not ""

### Steps to reproduce

1. Use a grid section of:
2. tariffs:
  currency: GBP # three letter ISO-4217 currency code (default EUR)
  grid:
    type: template
    template: octopus-api
    apiKey: sk_live_xxxxxxxxxxxxxxxxxxxxxxxx
2. Restart EVCC
3. See the error "creating tariff grid failed: cannot create tariff type 'template': cannot create tariff type 'octopusenergy': invalid apikey format"

### Configuration details

```yaml
tariffs:
  currency: GBP # three letter ISO-4217 currency code (default EUR)
  grid:
    type: template
    template: octopus-api
    apiKey: sk_live_xxxxxxxxxxxxxxxxxxxxxxxx
```

### Log details

```text
creating tariff grid failed: cannot create tariff type 'template': cannot create tariff type 'octopusenergy': invalid apikey format
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.202.1

andig, 2025-04-01:
/cc @duckfullstop could you kindly take a look?

duckfullstop, 2025-04-01:
Are you 100% sure that your API key is exactly 32 characters in length? The error you're hitting is super simple:

https://github.com/evcc-io/evcc/blob/b4988f4d7b5103d9ac94fcf73d42b818b743f0bf/tariff/octopus.go#L64

a-teece, 2025-04-01:
> Are you 100% sure that your API key is exactly 32 characters in length? The error you're hitting is super simple:
> 
> [evcc/tariff/octopus.go](https://github.com/evcc-io/evcc/blob/b4988f4d7b5103d9ac94fcf73d42b818b743f0bf/tariff/octopus.go#L64)
> 
> Line 64 in [b4988f4](/evcc-io/evcc/commit/b4988f4d7b5103d9ac94fcf73d42b818b743f0bf)
> 
>  if len(cc.ApiKey) != 32 || !strings.HasPrefix(cc.ApiKey, "sk_live_") {

Sorry, it is. I looked at the line of code and couldn't understand it either. In the sample config I supplied I swapped every char after the sk_live_ with an x, double checked the count and that also failed with the same error. I also manually re-typed the api key in case the copy-paste from the Octopus website was using a unicode _ that looks the same, but isn't (I've had that before with a - char in my own code).

Hopefully the replication is super-simple using the config I put above.

duckfullstop, 2025-04-01:
Able to replicate, investigating now...

(Was this working in a prior version? It seems like `cc.ApiKey` isn't being set from config properly...)

duckfullstop, 2025-04-01:
Fixed in #20304, just needs merge

PR #20304 (2025-04-02): tariff/octopus: Fix improper template render to struct
Closes #20298

... who knows why this broke, it's just a single character typo
closed 2025-04-02 by andig