evcc-io/evcc #20320: Fehlende lp.resetPVTimer() calls?

waynistan, 2025-04-02:
### Describe the bug

Ich habe jetzt ein paar Mal beobachtet, dass ein Ladevorgang sofort nach Ladestart ohne Verzögerung gestoppt wurde. Mein Verdacht ist, dass an 2 Stellen in core/loadpoint.go fehlende lp.resetPVTimer()-Aufrufe dieses Problem verursachen. Einmal zwischen den Zeilen 1395 & 1397 und einmal zwischen den Zeilen 1427 & 1429.

Hier ist der entsprechende Codeausschnitt:
https://github.com/evcc-io/evcc/blob/02938e00d026e78e83cc5e3da52db94825fac697/core/loadpoint.go#L1376-L1443

### Steps to reproduce

Unmittelbar nach einem Ladestart muss die disableThreshold-Bedingung erfüllt sein.
Evtl. funktioniert der Bug auch umgekehrt, sprich wenn sofort nach einem Ladestopp die enableThreshold-Bedingung erfüllt ist und enableDelay <= disableDelay (+ interval?) ist.

### Configuration details

```yaml
network:
  port: 7070

interval: 9s

sponsortoken: -

meters:
- name: my_grid
  type: custom
  power:
    source: mqtt
    topic: grid/power
    scale: -1
    timeout: 3s
- name: my_pv
  type: custom
  power:
    source: mqtt
    topic: pv/power
    timeout: 3s
- name: my_battery
  type: custom
  power:
    source: mqtt
    topic: battery/power
    timeout: 3s
  soc:
    source: mqtt
    topic: battery/soc
    timeout: 3s
  capacity: 9.83
- name: my_charger
  type: custom
  power:
    source: mqtt
    topic: charger/power
    timeout: 10s
  energy:
    source: mqtt
    topic: charger/energy
    timeout: 10s
  currents:
  - source: mqtt
    topic: charger/current_L1
    timeout: 10s
  - source: mqtt
    topic: charger/current_L2
    timeout: 10s
  - source: mqtt
    topic: charger/current_L3
    timeout: 10s

chargers:
- name: mennekes_amtron_charge_control
  type: custom
  status:
    source: mqtt
    topic: charger/vehicleState
    timeout: 10s
  enabled:
    source: mqtt
    topic: charger/enabled
    timeout: 11s
  enable:
    source: mqtt
    topic: charger/enable
  maxcurrent:
    source: mqtt
    topic: charger/maxCurrent

vehicles:
- name: opel_corsa_e
  type: custom
  title: Corsa-e
  capacity: 44
  soc:
    source: mqtt
    topic: corsa/soc
    timeout: 24h
  status:
    source: mqtt
    topic: corsa/status
    timeout: 24h
  range:
    source: mqtt
    topic: corsa/range
    timeout: 24h
  odometer:
    source: mqtt
    topic: corsa/odometer
    timeout: 24h
  climater:
    source: mqtt
    topic: corsa/climater
    timeout: 24h
- name: skoda_enyaq_85
  type: custom
  title: Enyaq 85
  capacity: 77
  soc:
    source: mqtt
    topic: enyaq/soc
    timeout: 24h
  status:
    source: mqtt
    topic: enyaq/status
    timeout: 24h
  range:
    source: mqtt
    topic: enyaq/range
    timeout: 24h
  odometer:
    source: mqtt
    topic: enyaq/odometer
    timeout: 24h
  climater:
    source: mqtt
    topic: enyaq/climater
    timeout: 24h
  limitsoc:
    source: mqtt
    topic: enyaq/limitSoc
    timeout: 24h

site:
- title: Opelring 8
  meters:
    grid: my_grid
    pv: my_pv
    battery: my_battery

loadpoints:
- title: Wallbox
  charger: mennekes_amtron_charge_control
  meter: my_charger
  #vehicle: opel_corsa_e
  mode: pv

mqtt:
  broker: 127.0.0.1:1883
  topic: evcc
  user: -
  password: -

log: debug
```

### Log details

```text
Leider nicht mehr verfügbar.
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.202.1

andig, 2025-04-02:
Wir können uns das gerne anschauen- wenn es ein aussagekräftiges Logfile gibt.

waynistan, 2025-04-02:
```text
[site  ] DEBUG 2025/04/02 17:01:36 ----
[lp-1  ] DEBUG 2025/04/02 17:01:36 charge power: 0W
[lp-1  ] DEBUG 2025/04/02 17:01:36 charge currents: [0 0 0]A
[site  ] DEBUG 2025/04/02 17:01:36 grid power: -4700W
[site  ] DEBUG 2025/04/02 17:01:36 pv 1 power: 5000W
[site  ] DEBUG 2025/04/02 17:01:36 battery 1 power: 0W
[site  ] DEBUG 2025/04/02 17:01:36 battery 1 soc: 100%
[site  ] DEBUG 2025/04/02 17:01:36 site power: -4600W
[lp-1  ] DEBUG 2025/04/02 17:01:36 charge total import: 6328.000kWh
[lp-1  ] DEBUG 2025/04/02 17:01:36 charger status: B
[lp-1  ] DEBUG 2025/04/02 17:01:36 pv charge current: 6.67A = 0A + 6.67A (-4600W @ 3p)
[lp-1  ] DEBUG 2025/04/02 17:01:36 site power -4600W <= -4240W enable threshold
[lp-1  ] DEBUG 2025/04/02 17:01:36 pv enable in 6s
[lp-1  ] DEBUG 2025/04/02 17:01:36 pv enable timer remaining: 6s
[site  ] DEBUG 2025/04/02 17:01:45 ----
[lp-1  ] DEBUG 2025/04/02 17:01:45 charge power: 0W
[lp-1  ] DEBUG 2025/04/02 17:01:45 charge currents: [0 0 0]A
[site  ] DEBUG 2025/04/02 17:01:45 grid power: -4700W
[site  ] DEBUG 2025/04/02 17:01:45 pv 1 power: 5000W
[site  ] DEBUG 2025/04/02 17:01:45 battery 1 power: 0W
[site  ] DEBUG 2025/04/02 17:01:45 battery 1 soc: 100%
[site  ] DEBUG 2025/04/02 17:01:45 site power: -4600W
[lp-1  ] DEBUG 2025/04/02 17:01:45 charge total import: 6328.000kWh
[lp-1  ] DEBUG 2025/04/02 17:01:45 charger status: B
[lp-1  ] DEBUG 2025/04/02 17:01:45 pv charge current: 6.67A = 0A + 6.67A (-4600W @ 3p)
[lp-1  ] DEBUG 2025/04/02 17:01:45 site power -4600W <= -4240W enable threshold
[lp-1  ] DEBUG 2025/04/02 17:01:45 pv enable in 0s
[lp-1  ] DEBUG 2025/04/02 17:01:45 pv enable timer elapsed
[lp-1  ] DEBUG 2025/04/02 17:01:45 max charge current: 6A
[lp-1  ] DEBUG 2025/04/02 17:01:45 charger enable
[lp-1  ] DEBUG 2025/04/02 17:01:45 wake-up timer: start
[site  ] DEBUG 2025/04/02 17:01:54 ----
[lp-1  ] DEBUG 2025/04/02 17:01:54 charge power: 4140W
[lp-1  ] DEBUG 2025/04/02 17:01:54 charge currents: [6 6 6]A
[site  ] DEBUG 2025/04/02 17:01:54 grid power: -0W
[site  ] DEBUG 2025/04/02 17:01:54 battery 1 power: 3000W
[site  ] DEBUG 2025/04/02 17:01:54 battery 1 soc: 100%
[site  ] DEBUG 2025/04/02 17:01:54 pv 1 power: 1440W
[site  ] DEBUG 2025/04/02 17:01:54 site power: 3100W
[lp-1  ] DEBUG 2025/04/02 17:01:54 charge total import: 6328.000kWh
[lp-1  ] DEBUG 2025/04/02 17:01:54 charger status: C
[lp-1  ] DEBUG 2025/04/02 17:01:54 charge start energy: 6328.000kWh
[lp-1  ] INFO 2025/04/02 17:01:54 start charging ->
[lp-1  ] DEBUG 2025/04/02 17:01:54 wake-up timer: stop
[lp-1  ] DEBUG 2025/04/02 17:01:54 soc estimated: 67.00% (vehicle: 67.00%)
[lp-1  ] DEBUG 2025/04/02 17:01:54 vehicle soc: 67%
[lp-1  ] DEBUG 2025/04/02 17:01:54 vehicle soc limit: 80%
[lp-1  ] DEBUG 2025/04/02 17:01:54 vehicle range: 314km
[lp-1  ] DEBUG 2025/04/02 17:01:54 pv charge current: 1.51A = 6A + -4.49A (3100W @ 3p)
[lp-1  ] DEBUG 2025/04/02 17:01:54 projected site power 3100W >= 100W disable threshold
[lp-1  ] DEBUG 2025/04/02 17:01:54 pv disable in 0s
[lp-1  ] DEBUG 2025/04/02 17:01:54 pv disable timer elapsed
[lp-1  ] DEBUG 2025/04/02 17:01:54 charger disable
[lp-1  ] DEBUG 2025/04/02 17:01:54 wake-up timer: stop
[site  ] DEBUG 2025/04/02 17:02:03 ----
[lp-1  ] DEBUG 2025/04/02 17:02:03 charge power: 0W
[lp-1  ] DEBUG 2025/04/02 17:02:03 charge currents: [0 0 0]A
[site  ] DEBUG 2025/04/02 17:02:03 grid power: -4700W
[site  ] DEBUG 2025/04/02 17:02:03 pv 1 power: 5000W
[site  ] DEBUG 2025/04/02 17:02:03 battery 1 power: 0W
[site  ] DEBUG 2025/04/02 17:02:03 battery 1 soc: 100%
[site  ] DEBUG 2025/04/02 17:02:03 site power: -4600W
[lp-1  ] DEBUG 2025/04/02 17:02:03 charge total import: 6328.000kWh
[lp-1  ] DEBUG 2025/04/02 17:02:03 charger status: B
[lp-1  ] DEBUG 2025/04/02 17:02:03 charge final energy: 0.000kWh
[lp-1  ] INFO 2025/04/02 17:02:03 stop charging <-
[lp-1  ] DEBUG 2025/04/02 17:02:03 pv timer reset
[lp-1  ] DEBUG 2025/04/02 17:02:03 pv timer inactive
[lp-1  ] DEBUG 2025/04/02 17:02:03 charge total import: 6328.000kWh
[lp-1  ] DEBUG 2025/04/02 17:02:03 soc estimated: 67.00% (vehicle: 67.00%)
[lp-1  ] DEBUG 2025/04/02 17:02:03 vehicle soc: 67%
[lp-1  ] DEBUG 2025/04/02 17:02:03 vehicle soc limit: 80%
[lp-1  ] DEBUG 2025/04/02 17:02:03 vehicle range: 314km
[lp-1  ] DEBUG 2025/04/02 17:02:03 pv charge current: 6.67A = 0A + 6.67A (-4600W @ 3p)
[lp-1  ] DEBUG 2025/04/02 17:02:03 site power -4600W <= -4240W enable threshold
[lp-1  ] DEBUG 2025/04/02 17:02:03 pv enable timer start: 1m0s
[lp-1  ] DEBUG 2025/04/02 17:02:03 pv enable in 1m0s
```
Dieses Logfile habe ich zur schnellen Reproduzierbarkeit mit simulierten Inputs erstellt. Der enable- und der disableDelay waren beide auf 60s eingestellt.

Wenn das Auto innerhalb eines intervals nach Ladefreigabe die Ladung beginnt und gleichzeitig die disableThreshold-Bedingung erfüllt ist, dann wird die Ladung sofort unterbrochen.
Umgekehrt scheint der Fehler nicht zu bestehen, da erfolgt ein pvTimer-Reset.

Also anders als im ersten Post vermutet fehlt wahrscheinlich nur ein lp.resetPVTimer()-Call, der aber evtl. nicht zwischen die im ersten Post erwähnten Zeilen 1427 & 1429 gehört, sondern in die evChargeStartHandler()-Funktion, das entsprechende Gegenstück in der evChargeStopHandler()-Funktion existiert nämlich bereits.

Hier die beiden Funktionen, die sich aktuell im Bezug auf den pvTimer-Reset nicht symmetrisch verhalten:
https://github.com/evcc-io/evcc/blob/25d36a5f486a8df1275ffe81bbe988a9a9e0caca/core/loadpoint.go#L424-L461

waynistan, 2025-04-02:
```text
[site  ] DEBUG 2025/04/02 20:57:53 ----
[lp-1  ] DEBUG 2025/04/02 20:57:53 charge power: 0W
[lp-1  ] DEBUG 2025/04/02 20:57:53 charge currents: [0 0 0]A
[site  ] DEBUG 2025/04/02 20:57:53 grid power: -4700W
[site  ] DEBUG 2025/04/02 20:57:53 pv 1 power: 5000W
[site  ] DEBUG 2025/04/02 20:57:53 battery 1 power: 0W
[site  ] DEBUG 2025/04/02 20:57:53 battery 1 soc: 100%
[site  ] DEBUG 2025/04/02 20:57:53 site power: -4600W
[lp-1  ] DEBUG 2025/04/02 20:57:53 charge total import: 7000.000kWh
[lp-1  ] DEBUG 2025/04/02 20:57:53 charger status: B
[lp-1  ] DEBUG 2025/04/02 20:57:53 pv charge current: 6.67A = 0A + 6.67A (-4600W @ 3p)
[lp-1  ] DEBUG 2025/04/02 20:57:53 site power -4600W <= -4240W enable threshold
[lp-1  ] DEBUG 2025/04/02 20:57:53 pv enable in 6s
[lp-1  ] DEBUG 2025/04/02 20:57:53 pv enable timer remaining: 6s
[site  ] DEBUG 2025/04/02 20:58:02 ----
[lp-1  ] DEBUG 2025/04/02 20:58:02 charge power: 0W
[lp-1  ] DEBUG 2025/04/02 20:58:02 charge currents: [0 0 0]A
[site  ] DEBUG 2025/04/02 20:58:02 grid power: -4700W
[site  ] DEBUG 2025/04/02 20:58:02 battery 1 power: 0W
[site  ] DEBUG 2025/04/02 20:58:02 battery 1 soc: 100%
[site  ] DEBUG 2025/04/02 20:58:02 pv 1 power: 5000W
[site  ] DEBUG 2025/04/02 20:58:02 site power: -4600W
[lp-1  ] DEBUG 2025/04/02 20:58:02 charge total import: 7000.000kWh
[lp-1  ] DEBUG 2025/04/02 20:58:02 charger status: B
[lp-1  ] DEBUG 2025/04/02 20:58:02 pv charge current: 6.67A = 0A + 6.67A (-4600W @ 3p)
[lp-1  ] DEBUG 2025/04/02 20:58:02 site power -4600W <= -4240W enable threshold
[lp-1  ] DEBUG 2025/04/02 20:58:02 pv enable in 0s
[lp-1  ] DEBUG 2025/04/02 20:58:02 pv enable timer elapsed
[lp-1  ] DEBUG 2025/04/02 20:58:02 max charge current: 6A
[lp-1  ] DEBUG 2025/04/02 20:58:02 charger enable
[lp-1  ] DEBUG 2025/04/02 20:58:02 wake-up timer: start
[site  ] DEBUG 2025/04/02 20:58:11 ----
[lp-1  ] DEBUG 2025/04/02 20:58:11 charge power: 0W
[lp-1  ] DEBUG 2025/04/02 20:58:11 charge currents: [0 0 0]A
[site  ] DEBUG 2025/04/02 20:58:11 grid power: -0W
[site  ] DEBUG 2025/04/02 20:58:11 pv 1 power: 5000W
[site  ] DEBUG 2025/04/02 20:58:11 battery 1 power: 300W
[site  ] DEBUG 2025/04/02 20:58:11 battery 1 soc: 100%
[site  ] DEBUG 2025/04/02 20:58:11 site power: 400W
[lp-1  ] DEBUG 2025/04/02 20:58:11 charge total import: 7000.000kWh
[lp-1  ] DEBUG 2025/04/02 20:58:11 charger status: B
[lp-1  ] DEBUG 2025/04/02 20:58:11 pv charge current: 0A = 0A + -0.58A (400W @ 3p)
[lp-1  ] DEBUG 2025/04/02 20:58:11 projected site power 400W >= 100W disable threshold
[lp-1  ] DEBUG 2025/04/02 20:58:11 pv disable in 0s
[lp-1  ] DEBUG 2025/04/02 20:58:11 pv disable timer elapsed
[lp-1  ] DEBUG 2025/04/02 20:58:11 charger disable
[lp-1  ] DEBUG 2025/04/02 20:58:11 wake-up timer: stop
[site  ] DEBUG 2025/04/02 20:58:20 ----
[lp-1  ] DEBUG 2025/04/02 20:58:20 charge power: 0W
[lp-1  ] DEBUG 2025/04/02 20:58:20 charge currents: [0 0 0]A
[site  ] DEBUG 2025/04/02 20:58:20 grid power: -4700W
[site  ] DEBUG 2025/04/02 20:58:20 pv 1 power: 5000W
[site  ] DEBUG 2025/04/02 20:58:20 battery 1 power: 0W
[site  ] DEBUG 2025/04/02 20:58:20 battery 1 soc: 100%
[site  ] DEBUG 2025/04/02 20:58:20 site power: -4600W
[lp-1  ] DEBUG 2025/04/02 20:58:20 charge total import: 7000.000kWh
[lp-1  ] DEBUG 2025/04/02 20:58:20 charger status: B
[lp-1  ] DEBUG 2025/04/02 20:58:20 pv charge current: 6.67A = 0A + 6.67A (-4600W @ 3p)
[lp-1  ] DEBUG 2025/04/02 20:58:20 site power -4600W <= -4240W enable threshold
[lp-1  ] DEBUG 2025/04/02 20:58:20 pv enable in 0s
[lp-1  ] DEBUG 2025/04/02 20:58:20 pv enable timer elapsed
[lp-1  ] DEBUG 2025/04/02 20:58:20 max charge current: 6A
[lp-1  ] DEBUG 2025/04/02 20:58:20 charger enable
[lp-1  ] DEBUG 2025/04/02 20:58:20 wake-up timer: start
[site  ] DEBUG 2025/04/02 20:58:29 ----
[lp-1  ] DEBUG 2025/04/02 20:58:29 charge power: 0W
[lp-1  ] DEBUG 2025/04/02 20:58:29 charge currents: [0 0 0]A
[site  ] DEBUG 2025/04/02 20:58:29 grid power: -0W
[site  ] DEBUG 2025/04/02 20:58:29 battery 1 power: 300W
[site  ] DEBUG 2025/04/02 20:58:29 pv 1 power: 5000W
[site  ] DEBUG 2025/04/02 20:58:29 battery 1 soc: 100%
[site  ] DEBUG 2025/04/02 20:58:29 site power: 400W
[lp-1  ] DEBUG 2025/04/02 20:58:29 charge total import: 7000.000kWh
[lp-1  ] DEBUG 2025/04/02 20:58:29 charger status: B
[lp-1  ] DEBUG 2025/04/02 20:58:29 pv charge current: 0A = 0A + -0.58A (400W @ 3p)
[lp-1  ] DEBUG 2025/04/02 20:58:29 projected site power 400W >= 100W disable threshold
[lp-1  ] DEBUG 2025/04/02 20:58:29 pv disable in 0s
[lp-1  ] DEBUG 2025/04/02 20:58:29 pv disable timer elapsed
[lp-1  ] DEBUG 2025/04/02 20:58:29 charger disable
[lp-1  ] DEBUG 2025/04/02 20:58:29 wake-up timer: stop
```

Ich habe noch ein bisschen experimentiert und es ist tatsächlich möglich mit halber interval-Frequenz "charger enable; charger disable"-Sequenzen zu erzeugen.
Im simulierten Fall reagiert das Fahrzeug nicht schnell genug und fängt gar keine Ladung an. Es muss sich nur im richtigen Moment der PV-Überschuss hinreichend stark ändern. Ich habe hier 300 Watt Grundlast und eine mit interval-Frequenz umschaltende 5 kW Last simuliert.
Vermutlich ist doch die ganz oben erwähnte Lösung, nämlich der sofortige pvTimer-Reset an den beiden o.g. Stellen in pvMaxCurrent() am besten, außer es spricht etwas anderes dagegen.

Mir ist klar, dass das in der Praxis nur selten passiert (wobei es bei mir jetzt min. 3x vorgekommen ist in wenigen Tagen, da mein Wechselrichter bei Lastschwankungen gerne mal kurzzeitig in der PV-Leistung einbricht).
Ich bin aber der Meinung, dass "zu schnelle" Schaltfolgen ausschließlich bei händischem Eingriff möglich sein sollten, niemals automatisch.

andig, 2025-04-12:
Magst du PR machen damit wir uns die konkreten Zeilen nochmal anschauen können? Die Timer Resets sollten nur bei Änderung ein/ausschalten notwendig sein, nicht bei reiner Stromänderung.

waynistan, 2025-04-12:
Ich habe wenig Erfahrung mit Github bisher, also entweder bin ich im Moment zu doof einen Branch + Pull request zu erstellen oder ich habe gar nicht die Rechte dazu, keine Ahnung. Aber was ich machen kann ist einfach hier die entsprechende Code-Änderung vorzuschlagen:

core/loadpoint.go ab Zeile 1395:
```
if elapsed >= lp.GetDisableDelay() {
	lp.log.DEBUG.Println("pv disable timer elapsed")

	// reset timer to prevent immediate re-enabling
	lp.resetPVTimer()

	return 0
}
```

core/loadpoint.go ab Zeile 1427:
```
if elapsed >= lp.GetEnableDelay() {
	lp.log.DEBUG.Println("pv enable timer elapsed")

	// reset timer to prevent immediate re-disabling
	lp.resetPVTimer()

	return minCurrent
}
```

Der Kommentar + der direkt folgende lp.resetPVTimer()-Afuruf sind jeweils von mir hinzugefügt.

closed 2025-04-14 by andig