evcc-io/evcc #20326: Add lower charging limit for cascading heating

Coernel82, 2025-04-02:
**Is your feature request related to a problem? Please describe.**
Heatpump and heating rod work at the same time wasting money: As the heatpump is more efficient than an electric heating rod (COP ≈ 4) it would make sense to heat the water up to the heatpump's limit before using excess energy for electric heating and feed the excess to the grid.

**Describe the solution you'd like**
Set a lower limit for the heating rod: is allowed to heat only when water is above x - which is the maximum the heatpump can achieve. Heatpump then switches off itself and heating rod adds some degrees to the water.

**Describe alternatives you've considered**
Simple plugin using the API.

Coernel82, 2025-04-02:
Problem:

Eine Wärmepumpe ist effizienter als eine elektrische Heizpatrone (COP ≈ 4 bedeutet, dass sie mit einer Einheit Strom vier Einheiten Wärme erzeugt).

Wenn beide gleichzeitig laufen, wird unnötig Geld verschwendet, da die Heizpatrone ineffizienter ist.

Statt die überschüssige PV-Energie direkt in die Heizpatrone zu leiten, sollte erst die Wärmepumpe das Wasser maximal aufheizen, bevor die Heizpatrone aktiv wird.

Gewünschte Lösung:

Es soll eine untere Temperaturlimit für die Heizpatrone geben.

Die Heizpatrone soll erst dann aktiv werden, wenn das Wasser bereits auf die maximale Temperatur aufgeheizt wurde, die die Wärmepumpe erreichen kann.

Die Wärmepumpe schaltet sich dann von selbst ab, und erst danach darf die Heizpatrone noch ein paar Grad drauflegen.

Alternative Lösung:

Der Nutzer überlegt, stattdessen ein einfaches Plugin zu schreiben, das die EVCC-API nutzt, um das Verhalten zu steuern.

Im Kern geht es darum, dass erst die effizientere Wärmepumpe das Maximum rausholt, bevor die ineffizientere Heizpatrone genutzt wird.

andig, 2025-04-02:
… und genau das steuert evcc überhaupt nicht sondern wir überlassen die Entscheidung der WP.

Coernel82, 2025-04-03:
Wenn ich die Wärmepumpe und einen Heizstab als Loadpoints habe, über z.B. ein Shelly mit MQTT angesteuert, dann kann ich den Boostmode setzen und sogar mit bisschen Shelly Logik sie ausschalten, was aber Off topic ist.
D.h. ich habe folgende Konstellation:

Wärmepumpe mit hoher Priorität fängt an zu heizen. Freigabe von evcc bleibt bestehen, weil Überschuss. Irgendwann kann die Wärmepumpe aber nicht mehr, bei sagen wir 60 Grad, Wärmepumpe schaltet ab, logischerweise weiß evcc das nur indirekt, da kein Stromverbrauch mehr, ist aber ohnehin irrelevant. 

Da viel Sonnenstrom da ist wird aber parallel schon per Heizstab geheizt, was Geldverschwendung ist, es wäre ökonomischer, erst ab dem Limit der Wärmepumpe ein wenig nachzuheizen und vorher einzuspeisen.

D.h. wenn jetzt der Heizstab ein unteres Limit hat (du darfst ab 58 Grad erst anfangen und Horst bei 65 Grad auf) dann ist die Priorisierung eleganter.

Ich hoffe ich konnte es jetzt verständlich erklären.

andig, 2025-04-03:
https://github.com/evcc-io/evcc/issues/20326#issuecomment-2772745924

premultiply, 2025-04-03:
evcc ist ein Energiemanagementsystem und keine Heizungssteuerung oder sonstiges Automationsprogramm.
Solche Logiken etc. kannst du dir prima z. B. in einer Hausautomation nach Bedarf zusammenstellen.
Eventuell reicht dafür aber auch ein einfacher Thermostat der deinen Heizstab erst _ab_ einer bestimmten Temperatur aktiviert...
Auch Tasmota, Shelly usw. sollten das hinbekommen.

Coernel82, 2025-04-04:
Also wenn wir vom ursprünglichen Namen ausgehen ist evcc eine Ladesteuerung für Autos und wir müssen dann auch sämtlichen Shelly support etc. rausnehmen.
Letztendlich ist aber mein Vorschlag genau das, was du sagst. Wie nennt man es denn dann, wenn der Heizstab erst später angeht und die Energie anderweitig genutzt wird?

Es ist ein Vorschlag und eine Idee und das Entwicklerteam hat alles Recht, diesen für gut oder schlecht zu befinden.

Mit einem Scheinargument der Kategorie "falsche Analogie" oder auch argumentum ad antiquitatem würden wir immer noch Hexen verbrennen und Aderlass betreiben.