evcc-io/evcc #20334: Skoda Citigo hat falschen Kilometerstand

pencoe, 2025-04-02:
### Describe the bug

Für meinen Skoda Citigo wird ein falscher Kilometerstand abgerufen: seit 12.03. wird immer unverändert ein Stand von 45379 km geliefert, der reale Kilometerstand ist derzeit 45955 km. Die MySkoda-Integration in HomeAssistant liefert den korrekten Wert.

Das Problem ist die Abfrage bei Skoda (siehe Log-Eintrag).

### Steps to reproduce

Skoda Citigo konfigurieren und Daten abrufen...

### Configuration details

```yaml
Konfig in UI, nicht in evcc.yaml
```

### Log details

```text
[enyaq ] TRACE 2025/04/02 16:41:22 GET https://mysmob.api.connect.skoda-auto.cz/api/v1/vehicle-health-report/warning-lights/***
[enyaq ] TRACE 2025/04/02 16:41:22 {"capturedAt":"2025-03-12T07:55:25Z","mileageInKm":45379,"warningLights":[{"category":"ASSISTANCE","defects":[]},{"category":"COMFORT","defects":[]},{"category":"BRAKE","defects":[]},{"category":"ENGINE","defects":[]},{"category":"LIGHTING","defects":[]},{"category":"TIRE","defects":[]},{"category":"OTHER","defects":[]}]}
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.202.1

andig, 2025-04-02:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2025-04-02 by andig

pencoe, 2025-04-03:
Es gibt im Log ja eben nichts zu sehen außer der Tatsache, dass bei der Abfrage halt ein falscher (alter) Wert zurückkommt. Exakt die beiden Log-Zeilen mit dem Abruf (insbesondere der Abruf-URL) und dem Ergebnis habe ich geliefert und darauf hingewiesen, dass das MySkoda-Projekt (https://github.com/skodaconnect/myskoda) die richtigen Daten liefert!