evcc-io/evcc #20365: Review and validate logic from #20183

andig, 2025-04-04:
See https://github.com/evcc-io/evcc/pull/20183

andig, 2025-04-04:
@mfuchs1984 magst Du Dich mal bei uns auf Slack melden?

mfuchs1984, 2025-04-06:
Wenn ich nicht ganz falsch liege, könnte der ganze Aufwand umsonst gewesen sein und es hätte einfach nur der falsche Parameter korrigiert werden müssen :unamused: :grinning:

premultiply, 2025-04-06:
In der Tat ist die Bezeichnung `lp.chargeCurrent` für die Stromvorgabe vs `lp.chargeCurrents` / `lp.chargePower` maximal unglücklich und verwirrend.
Da bekomme ich auch einen Knoten ins Hirn.

Da wäre `lp.maxCurrent` oder `lp.offeredCurrent` wohl eine passendere Bezeichnung...

mfuchs1984, 2025-04-06:
`lp.offeredCurrent` oder `lp.currentOffered` wären für mich am intuitivsten.

mfuchs1984, 2025-04-19:
https://github.com/evcc-io/evcc/pull/20713 

- "charging logik" entfernt, alte logik wiederhergestellt. stattdessen wird ValidateCurrent jetzt korrekt mit der richtigen old chargeCurrent aufgerufen + Bugfix für ein Szenario, wo der alte code buggy war  (incl. Tests)

mfuchs1984, 2025-04-19:
Bei den ganzen Änderungen und Szenarien ging total unter, dass die circuit power / currents ja auch negativ sein können. Das geht jetzt leider schief, siehe https://github.com/evcc-io/evcc/discussions/20712

Deshalb machte die Logik der alten Implementierung total Sinn und eine stumpfe Limitierung auf max circuit power / current eben nicht.

andig, 2025-04-21:
1000 Dank @mfuchs1984- das hätten wir ohne dich nicht geschafft!

closed 2025-04-21 by andig

mfuchs1984, 2025-04-21:
Danke, am Ende haben wir gemeinsam eine einfach und gute Lösung gefunden 👍