evcc-io/evcc #20368: Identifier werden beim Vehicle als Zahl interpretiert

TobiasHuber1980, 2025-04-04:
### Describe the bug

Identifier werden beim vehicle als Zahl interpretiert -> z.B. e12 erzeugt dann 12 Nullen

### Steps to reproduce

follow up: https://github.com/evcc-io/evcc/discussions/19565

### Configuration details

```yaml
alles im UI konfiguriert
```

### Log details

```text
evcc dump:
vehicle: VW ID.5 (Tobias Huber)
-------------------------------
Soc:            80%
Capacity:       77.0kWh
Charge status:  B
Range:          333km
Odometer:       25761km
Finish time:    2025-04-03 15:19:00 +0200 CEST
Climate active: false
Limit Soc:      80%
Identifiers:    [4115000000000000]
OnIdentified:   Priority:1, MinCurrent:6, MaxCurrent:16

evcc charger:
----
Power:          0W
Energy:         8762.8kWh
Current L1..L3: 0A 0A 0A
Charge status:  B
Status reason:  unknown
Enabled:        true
Identifier:     04115e12
Features:       [WelcomeCharge]
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.202.1

naltatis, 2025-04-04:
@TobiasHuber1980 füg mal bitte deine Config hinzu.

naltatis, 2025-04-04:
Ok, der Effekt tritt unabhängig von Config UI auf.

Auch diese Config

```
  - name: vehicle_1
    type: template
    template: offline
    identifiers:
      - "04115e12"
```

führt zur `Identifiers: [4115000000000000]` in `evcc vehicle`.

@andig das scheint schon direkt in `util.DecodeOther` zu passieren.
https://github.com/evcc-io/evcc/blob/1c1ec03aac1d63a5502214a191b02534181a4ad2/vehicle/vehicle.go#L42-L43

naltatis, 2025-04-04:
Scheint auch nicht nur `identifiers` sondern generell String-Felder zu betreffen.

```
  - name: vehicle_6
    type: template
    template: offline
    title: "14e12"
```

wird auch zu `14000000000000`.

andig, 2025-04-04:
Ja, Ursache gefunden. Das passiert beim umverpacken von Template in custom. Da müssen die Quotes erhalten bleiben.

closed 2025-04-04 by andig