evcc-io/evcc #20371: Solar Forecast: Erträge vor Sonnenaufgang

marcelGoerentz, 2025-04-04:
### Describe the bug

Die solare Vorhersage sagt, dass Erträge vor Sonnenaufgang zu erwarten sind (Sonnenaufgang ist ca. 7:00):
[image]

### Steps to reproduce

1. PV Vorhersage konfigurieren

### Configuration details

```yaml
network:
  schema: http
  host: ****** # .local suffix announces the hostname on MDNS
  port: ******

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 10s # control cycle interval

meters:
  - name: Netz
    type: template
    template: lg-ess-home-8-10
    usage: grid
    host: ***** # IP-Adresse oder Hostname
    password: *****
  - name: PV
    type: template
    template: lg-ess-home-8-10
    usage: pv
    host: ***** # IP-Adresse oder Hostname
  - name: Batterie
    type: template
    template: lg-ess-home-8-10
    usage: battery
    host: ***** # IP-Adresse oder Hostname
    capacity: 9.8 # Akkukapazität in kWh (optional)
    minsoc: 20 # Untere Grenze beim Entladen der Batterie (normaler Betrieb) (optional)
    maxsoc: 80 # Oberes Limit beim Laden der Batterie aus dem Netz (optional)

mqtt:
  broker: localhost:1883
  topic: evcc
  clientid: evcc-server

site:
  title: Mein Haus

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.31 # EUR/kWh
  feedin:
    type: fixed
    price: 0.08 # EUR/kWh
  co2:
    type: template
    template: grünstromindex
    zip: *****
  solar:
    - type: template
      template: forecast-solar
[... 730 lines cut ...]
      "energy": 21597.3377782126
    },
    "tomorrow": {
      "complete": true,
      "energy": 29420
    }
  }
}
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.202.1

Chris45671, 2025-04-04:
[image]
Bei mir ist der Sonnenaufgang nach 6 Uhr. 
So wie ich den forecast verstehe, wird hier der Wert für die gesamte Stunde angezeigt. 
Ich habe heute in der Zeit von 6 Uhr bis 7 Uhr schon rund 100 Watt erzeugt.

marcelGoerentz, 2025-04-04:
Ich wohne ziemlich im Westen von Deutschland, da ging heute die Sonne erst um 7:03 auf. Selbst wenn mein Dach nach Osten gerichtet wäre würde ich vorher kein Ertrag erzielen können.

naltatis, 2025-04-04:
@marcelGoerentz was liefert denn forecast.solar für einen Standpunkt zurück, wenn du die API direkt aufrufst?

marcelGoerentz, 2025-04-05:
@naltatis Das Problem scheint sich von alleine gelöst zu haben.

closed 2025-04-05 by marcelGoerentz