evcc-io/evcc #20374: Fahrzeug wird nicht mehr erkannt seit V202.1

deepsee1, 2025-04-04:
### Describe the bug

Hallo EVCC,
seit dem Update auf 202.1 wird Fahrzeug nicht mehr automatisch erkannt, wenn es an der Wallbox eingesteckt wird. Es wurde nichts verändert, ausser das Update auf V202.1 durchgeführt

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
vehicles:
  - name: Mini
    type: template
    template: mini
    user: xxx.xxx@xxx.de
    password: xxxxxxxxxxx
    #vin: WBMW... # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
    title: Mini
    #capacity: 50 # Akkukapazität in kWh (optional)
    welcomecharge: true
    cache: 15m # Zeitintervall nach dem Daten erneut vom Fahrzeug abgefragt werden (optional)
    region: EU
    hcaptcha: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
```

### Log details

```text
site ] DEBUG 2025/04/04 11:30:04 solar forecast: accumulated 36.574kWh, produced 53.646kWh, scale 1.467
[site ] DEBUG 2025/04/04 11:30:35 ----
[lp-1 ] DEBUG 2025/04/04 11:30:35 charge power: 4140W
[site ] DEBUG 2025/04/04 11:30:35 grid power: -381W
[site ] DEBUG 2025/04/04 11:30:35 pv 2 power: 2872W
[site ] DEBUG 2025/04/04 11:30:35 pv 1 power: 2609W
[site ] DEBUG 2025/04/04 11:30:35 pv power: 5481W
[site ] DEBUG 2025/04/04 11:30:35 site power: -381W
[lp-1 ] DEBUG 2025/04/04 11:30:35 charger status: C
[lp-1 ] DEBUG 2025/04/04 11:30:35 pv charge current: 6.55A = 6A + 0.552A (-381W @ 3p)
[site ] DEBUG 2025/04/04 11:30:35 solar forecast: accumulated 36.600kWh, produced 53.693kWh, scale 1.467
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

2002.1

andig, 2025-04-04:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2025-04-04 by andig

deepsee1, 2025-04-04:
evcc runs under homeassist on a raspberry pi 5. I do not get an error message in the logfile of evcc neither per trace nor per debug. As far as I know, it is not possible to create the logfile on the command line when evcc is running on homeassist