evcc-io/evcc #20375: No Entsoe data anymore

enboer, 2025-04-04:
### Describe the bug

Since yesterday I do not get price info from Entsoe anymore. I do not have the url link tomtest if it is on the Entsoe side. And I do not know if the data is correct.

### Steps to reproduce

1. No specific steps on error that there is no matching data found
2.
3.
...

### Configuration details

```yaml
202.1
```

### Log details

```text
entsoe] ERROR 2025/04/04 10:11:38 No matching data found for Data item ENERGY_PRICES and interval 2025-04-04T10:00:00.000Z/2025-04-05T10:00:00.000Z
[entsoe] TRACE 2025/04/04 10:11:39 <?xml version="1.0" encoding="UTF-8"?> <Acknowledgement_MarketDocument xmlns="urn:iec62325.351:tc57wg16:451-1:acknowledgementdocument:7:0"> <mRID>0311fee9-0805-4</mRID> <createdDateTime>2025-04-04T08:11:39Z</createdDateTime> <sender_MarketParticipant.mRID codingScheme="A01">10X1001A1001A450</sender_MarketParticipant.mRID> <sender_MarketParticipant.marketRole.type>A32</sender_MarketParticipant.marketRole.type> <receiver_MarketParticipant.mRID codingScheme="A01">10X1001A1001A450</receiver_MarketParticipant.mRID> <receiver_MarketParticipant.marketRole.type>A39</receiver_MarketParticipant.marketRole.type> <received_MarketDocument.createdDateTime>2025-04-04T08:11:39Z</received_MarketDocument.createdDateTime> <Reason> <code>999</code> <text>No matching data found for Data item ENERGY_PRICES and interval 2025-04-04T10:00:00.000Z/2025-04-05T10:00:00.000Z</text> </Reason> </Acknowledgement_MarketDocument>
[entsoe] ERROR 2025/04/04 10:11:39 No matching data found for Data item ENERGY_PRICES and interval 2025-04-04T10:00:00.000Z/2025-04-05T10:00:00.000Z
[entsoe] TRACE 2025/04/04 11:11:38 <?xml version="1.0" encoding="UTF-8"?> <Acknowledgement_MarketDocument xmlns="urn:iec62325.351:tc57wg16:451-1:acknowledgementdocument:7:0"> <mRID>0deac184-650c-4</mRID> <createdDateTime>2025-04-04T09:11:38Z</createdDateTime> <sender_MarketParticipant.mRID codingScheme="A01">10X1001A1001A450</sender_MarketParticipant.mRID> <sender_MarketParticipant.marketRole.type>A32</sender_MarketParticipant.marketRole.type> <receiver_MarketParticipant.mRID codingScheme="A01">10X1001A1001A450</receiver_MarketParticipant.mRID> <receiver_MarketParticipant.marketRole.type>A39</receiver_MarketParticipant.marketRole.type> <received_MarketDocument.createdDateTime>2025-04-04T09:11:38Z</received_MarketDocument.createdDateTime> <Reason> <code>999</code> <text>No matching data found for Data item ENERGY_PRICES and interval 2025-04-04T11:00:00.000Z/2025-04-05T11:00:00.000Z</text> </Reason> </Acknowledgement_MarketDocument>
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.202.1

RenatusRo, 2025-04-04:
das passiert alle paar wochen mal. 
ist  ein entso-problem (bzw. eines von deren data-provider)

RenatusRo, 2025-04-04:
see 
https://transparency.entsoe.eu/
-> outages

enboer, 2025-04-04:
Thanks. I filled in the webbrowser the http call that is in the log file. For yesterday it gives data, but not for today.....

RenatusRo, 2025-04-04:
> Thanks. I filled in the webbrowser the http call that is in the log file. For yesterday it gives data, but not for today.....

yep, if they don't have data they can't give you data:

Dear Transparency Platform users,
Please be informed that Energy Prices 12.1.D values are missing  for Areas BZN HU, GR, SI, FR, AT (Seq 1), CZ, SK, NL, BE, RO, HR, BG, DE-LU (Seq 1), PL in Transparency Platform since 03.04.2025. Data provider is informed.
Apologies for any disruption this may have caused.
Kind regards,
Transparency Platform team

closed 2025-04-08 by enboer