evcc-io/evcc #20384: EVCC Restart während Ladevorgang führt zu falscher Berechnung der geladenen kWh

sammy708, 2025-04-04:
### Describe the bug

Eine Neustart von EVCC während gerade ein Ladevorgang läuft, führt zu einer falschen Berechnung der Einträge in der Statistik der Ladevorgänge. Habe einen Screenshot angehängt. Die 44,1 kwH sind korrekt, Dann erfolgte nach 12:40 Uhr ein restart und dann hätten innerhalb 20 Min 45,2 kwh gelanden werden müssen. Das gibt mein Stromanschluss gar nicht her ;-)

[image]

### Steps to reproduce

1. Ladevorgang starten
2. systemctl restart evcc
3. Auswerten Ladevorgänge nach Beenden des Ladevorgangs
...

### Configuration details

```yaml
wird nachgereicht
```

### Log details

```text
no specific logs
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.202.1

andig, 2025-04-04:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

GrimmiMeloni, 2025-04-04:
Wir brauchen dafür insbesondere auch ein Trace log, da wir sonst nicht sehen welche Daten der Charger sendet.

Dazu in `evcc.yaml` die Logging Section entsprechend anpassen, z.B. so:

```
log: debug
levels:
  cache: error
  easee: trace
```

sammy708, 2025-04-04:
> Wir brauchen dafür insbesondere auch ein Trace log, da wir sonst nicht sehen welche Daten der Charger sendet.
> 
> Dazu in `evcc.yaml` die Logging Section entsprechend anpassen, z.B. so:
> 
> ```
> log: debug
> levels:
>   cache: error
>   easee: trace
> ```

alles klar. danke. yaml habe ich jetzt ergänzt. Logs werdeb NACHGEREICHT - muss das morgen mit erweiterten Log/trace erst reproduzieren.

sammy708, 2025-04-05:
hier die Logfiles:
```text
Apr 05 12:20:00 evcc evcc[8766]: [easee ] TRACE 2025/04/05 12:20:00 ProductUpdate EHCDJ7A8: (2025-04-05 10:20:00 +0000 UTC) SESSION_ENERGY 14.23585319519043
Apr 05 12:20:09 evcc evcc[8766]: [site  ] DEBUG 2025/04/05 12:20:09 ----
Apr 05 12:20:09 evcc evcc[8766]: [site  ] WARN 2025/04/05 12:20:09 planner: tariff not available: cannot create tariff type 'template': cannot create tariff type 'grünstromindex': unexpected status: 503 (Service Unavailable)
Apr 05 12:20:09 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:09 charge power: 3532W
Apr 05 12:20:09 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:09 charge currents: [15.2 0.02 0.009]A
Apr 05 12:20:09 evcc evcc[8766]: [site  ] DEBUG 2025/04/05 12:20:09 grid power: -7W
Apr 05 12:20:09 evcc evcc[8766]: [site  ] DEBUG 2025/04/05 12:20:09 pv 1 power: 4913W
Apr 05 12:20:09 evcc evcc[8766]: [site  ] DEBUG 2025/04/05 12:20:09 battery 1 power: -772W
Apr 05 12:20:09 evcc evcc[8766]: [site  ] DEBUG 2025/04/05 12:20:09 battery 1 soc: 55%
Apr 05 12:20:09 evcc evcc[8766]: [site  ] DEBUG 2025/04/05 12:20:09 site power: -579W
Apr 05 12:20:09 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:09 detected active phases: 1p
Apr 05 12:20:09 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:09 charge total import: 4537.070kWh
Apr 05 12:20:09 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:09 charger status: C
Apr 05 12:20:09 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:09 soc estimated: 72.44% (vehicle: 71.00%)
Apr 05 12:20:09 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:09 vehicle soc: 72%
Apr 05 12:20:09 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:09 vehicle soc limit: 100%
Apr 05 12:20:09 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:09 vehicle range: 228km
Apr 05 12:20:09 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:09 pv charge current: 18.5A = 16A + 2.52A (-579W @ 1p)
Apr 05 12:20:28 evcc evcc[8766]: [easee ] TRACE 2025/04/05 12:20:28 ProductUpdate EHCDJ7A8: (2025-04-05 10:20:26 +0000 UTC) TOTAL_POWER 3.507999897003174
Apr 05 12:20:29 evcc evcc[8766]: [site  ] DEBUG 2025/04/05 12:20:29 ----
Apr 05 12:20:29 evcc evcc[8766]: [site  ] WARN 2025/04/05 12:20:29 planner: tariff not available: cannot create tariff type 'template': cannot create tariff type 'grünstromindex': unexpected status: 503 (Service Unavailable)
Apr 05 12:20:29 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:29 charge power: 3508W
Apr 05 12:20:29 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:29 charge currents: [15.2 0.02 0.009]A
Apr 05 12:20:29 evcc evcc[8766]: [site  ] DEBUG 2025/04/05 12:20:29 grid power: -16W
Apr 05 12:20:29 evcc evcc[8766]: [site  ] DEBUG 2025/04/05 12:20:29 pv 1 power: 4910W
Apr 05 12:20:29 evcc evcc[8766]: [site  ] DEBUG 2025/04/05 12:20:29 battery 1 power: -762W
Apr 05 12:20:29 evcc evcc[8766]: [site  ] DEBUG 2025/04/05 12:20:29 battery 1 soc: 55%
Apr 05 12:20:29 evcc evcc[8766]: [site  ] DEBUG 2025/04/05 12:20:29 site power: -578W
Apr 05 12:20:29 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:29 detected active phases: 1p
Apr 05 12:20:29 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:29 charge total import: 4537.070kWh
Apr 05 12:20:29 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:29 charger status: C
Apr 05 12:20:29 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:29 soc estimated: 72.44% (vehicle: 71.00%)
Apr 05 12:20:29 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:29 vehicle soc: 72%
Apr 05 12:20:29 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:29 vehicle soc limit: 100%
Apr 05 12:20:29 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:29 vehicle range: 228km
Apr 05 12:20:29 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:29 pv charge current: 18.5A = 16A + 2.51A (-578W @ 1p)
Apr 05 12:20:49 evcc evcc[8766]: [site  ] DEBUG 2025/04/05 12:20:49 ----
Apr 05 12:20:49 evcc evcc[8766]: [site  ] WARN 2025/04/05 12:20:49 planner: tariff not available: cannot create tariff type 'template': cannot create tariff type 'grünstromindex': unexpected status: 503 (Service Unavailable)
Apr 05 12:20:49 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:49 charge power: 3508W
Apr 05 12:20:49 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:49 charge currents: [15.2 0.02 0.009]A
Apr 05 12:20:49 evcc evcc[8766]: [site  ] DEBUG 2025/04/05 12:20:49 grid power: -133W
Apr 05 12:20:49 evcc evcc[8766]: [site  ] DEBUG 2025/04/05 12:20:49 battery 1 power: -778W
Apr 05 12:20:49 evcc evcc[8766]: [site  ] DEBUG 2025/04/05 12:20:49 pv 1 power: 4920W
Apr 05 12:20:49 evcc evcc[8766]: [site  ] DEBUG 2025/04/05 12:20:49 battery 1 soc: 55%
Apr 05 12:20:49 evcc evcc[8766]: [site  ] DEBUG 2025/04/05 12:20:49 site power: -711W
Apr 05 12:20:49 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:49 detected active phases: 1p
Apr 05 12:20:49 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:49 charge total import: 4537.070kWh
Apr 05 12:20:49 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:49 charger status: C
Apr 05 12:20:49 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:49 soc estimated: 72.44% (vehicle: 71.00%)
Apr 05 12:20:49 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:49 vehicle soc: 72%
Apr 05 12:20:49 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:49 vehicle soc limit: 100%
Apr 05 12:20:49 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:49 vehicle range: 228km
Apr 05 12:20:49 evcc evcc[8766]: [lp-1  ] DEBUG 2025/04/05 12:20:49 pv charge current: 19.1A = 16A + 3.09A (-711W @ 1p)
[... cut ...]
Apr 05 12:26:39 evcc evcc[9559]: [lp-1  ] DEBUG 2025/04/05 12:26:39 vehicle soc: 72%
Apr 05 12:26:39 evcc evcc[9559]: [lp-1  ] DEBUG 2025/04/05 12:26:39 vehicle soc limit: 100%
Apr 05 12:26:39 evcc evcc[9559]: [lp-1  ] DEBUG 2025/04/05 12:26:39 vehicle range: 230km
Apr 05 12:26:39 evcc evcc[9559]: [lp-1  ] DEBUG 2025/04/05 12:26:39 pv charge current: 19A = 16A + 2.97A (-684W @ 1p)
Apr 05 12:26:39 evcc evcc[9559]: [site  ] DEBUG 2025/04/05 12:26:39 solar forecast: accumulated 0.086kWh, produced 0.118kWh, scale 1.365
```

sammy708, 2025-04-05:
hier noch ein aktualisierter Screenshot der Ladeübersicht:

[image]

GrimmiMeloni, 2025-04-05:
Looks valid. Reopening, but will need time later today (or tomorrow) to have a closer look.

GrimmiMeloni, 2025-04-06:
Was able to reproduce in a test. Not sure if it is possible to fix.

The underlying problem is that the charger's session keeps running (which makes sense from Easee side - they don't know the controlling entity is restarting).
When evcc stops it closes the session based on the session energy so far.
When evcc starts and detect a running charge, it starts a new session. 
So for Easee this is one big session, but for evcc these are two.

The problem stems from the fact that the charger keep reporting its own session's energy (which includes the energy prior to the evcc restart). At the same time, we currently do not account for the slice of energy from the previous charge already being "booked". 
As far as I can see, there's also no way to tell this apart based purely on the charger's data.

Options:
* Do nothing - simply don't restart evcc while charging. 🤷‍♂  
* Do not use the session energy from the charger, and make calculations inside evcc based purely on the charger's meter. This has a known drawback, that the meter values are reported by Easee very irregularly (once an hour?). We actually spent quite some time in #9940 to come up with a somewhat elegant heuristic to reduce these errors in measuring to a minimum. I am therefore not in favor of this option, as it would raise this error again, and also it impacts all regular charges. 
* Keep behavior of evcc charger as is, but use the charge meter values from the sessions to calculate some reasonable max values when closing a session. Based on what I see so far, I think that for the incorrect sessions the charge meter start value will also be unreasonable.

@sammy708 to confirm the latter theory, I would need to have a look at your charge details (just the one's from April suffice). Can you please share the CSV download for April here (or if you don't want to make this public, send it to me at grund.heuboden0v@icloud.com )

cc @allcoolusernamesaregone

sammy708, 2025-04-07:
Thanks very much for the analysis. I will provide the April CSV latest by end of the week, as I am without access to the site, cause of traveling at the moment.

sammy708, 2025-04-11:
Comming back to your previous post, I have atteched the april csv as requested. just anonymized the "Kennung" field.
Thanks for investigating!

[session-2025-04.csv](https://github.com/user-attachments/files/19702482/session-2025-04.csv)

andig, 2025-04-12:
> At the same time, we currently do not account for the slice of energy from the previous charge already being "booked".

@GrimmiMeloni eigentlich tun wir das, zumindest wenn evcc neu startet. Schau mal im `chargerater` nach `cr.startEnergy`. Kann natürlich ein Fehler drin sein?

GrimmiMeloni, 2025-04-12:
I’m still looking for a free time slot to check this.

GrimmiMeloni, 2025-04-14:
OK, so I dug into this a bit more, found two issues, and learned once again a lot about evcc. 😅 

First, thanks for the hint @andig regarding the existing charge rater behavior, that was helpful to know. I was about to suggest an offset implementation inside `easee.go` - which luckily we don't need given it is already there.

So, the problem with Easee is, when (re)starting, we do not wait for the **complete** state of the charger, only for opMode (see https://github.com/evcc-io/evcc/issues/11431 and https://github.com/evcc-io/evcc/pull/11342). This means during setup of the loadpoint we now have a data race. If OP_MODE Product Update arrives before the SESSION_ENERGY update, we effectively start with a 0 session energy, even thought the session is ongoing. That in turn makes the existing offset logic in chargerater ineffective, and hence energy gets counted twice, which is the issue reported here. 

This one is simple enough to fix (just wait for more than just OP_MODE, apparently there's a few more "essential" states we need).

I will create a PR later (hopefully today).

For second issue I will write a follow up in #18401; TLDR - the charge meter is broken and this will lead to incorrect meter start and stop values in the charge history. I am only mentioning it here, to indicate that I am aware - would still like to keep this issue focussed on the session's energy.

PR #20646 (2025-04-16): Easee: wait for essential state during boot
fix #20384

@andig this likely works, but I don't like it - so I would appreciate your input on my thinking below.

The former solution waited for a single ProductUpdate (CHARGER_OP_MODE) and this signalling was quite elegant.
Also, since OP_MODE does not change often and no compute was needed this was fine.

Now, the situation is different, as we need to check for presence of multiple ProductUpdates to have arrived.
Currently, I am reusing the map with the observation timestamps (`c.obsTime`) and check for the respective keys to be present.

What I  [... cut]
PR #20871 (2025-05-01): Easee: wait for optional state during boot
fix #20384 

changed approach compared to the previous solution:
* charger op mode remains the sole mandatory state
* once that is available, we wait for up to 3 seconds max for further optional states (session energy, and meter values)
closed 2025-05-01 by andig