evcc-io/evcc #20392: Modbus Template Mennekes AMTRON 4You 560

opitzb86, 2025-04-04:
**Is your feature request related to a problem? Please describe.**
- Für die Mennekes AMTRON 5You 560 ist aktuell kein Modbus Template vorhanden 
- Einbindung über OCPP möglich aber:
    - scheint durch Mennekes nicht gut implementiert, daher häufiger timeout, out of sync, neustart wallbox notwendig
    - automatische Phasenumschalltung nicht richtig
    - Authorisierung nur über Backend möglich, nicht über App oder lokale Whitelist. Dadurch keine Zuordnung zu Fahrzeug möglich

**Describe the solution you'd like**
- Modbus Template mit automatischer Phasenumschalltung.

**Describe alternatives you've considered**
- OCPP, sieht oben.

opitzb86, 2025-04-04:
Hi,
ich habe das entsprechende Modbus Template schon lokal programmiert und getestet und dachte das ich über das **issue** einen **branch** erzeuge wo ich dass dann pushen könnte. 

Kann mir hier bitte jemand helfen wie der "korrekte" weg ist um das Template zur Verfügung zu stellen?
Danke.

StefanSchoof, 2025-04-04:
Du musst evcc forken. Dann erzeugst du einen Branch in dem fork und kannst dann ein Pull Request erzeugen.

PR #20425 (2025-04-05): Add Mennekes AMTRON 4You
Replaces #20420
Fixes #20392
closed 2025-04-05 by premultiply