evcc-io/evcc #20396: Use VIN as vehicle identifier within the database instead of title

starwarsfan, 2025-04-04:
**Is your feature request related to a problem? Please describe.**
I'd like to wipe out my yaml configuration and use the ui to configure my vehicles. But I don't want to loose the load history of the vehicles, so I asked [here](https://github.com/evcc-io/evcc/discussions/20394#discussioncomment-12729907) if and how it's possible. The given answer stated that the "title" is the mapping key between a configured vehicle and it's load history in the database. IMHO this is not a user friendly approach, so I'd like to suggest the usage of the VIN as key for the vehicles.

**Describe the solution you'd like**
With the current approach it's not possible to rename the vehicles without losing the history. Let's say I have a vehicle called "Enyaq" and later on another one joins the family. Then one should be able to rename the first one like "Orange Enyaq" or whatever. The VIN is a clear identifier of a vehicle, which can't be changed at all. So if it is used to identify a vehicle, the user would be free to (re-)name them without any loss of load history.

andig, 2025-04-04:
> IMHO this is not a user friendly approach

Its unclear why that would be or what the problem with this is.

starwarsfan, 2025-04-05:
Ok, so let me explain.

From a consistency point of view it's a bad idea to use a modifiable key to identify an object. So if that key is changed, the initial object won't be referenced anymore. Here in this case this would lead to a separate set of data, if the user modify the Title of a vehicle.

My suggestion is to use a distinct and unmodifiable key to identify the configured vehicles and this would be the serial number.

andig, 2025-04-05:
You make it sound like a huge thing. It‘s a single field in a SQL table. If you don‘t like it you can always UPDATE it to whatever you like.

closed 2025-04-05 by andig

starwarsfan, 2025-04-05:
I'm not talking about me, as a developer I know how to deal with databases on that level. What I'm talking about is the average user who might come into the described situation! For that case it would be nice to prevent such unwanted side effects.

But never mind, it was just a suggestion.

laasfuus, 2026-02-02:
I happily migrated from yaml based config to native evcc bluelink integration. I kept the title same and that worked. 

Still I’m bit uncomfortable of keeping my data intact for years as I already have double entry for my car. (It happened during first setup that included some trial and error).