evcc-io/evcc #20411: Ladestrom wird zu gering berechnet bei Verwendung von maxAcPower

vedje, 2025-04-05:
### Describe the bug

Hallo,

ich habe das Problem festgestellt, dass ein pausierter Ladevorgang nicht gestartet bzw. ein laufender Ladevorgang abgebrochen wird, obwohl ein mehr als ausreichender PV-Überschuss zur Verfügung steht und auch der Wechselrichter noch mehr als 1400W bzw. 6A zusätzlich generieren könnte. Ursache dafür ist (laut evcc.log), dass der Wert für `pv charge current` zu niedrig bestimmt wird und `site power` für mich überhaupt nicht nachvollziehbare Werte aufweist. Jedenfalls liegt der in evcc angezeigte (Haus)Verbrauch bis zu 500W über dem realen Verbrauch.

Meine Vermutung ist, dass die Lösung in #18943 bzgl. `maxAcPower` noch nicht hundertprozentig funktioniert. Ich hatte die hier geschilderte Problematik bereits dort erwähnt, aber leider keine Reaktion erhalten, weswegen ich das Regelverhalten von evcc nochmals detailliert darstellen möchte.

Zunächst die relevanten Bedingungen.

**Geräte**
Wechselrichter: Victron Multiplus II 5000 als ESS mit Nulleinspeisung (bzw. -30W) konfiguriert
Energiemessgerät: Carlo Gavazzi EM24 
Ladepunkt: go-e Charcher (einphasig betrieben)

**Konfiguration evcc**
maxAcPower: 4000W
priority soc: 70%

[evcc.yaml.txt](https://github.com/user-attachments/files/19615884/evcc.yaml.txt)

Als Erstes nun das fast täglich vorkommende Szenario, bei dem über mehrere Stunden nahezu identische Bedingungen vorliegen, da die PV-Anlage dauerhaft über 6kW produziert, wovon konstant ca. 2,5 kW verbraucht werden (Haus + Heizstab für Warmwasser) und der Rest in die Hausbatterie geladen wird. Da der Wechselrichter maximal 4,4kW liefern kann, diese Leistung unter Dauerlast aber auf ca. 4kW sinkt, bleiben für die Autobatterie theoretisch noch mindestens 1,5kW übrig. evcc berechnet jedoch gerade mal ca. 1,1kW (ca. 5 A), was nicht ausreicht um den Ladevorgang zu starten. Die Differenz zwischen "Verbrauch" in evcc "AC-Lasten" in der Victron-App beträgt ca. 300W.

[evcc.log](https://github.com/user-attachments/files/19615885/evcc.log)

```
[site  ] DEBUG 2025/04/04 10:49:06 ----
[lp-1  ] DEBUG 2025/04/04 10:49:06 charge power: 0W
[lp-1  ] DEBUG 2025/04/04 10:49:06 charge currents: [0 0 0]A
[site  ] DEBUG 2025/04/04 10:49:06 battery 1 power: -3360W
[site  ] DEBUG 2025/04/04 10:49:06 grid power: -23W
[site  ] DEBUG 2025/04/04 10:49:06 battery 1 soc: 80%
[site  ] DEBUG 2025/04/04 10:49:06 grid currents: [-10.8 10.1 1.3]A
[site  ] DEBUG 2025/04/04 10:49:06 pv 1 power: 6185W
[site  ] DEBUG 2025/04/04 10:49:06 pv 1 excess DC: 2185W
[site  ] DEBUG 2025/04/04 10:49:06 site power: -1098W
[lp-1  ] DEBUG 2025/04/04 10:49:06 charge total import: 830.186kWh
[lp-1  ] DEBUG 2025/04/04 10:49:06 charger status: B
[lp-1  ] DEBUG 2025/04/04 10:49:06 next soc poll remaining time: 10s
[lp-1  ] DEBUG 2025/04/04 10:49:06 pv charge current: 4.77A = 0A + 4.77A (-1098W @ 1p)
[site  ] DEBUG 2025/04/04 10:49:16 ----
```

[image]

[image]

Als Zweites die Darstellung eines Ladevorgangs im Modus Min+PV. Hier kommen mir folgende Punkte sehr merkwürdig vor:
1. Laut evcc-Log werden nur ca. 3,5A! verfügbarer Ladestrom berechnet!
2. Laut evcc-Screenshot werden insgesamt 4,5kW ausgegeben (Verbrauch 3,1kW + Ladepunkt 1,4kW) bei 0W Netzbezug! Das schafft der Wechselrichter nicht und tut er laut Victron-Screenshot (AC-Lasten 4076W) auch gar nicht. Auch hier ergibt sich wieder eine Differenz von bis zu 500W zwischen Berechnung tatsächlichem Verbrauch.

[evcc.log](https://github.com/user-attachments/files/19616021/evcc.log)

```
[site ] DEBUG 2025/03/28 11:50:40 ----
[lp-1 ] DEBUG 2025/03/28 11:50:40 charge power: 1357W
[lp-1 ] DEBUG 2025/03/28 11:50:40 charge currents: [6.19 0 0]A
[site ] DEBUG 2025/03/28 11:50:40 battery 1 power: -2816W
[site ] DEBUG 2025/03/28 11:50:40 battery 1 soc: 86%
[site ] DEBUG 2025/03/28 11:50:40 grid power: -1W
[site ] DEBUG 2025/03/28 11:50:40 pv 1 power: 7301W
[site ] DEBUG 2025/03/28 11:50:40 pv 1 excess DC: 3301W
[site ] DEBUG 2025/03/28 11:50:40 grid currents: [-11.3 10.3 1.4]A
[site ] DEBUG 2025/03/28 11:50:40 site power: 584W
[lp-1 ] DEBUG 2025/03/28 11:50:40 detected active phases: 1p
[lp-1 ] DEBUG 2025/03/28 11:50:40 charge total import: 817.769kWh
[lp-1 ] DEBUG 2025/03/28 11:50:40 charger status: C
[lp-1 ] DEBUG 2025/03/28 11:50:40 vehicle soc: 48%
[lp-1 ] DEBUG 2025/03/28 11:50:40 pv charge current: min 6A > 3.46A (584W @ 1p, battery: false)
[site ] DEBUG 2025/03/28 11:50:50 ----
[... 155 lines cut ...]
[site  ] DEBUG 2025/04/04 10:49:26 pv 1 power: 6193W
[site  ] DEBUG 2025/04/04 10:49:26 pv 1 excess DC: 2193W
[site  ] DEBUG 2025/04/04 10:49:26 site power: -1155W
[lp-1  ] DEBUG 2025/04/04 10:49:26 charge total import: 830.186kWh
[lp-1  ] DEBUG 2025/04/04 10:49:26 charger status: B
[lp-1  ] DEBUG 2025/04/04 10:49:26 next soc poll remaining time: 4m50s
[lp-1  ] DEBUG 2025/04/04 10:49:26 pv charge current: 5.02A = 0A + 5.02A (-1155W @ 1p)
[site  ] DEBUG 2025/04/04 10:49:36 ----
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc 0.202.1

mfuchs1984, 2025-04-05:
Im der Energiebilanz der victron Screenshots fehlen immer ein paar 100 Watt, dazu kommt noch die residualpower.

>Victron Multiplus II 5000 als ESS mit **Nulleinspeisung** (bzw. -30W) konfiguriert

Nulleinspeisung ist grundsätzlich ein Problem, evcc braucht den Überschuss am Übergabepunkt ins Netz (und berechnet das, was in die Batterie geht noch unter Berücksichtigung von maxacpower ein). Wenn der Wechselrichter die Leistung künstlich begrenzt, um nicht einzuspeisen, sieht evcc diesen Überschuss nicht.

Alternativ kann man evcc auch ohne grid meter betreiben und den typischen Hausverbrauch in der residualpower modellieren https://docs.evcc.io/docs/faq#kann-ich-evcc-ohne-netzz%C3%A4hler-betreiben

VolkerK62, 2025-04-05:
```
[site  ] DEBUG 2025/04/04 10:49:16 battery 1 power: -3327W
[site  ] DEBUG 2025/04/04 10:49:16 grid power: -31W
[site  ] DEBUG 2025/04/04 10:49:16 pv 1 power: 6188W
[site  ] DEBUG 2025/04/04 10:49:16 pv 1 excess DC: 2188W
[site  ] DEBUG 2025/04/04 10:49:16 site power: -1070W
[lp-1  ] DEBUG 2025/04/04 10:49:16 pv charge current: 4.65A = 0A + 4.65A (-1070W @ 1p)
```
Du erzeugst 6188W ` pv 1 power: 6188W`
somit 2188W mehr als acmaxpower `pv 1 excess DC: 2188W`
Die Batterieladeleistung ` battery 1 power: -3327W`  wird als verfügbar angesehen, aber mit `excess DC` verrechnet
Verbleibt `2188W - 3327W = 1139W` zzgl `grid power: -31W` verbleiben -1170W `site power`.

closed 2025-04-05 by andig

vedje, 2025-04-06:
@VolkerK62: Vielen Dank für die Erläuterung der Berechnung von `site power`.

> Im der Energiebilanz der victron Screenshots fehlen immer ein paar 100 Watt

@mfuchs1984: Das sind die von mir erwähnten Verluste des Wechselrichters. In meinem Fall können die laut technischem Datenblatt von Victron bei Volllast ca. 10%, also bis zu 440W betragen. Idealerweise dürften sie bei der Berechnung des Ladestroms nicht berücksichtigt werden.

Anhand der Victron Daten lassen sich die Verluste exakt beziffern:
`PV-Ladegerät` - `Aufladen` - `AC-Lasten` - `Netz`

Victron-Screenshot 1: 6184 - 3360 - 2570 - 2 = **252W**
Victron-Screenshot 2: 7292 - 2816 - 4076 - 24 = **376W**

Ich frage mich daher, ob evcc die Verluste nicht ebenso per modbus ermitteln und in die Berechnung des Ladestroms einbeziehen kann oder ob ich wenigstens die Möglichkeit habe, per API Einfluss auf den Ladestrom zu nehmen, indem ich die Verluste regelmäßig bekannt mache?

Ich hatte auch schon versucht, die Verluste pauschal mit 400W zu definieren, indem ich `maxAcPower` auf 4400W erhöht habe, aber aus einem mir noch unbekannten Grund hat das nicht gereicht. Erst bei 4800W wurden 6A Ladestrom erreicht. Problem war dann allerdings, dass nach einiger Zeit der Wechselrichter die nun benötigten 4.400 Leistung nicht mehr zur Verfügung stellen konnte und letztlich 400W aus dem Netz bezogen hat.

premultiply, 2025-04-06:
Relevant ist primär die Messung des Überschusses als Wirkleistung durch das Gridmeter.

Ich denke hier liegt das Grundproblem.

Und OT: Verluste gehen _ab_, also würde nach deiner Rechnung evcc die Lage dann eher zu optimistisch sehen.

vedje, 2025-04-06:
> Verluste gehen _ab_, also würde nach deiner Rechnung evcc die Lage dann eher zu optimistisch sehen.

@premultiply: Das verstehe ich noch nicht ganz, denn in meinen konkreten Fällen wären die Werte doch eher realistisch. Aber davon abgesehen, kann ich das "Grundproblem" irgendwie lösen? Der oben erwähnte Vorschlag mit `residualPower` wäre für mich jedenfalls keine geeignete Alternative.

premultiply, 2025-04-06:
Wenn der Batteriespeicher selbst ohnehin versucht einzuspeisen (-30W?) kannst du höchstens noch probieren `residualPower` auf 0 zu setzen oder mit den Thresholds zu experimentieren.

vedje, 2025-04-06:
Danke, dann probiere ich das mal.