evcc-io/evcc #20416: Support for 3 configurations for 3 different strings on Open-Meteo forecast

fugazzy, 2025-04-05:
I have 3 strings of PV with different KwP (amount of modules) on different angles dec (15, 35) and ac (70, 160, 340)

I would like to be able to define these 3 strings in 3 config entries, which then are summated.

SolarPower2024, 2025-04-05:
You already can do that. https://docs.evcc.io/en/docs/tariffs#pv-forecast

thecem, 2025-04-05:
There is no az of 340! Only -180 to 180! South azimuth!

fugazzy, 2025-04-05:
> You already can do that. https://docs.evcc.io/en/docs/tariffs#pv-forecast

Thank you I missed that one. I still need to get it working . . .

fugazzy, 2025-04-05:
> There is no az of 340! Only -180 to 180! South azimuth!

Thank you for pointing this out. I use the same provider in HomeAssistant and there the config is 0 to 360 deg. . . 
`Azimuth

The azimuth range for this integration is 0 to 360 degrees, with 0 being North, 90 being East, 180 being South, and 270 being West. If you have a negative azimuth, add 360 to it to get the correct value. For example, -90 degrees should be entered as 270 degrees.`

closed 2025-04-05 by andig